tttapa/MIDI_controller #1: Encoder_example compilation terminated

DH1KLM, 2017-04-14:
encoder_example did not compile.

Arduino: 1.6.5 (Windows 7), TD: 1.36, Platine: "TeeOnArdu (Leo on TeensyCore), MIDI, German"

In file included from C:\Users\DH1KLM\Documents\Arduino\libraries\MIDI_controller-master\src/MIDI_controller.h:9:0,
                 from Encoder_example.ino:20:
C:\Users\DH1KLM\Documents\Arduino\libraries\MIDI_controller-master\src/RotaryEncoder.h:6:21: fatal error: Encoder.h: No such file or directory
 #include "Encoder.h"
                     ^
compilation terminated.
Fehler beim Kompilieren.

Same happens if i change to Teensy.

tttapa, 2017-04-14:
Please follow the instructions in the [Readme](https://github.com/tttapa/MIDI_controller/blob/master/README.md), and install the dependencies. You don't have to use TeeOnArdu, you can just delete it from your _hardware_ folder.
You'll probably have to upgrade the Arduino core as well, you need version 1.6.8 or higher.

closed 2017-04-14 by tttapa

DH1KLM, 2017-04-14:
TeeOnArdu is needed for my Sparkfun Pro Micro to add MIDI.
Also own a Teensy++ 2.0, all dependencies are installed as mentioned in your readme.
I have already compiled other MIDI sketches successfully.
1.6.8 is not supported by Teensyduino. I will try a later version and report back.

tttapa, 2017-04-14:
No, TeeOnArdu is not needed, use the Arduino Micro as the board setting, the MIDI Controller library uses the MIDIUSB library.
You need the PJRC encoder library.
Teensyduino 1.36 supports up to Arduino 1.8.2.

DH1KLM, 2017-04-14:
Teensyduino 1.36 supports Arduino versions 1.0.6 and 1.6.5-r5 and 1.6.9 and 1.6.12 and 1.6.13 and 1.8.1 and 1.8.2.
Installed 1.6.9, also installed all required libraries, have choosen Leonardo as Board and
now it compiled and upload the sketch without errors.
I could not find a Arduino Micro in the board list so i used Leonardo.
TeeOnArdu was used for another Sketch. 
So i will test in the next days with my Sketch.
Thank you for your support