tttapa/MIDI_controller #2: No midi output over usb Teensy

mattlamperouge, 2017-05-20:
Hello, I was trying your encoder example, and everything compiled fine but i don't get midi output.
Im using a Teensy 3.1 with MIDI as the Usb type and all the needed libraries and software installed.
The Led lights up when i move the encoder so the enc.refresh(); seems to work.
I'm used to send midi using the pjrc library, like usbMIDI.sendControlChange for example.
How does your library send midi over USB?
Do i need to setup anything else?

tttapa, 2017-06-07:
The library uses the same function to send MIDI as usbMIDI.sendControlChange. Not sure why it's not working. Have you tried using a MIDI event analyzer? 
[https://github.com/tttapa/MIDI_controller/blob/master/src/USBMidi.cpp#L47](https://github.com/tttapa/MIDI_controller/blob/master/src/USBMidi.cpp#L47)
[https://github.com/PaulStoffregen/cores/blob/master/teensy3/usb_midi.h#L110](https://github.com/PaulStoffregen/cores/blob/master/teensy3/usb_midi.h#L110)

Pieter

mattlamperouge, 2017-06-07:
Hi I used a midi analyser but didn't get any cc msg or value..

tttapa, 2017-06-07:
You could try uncommenting the DEBUG preprocessor directive: [https://github.com/tttapa/MIDI_controller/blob/master/src/USBMidi.cpp#L4](https://github.com/tttapa/MIDI_controller/blob/master/src/USBMidi.cpp#L4)
And check what it's doing in the serial monitor.

I probably won't have time to look further into it until July, my apologies.

Pieter

mattlamperouge, 2017-06-08:
Thanks i will rey and report back

Aldbz, 2017-06-25:
Hello,
I try to use the MIDI_Controller_finished_example, and when i verify the code, i have a error message:
```
Arduino : 1.6.4 (Linux), Carte : "Arduino Uno"

In file included from /home/Tungsten/Arduino/libraries/MIDI_controller-master/src/MIDI_controller.h:9:0,
                 from MIDI_Controller_finished_example.ino:21:
/home/Tungsten/Arduino/libraries/MIDI_controller-master/src/RotaryEncoder.h:6:21: fatal error: Encoder.h: No such file or directory
 #include "Encoder.h"
                     ^
compilation terminated.
```

I think i don't have the encoder.h, it was not in the files...

Thanks for all you made! It's huge!

tttapa, 2017-06-26:
Did you get it to work?
I tried it with a Teensy 3.2 today, and it worked just fine ...

Pieter

tttapa, 2017-08-17:
Closing this issue as I've not been able to reproduce it, and no activity for well over a month.

deksprime, 2017-09-14:
Hello all, I have also come across the same issue with Teensy 3.2. Maybe we could reopen this issue and discuss it more thoroughly? Thanks in advance.

deksprime, 2017-09-14:
@mattlamperouge  have you been able to fix this issue? I'm having the same problem, the LED is lighting up while I rotate the potentiometer knob but there isn't any midi output to my DAW of choice, which is in this case Mixxx. I will try to analyze it with a midi analyzer today and post the results here

tttapa, 2017-09-14:
Very strange. I do most of my testing on a Teensy 3.2, and I've never had any problems. 
Please check with a MIDI analyzer. Also post your exact code and library versions you are using.

Note that you have to close your DAW and reopen it after uploading new firmware to the Teensy/Arduino.
Unplugging the Teensy is not necessary (at least not in Linux, but your mileage may vary if you're on Windows or Mac).

Pieter

deksprime, 2017-09-14:
-Arduino version 1.8.1 (it was included in the supported versions on Teensyduino installer)
-Teensy Loader 1.33 (don't know if this is the same version as teensyduino)
-Teensyduino 1.38
-I'm on Ubuntu Studio 16.04 with kernel 4.4.0-93
-USB type is set to MIDI from the Arduino IDE
-Teensy loader is opened (does this matter or should this be opened for only uploading?)
Here the led is blinking while I rotate the potentiometer so that I know there's nothing wrong with the hardware setup(at least with the pot setup), but no midi over USB is sent.

the exact code is: 
```
#include <MIDI_controller.h> // include the library

const static byte Channel_Volume = 0x7; // controller number 7 is defined as Channel Volume in the MIDI implementation.
const static size_t analogAverage = 8; // Use the average of 8 samples to get smooth transitions and prevent noise
//________________________________________________________________________________________________________________________________

Analog fader(A0, Channel_Volume, 1); // Create a new instance of the class 'Analog, called 'fader1', on pin A0, that sends MIDI messages with controller 7 (channel volume) on channel 1.

//________________________________________________________________________________________________________________________________

void setup(){
  USBMidiController.blink(LED_BUILTIN);  // flash the built-in LED (pin 13 on most boards) on every message
  USBMidiController.setDelay(15);  // wait 15 ms after each message not to flood the connection
  USBMidiController.begin();  // Initialise the USB MIDI connection
  delay(1000); // Wait a second...
  fader.average(analogAverage); // Use the average of 8 samples to get smooth transitions and prevent noise
  fader.map([](int value){return (int)((unsigned long)value*value/1023);}); // Map to a quadratic curve
}

//________________________________________________________________________________________________________________________________

void loop(){
  fader.refresh(); // refresh the fader (check whether the input has changed since last time, if so, send it over MIDI)
}
```
Which is the Analog_map_example from this repo. I suppose it is something not related to this specific code since I have tried many other analog potentiometer examples using the USBMidi library but no luck. Although my problem might be unrelated, the only similar problem on the web I could find was underneath this issue,so a little help or any suggestions would make a huge difference on my side and be so much appreciated! :)

tttapa, 2017-09-19:
I just tried it in Ubuntu 16.04 and Ubuntu 14.04 Studio, without any problems.
I used the Analog_map_example on a Teensy 3.2 with Arduino 1.8.4, Teensyduino 1.39 and MIDI_Controller 2.1.0.
Here are my ALSA MIDI connections:
[image]
And here's the output from GMIDImonitor (also works in KMidimon and Midisnoop):
[image]

Is the MIDI device detected? 
Open a terminal and execute `lsusb`. It should show up as _Van Ooijen Technische Informatica Teensyduino MIDI_
Then try `amidi -l`. It should show _IO  hw:1,0,0  Teensy MIDI MIDI 1_ or something similar.

deksprime, 2017-09-27:
Tried the steps that you have recommended. This time it worked, although I don't have any previous reference whether if the lsusb and amidi -l worked becaused I have only tried them this time and this time the Teensy happened to work. Thanks for your time and the extensive reply though. I think this issue can be closed since the other guy on the thread is not responding. Thank you dude.

closed 2017-09-27 by tttapa