tttapa/MIDI_controller #6: Uno issue

melph11, 2017-08-13:
Hi
Thanks for your work on this. 
I have been trying to make an uno work for midi CC, say CC11 and CC01.

I can successfully upload your uno example script (MIDI_Controller_Uno_Analog) and can flash the uno asper your instructions. I amended the script to use just one slide slider (fader) as a test..

#define CHANNEL  1 // Send all messages on channel 1.

int controllers[NUMBER_OF_ANALOG_INPUTS] = { 
  0x11
};  

If I connect to the computer, Midi is sent but as CC17? By changing to 0x11 shouldCC11 be sent?

I'm not going to use more than 4 faders so your script is brilliant, if only I can send the correct midi CC
Could you help steer me in the right direction

Many thanks

Mark

melph11, 2017-08-13:
Update - if I change0x11 to 0x05, it sends the CC message05. If I change this back to 0x11,it sends CC17?

Sorry, its probably really obvious.

melph11, 2017-08-14:
My Apologies......

It was late when I posted! Yes, change the code 0x11 to 0x0Bto get CC11...duh!

Great indestructible sketch and instructions. Manythanks

closed 2017-08-14 by melph11

tttapa, 2017-08-14:
You don't have to use hexadecimal representation, you could use 
``` int controllers[NUMBER_OF_ANALOG_INPUTS] = { 11 }; ``` as well, the actual machine code will be exactly the same, it's just a matter of preference.

Pieter