tttapa/MIDI_controller #8: Compilation error

vjdobermann, 2017-08-20:
Hi,
I'm trying to use your projet for mine.

when I want to upload an exemple (Analog_array_exemple.ino), the arduino IDE display an error message : 

```
Arduino : 1.8.3 (Mac OS X), Carte : "Arduino/Genuino Uno"

In file included from /Users/vjdobermann/Documents/Arduino/libraries/MIDI_controller-master/src/MIDI_controller.h:9:0,
                 from /Users/vjdobermann/Documents/Arduino/libraries/MIDI_controller-master/examples/Analog_array_example/Analog_array_example.ino:14:
/Users/vjdobermann/Documents/Arduino/libraries/MIDI_controller-master/src/RotaryEncoder.h:6:21: fatal error: Encoder.h: No such file or directory
 #include "Encoder.h"
                     ^
compilation terminated.
exit status 1
Erreur de compilation pour la carte Arduino/Genuino Uno
```
what does it means ?

vjdobermann, 2017-08-20:
finally I've founded the root cause, a file called Encoder.h is missing, it can be downloaded [here](https://github.com/ericbarch/arduino-libraries/blob/master/Encoder/Encoder.h).
No more compilation error after installing the files in the Libraries folder ;)
But it opens another issu but I close this topic and open a new one ;)

tttapa, 2017-08-20:
That's not the right Encoder library, it's an ancient copy of [PJRC's Encoder library](https://github.com/PaulStoffregen/Encoder).
Please find the [installation instructions](https://github.com/tttapa/MIDI_controller#installation) in the ReadMe.

vjdobermann, 2017-08-20:
thanks for the precision ;)
I've downloaded the dmg [here](https://www.pjrc.com/teensy/td_download.html). It asked me to select the arduino.app and install all the files.
I removed the Encoder folder from the libraries folder but the error message has came back...

tttapa, 2017-08-20:
Yes, you have to install the library (that can be found on GitHub), not Teensyduino. You don't really have to rename it.

Have you read the ReadMe and the [Wiki](https://github.com/tttapa/MIDI_controller/wiki/MIDI-over-USB) on how to use Hairless?

vjdobermann, 2017-08-20:
yes I read on the wiki how to use Hairless, I add `USBMidiController.begin(115200);` in the `void setup(){}` and set the same value in Hairless preferences. Here is [screen capture video](https://youtu.be/tez7KxQd_ek) 

I'm usine an Arduino UNO with an ATmega328p. Maybe this cause the issue. Should I use the Arduino Leo that seems have a native midi support ?
I'm a noob with Arduino, flashing the firmware with Hiduino seems too hardcore for me...

tttapa, 2017-08-20:
I haven't tried the newest version with Hairless, I'll try it when I get home!

Personally, I prefer to use the Leonardo, it's much easier. Or even better, use a Teensy, it costs the same as an official Leonardo, but has more I/O. 
A Chinese Leonardo clone will be much cheaper, of course.

tttapa, 2017-08-21:
Hi, 
I just tried it with an Arduino Uno Rev. 3 in Hairless 0.4 on Windows 10, and it worked just fine.
Here's the exact code I used:
```cpp
#include <MIDI_controller.h> // include the library

const static size_t analogAverage = 8; // Use the average of 8 samples to get smooth transitions and prevent noise

Analog* potentiometers[2]; // create an empty array of pointers to Analog objects

void setup(){
  USBMidiController.blink(LED_BUILTIN);  // flash the built-in LED (pin 13 on most boards) on every message
  USBMidiController.setDelay(15);  // wait 15 ms after each message not to flood the connection
  USBMidiController.begin(115200);  // Initialise the USB MIDI connection

  for(int i = 0; i < 2; i++) {
    potentiometers[i] = new Analog(A0+i, i, 1); // create a new Analog object and store it in the array
    potentiometers[i]->average(analogAverage); // Use the average of 8 samples to get smooth transitions and prevent noise
  }
}

void loop(){
  for(int i = 0; i < 2; i++) { // refresh all Analog inputs
    potentiometers[i]->refresh();
  }
}
```
Are you absolutely sure that the baud rate is correct?

For the record, I also tried it with HIDUINO, and it works as well.

Pieter

vjdobermann, 2017-08-23:
yes my baud rate is at 115200 in the arduino code + hairless

tttapa, 2017-08-23:
You can use something like this (untested):
```cpp
#include <MIDI_controller.h> // include the library

const size_t analogAverage = 8; // Use the average of 8 samples to get smooth transitions and prevent noise
const uint8_t channel = 1; // MIDI channel 1

Analog potentiometers[] = { // create an array of Analog objects
  {A0, 60, channel}, // a potentiometer connected to pin A0, sends MIDI CC events on controller number 60, on channel 1
  {A1, 61, channel}
};
const size_t nb_potentiometers = sizeof(potentiometers) / sizeof(Analog);

void setup(){
  USBMidiController.blink(LED_BUILTIN);  // flash the built-in LED (pin 13 on most boards) on every message
  USBMidiController.setDelay(15);  // wait 15 ms after each message not to flood the connection
  USBMidiController.begin(115200);  // Initialise the USB MIDI connection

  for(int i = 0; i < nb_potentiometers; i++) {
    potentiometers[i].average(analogAverage); // Use the average of 8 samples to get smooth transitions and prevent noise
  }
}

void loop(){
  for(int i = 0; i < nb_potentiometers; i++) { // refresh all Analog inputs
    potentiometers[i].refresh();
  }
}
```

closed 2017-09-11 by tttapa