tttapa/MIDI_controller #9: Mackie Protocol

RussCoty, 2017-09-11:
I have a prototype for mackie protocol controller that has multiple encoders controlling the same V-Pot output (controller 0x10 -0x17) but in different modes. (V-Pot is just mackie's name for the 8 encoders on a mackie control and modes are changed with a midi note)

Library works great thanks. But I have to add a note on command before each of my encoders sends a CC, this works fine...except in JOG mode where the note commands overwhelm the CC.

So really I just want to send the note commad each time a NEW controller is touched not before each CC command.

I added a variable to the function like this in RotaryEncoder.cpp and RotaryEncoder.h 

RotaryEncoder::RotaryEncoder(uint8_t pinA, uint8_t pinB, uint8_t controllerNumber, uint8_t channel, int speedMultiply, uint8_t pulsesPerStep, relativeCCmode mode, int mackieID)

Where MackieID is the note command that would normally change the V-Pot mode for the 8 encoders on a mackie control. 

so I have this in RotaryEncoder.cpp:

`void RotaryEncoder::refresh() // Check if the encoder position has changed since last time, if so, send the relative movement over MIDI
{
  long value = enc.read();
  long difference = (value - oldVal) / pulsesPerStep;   
  if (difference != 0)
    {
    USBMidiController.send(0x80, 1, mackieID, 127); // Send note to change the function of CCs 0x10 to 0x17 
    }

 while  (difference != 0)
   {
    if (difference > 15) // constrain relative movement to +/-15 for Mackie Control Universal compatibility
      difference = 15;
    if (difference < -15)
      difference = -15;
      
    uint8_t msgVal = mapRelativeCC(difference * speedMultiply, mode);
    if (bankEnabled && !digitalRead(bankPin))                        // if the bank mode is enabled, and the bank switch is in the 'alternative' position (i.e. if the switch is on (LOW))                                                                
      USBMidiController.send(CC, altChannel, altController, msgVal); // send a Control Change MIDI event with the 'alternative' channel and controller number
    else                                                             // if the bank mode is disabled, or the bank switch is in the normal position
      USBMidiController.send(CC, channel, controllerNumber, msgVal); // send a Control Change MIDI event with the normal, original channel and controller number
      oldVal += difference * pulsesPerStep; // If difference was in [-15, 15], difference * pulsesPerStep == value,
    // otherwise difference * pulsesPerStep == the value on the computer (after receiving the MIDI event above)

    value = enc.read();
    difference = (value - oldVal) / pulsesPerStep;
   }
}
`

I have tried adding another variable to store the lastSentMackieID either globally or for each instance of the RotaryEncoder class but can't seem to get it working. Any ideas would be greatly appreciated.

tttapa, 2017-09-14:
Have you tried making lastSentMackieID static?
In `RotaryEncoder.h`:
```cpp
...
  private:
    static uint8_t lastSentMackieID = 0;
...
```
In `RotaryEncoder::refresh`:
```cpp
...
  if (difference != 0 && lastSentMackieID  != this->mackieID)
  {
    USBMidiController.send(NOTE_ON, 1, this->mackieID, 127); // Send note to change the function of CCs 0x10 to 0x17 
    lastSentMackieID = this->mackieID;
  }
...
```
Pieter

RussCoty, 2017-09-18:
Brilliant thanks...I thought I had tried something similar but I'm going to try again today. 

I'll let you know how it goes.

RussCoty, 2017-09-18:
Thanks Pieter,

IDE Doesn't seem to like that 

error: ISO C++ forbids in-class initialization of non-const static member 'RotaryEncoder::lastSentMackieID'
   static uint8_t lastSentMackieID = 0;
                                     
Though this works:
at the top of the main sketch:

`static uint8_t lastSentMackieID;`

at the top of Rotary Encoder.cpp

`#include "RotaryEncoder.h"
uint8_t lastSentMackieID = 0;`

Not sure if others would find this useful but it allows you to have an encoder that always calls EQ for example and one that will always call pan etc. when using mackie protocol. Could be a cool feature.

I'm just working on multiplexing the encoders now with some 4051 ICs, turning off interrupts and manually polling seems to be my plan. Do you know how I might get arround the bank switches? I want to be able to multiplex the swirches too but I'm not sure how this would work if the address line is scanning all the time and I am looking at a differnt switch....maybe an array of all the last read switch values?

tttapa, 2017-09-19:
My bad. Moving the definition to the cpp file is the right solution:
```cpp
uint8_t RotaryEncoder::lastSentMackieID = 0;
```
Keep the declaration in the header file:
```cpp
private:
    static uint8_t lastSentMackieID = 0;
```
Have you tried the multiplexers yet? I'm not sure if it's going to be fast enough though.

I'm not sure what you mean by getting around the bank switches.
The latest version has an improved bank system, but I'm still working on the documentation and examples.

Pieter

RussCoty, 2017-09-19:
Yeah just trying the multiplexers now and it looks to be fast enough for manual control even with an UNO. A teensy should presumably be faster I hope. But it feels responsive in my DAW. 

I'm currently using digitalwrite commands on  the address pins to test it, but hopefully direct port access will speed that up too. 

Regards the bank pins I want to use the push switches on the encoders to alter the function of each encoder (plus have some other buttons too). 

I'm thinking a button matrix. 

I want to multiplex these too, at least for the columns, ideally will I have one master set of address pins like an address bus. 

My concern is that the multiplexing will mean that the push switch in the encoder might not be connected to the input pins at the correct time. Maybe that's a circuit design thing.

tttapa, 2017-09-19:
I think the best place to get help with the buttons would be the Arduino forum.
If you need heavy multiplexing, it's probably best to write your own library.
If you have any questions about how to adapt the library, feel free to ask.

Pieter

closed 2017-09-19 by tttapa

RussCoty, 2017-09-19:
I tell a lie it gets a bit slow when you poll 8 encoders...but I think I can remedy that.

RussCoty, 2017-09-19:
Awesome thanks Pieter.

tttapa, 2017-09-19:
You could probably use multiplexers or I/O expanders with a pin change interrupt output. This means that you won't have to poll it.
If you use a Teensy 3, you won't need to do this, because all digital pins have interrupt capabilities. It makes more sense to save pins on the buttons and analog inputs, because they don't require accurate timing.

tttapa, 2017-09-19:
If you open a thread on the Arduino forum, you can post the link here, in case I miss it.

RussCoty, 2017-09-19:
I wanted to be able to scale this to a rather large encoder count.

I suspect a larger IO number MCU might be the answer, or maybe using a teensy for each group of encoders and merging i2c data in another unit. 

Interestingly the multiplexing works ok up to about 5 or 6  encoders but a 1ms delay seems to be needed after the address bus port manipulation. Otherwise extra midi data gets sent.

I'll do a bit more work and start a forum post as you suggest.

RussCoty, 2017-09-20:
One final post, I know the issue is closed but I think you will like this.

Adding a 5 microsecond delay after the address port manipulation before the refresh seems to allow me to read plenty of encoders.

I have 16 running quite smoothly on a teensy and an increase in the delay to 100 microseconds lets me run the sketch successfully on a cheap Chinese uno.

PORTB = B00000001;
delayMicroseconds(5);
encoder0.refresh; //an encoder on multiplexers going to pins 2 and 3 
encoder 8.refresh; //an encoder on multiplexers going to pins 4 and 5

As you suggested I'll try something like a PCF8575 IO expanded with interrupts but for now this seems to feel super smooth and responsive. 

Thanks again for all the help.

tttapa, 2017-09-20:
Interesting, I didn't know that multiplexing encoders works so well.