tttapa/MIDI_controller #12: Add LED on/off with MIDI send

devwmbr, 2017-10-14:
Hello,
I've decided to use your great library in guitar foot controller for Guitar Rig 5 project.
May I ask how to turn on/off LED for momentary button during MIDI send? For example for Delay on/off or Reverb on/off. I button pressed, MIDI is send do Guitar Rig, LED is turned on. Press button again send MIDI and LED is turned off.

Regards
devwmbr

tttapa, 2017-10-22:
You could edit the source code for the Digital class to add this functionality. 
Or you could use my Control Surface library. It should be fully functional, but I haven't added the documentation yet, and the API will probably change a little bit in upcoming releases. 

An example would look like this:
```cpp
#include <Control_Surface.h>

const uint8_t controllerNumber = 0x10; // Controller number for delay effect
const uint8_t channel = 1; // MIDI channel 1
const uint8_t velocity = 127; 

const pin_t buttonPin = 2;
const pin_t ledPin = 13;

Digital delayButton = { buttonPin, controllerNumber, channel, 127 };

MIDI_LED delayLed = { ledPin, controllerNumber, channel };

void setup() {}

void loop() {
  Control_Surface.refresh();
}
```

devwmbr, 2017-10-22:
> Wysłana:Sun, 22 Oct 2017 17:28:01 +0200
> Do:tttapa/MIDI_controller <MIDI_controller@noreply.github.com>
> DW:devwmbr <beardini@gmail.com>,Author <author@noreply.github.com>

Hi,

Thank you very much for answer. I edited your srce files and I added some methods as much as I possibly  can. Everyday I develop in Asp.net/C#, C++ is little bit difference with .h files.

I will send you my changes and results with LEDs and LCD tomorrow morning.

I will look of course your proposal.

Thank you again.

Waldemar

Wysłane z mojego ASUS

-------- Wiadomość oryginalna --------

devwmbr, 2017-10-23:
Hello,
Surface library is all-in-one with MIDI Controller within?

tttapa, 2017-10-31:
Yes, they use some of the same files. The Control Surface library was created first, and then everything related to MIDI input was removed, resulting in the MIDI Controller library.

closed 2017-12-25 by tttapa