tttapa/MIDI_controller #18: Can I receive values over midi?

Jorgen-VikingGod, 2018-01-14:
## Can I receive values over MIDI?
Is there a way to receive MIDI values? As I can see in the examples, they all just send values over MIDI. But I can not find a example to receive values...

I want to have a kind of feedback. If my midi software changes a value, I want to receive this and light up some LEDs.
f.e. I want a LED-ring around my rotary encoder and want to highlight the current value.

## Steps to reproduce the problem

### Hardware
**Arduino board**:  Arduino MEGA, Teensy 3.2

### Software versions:
**MIDI Controller library**: 3.0.0

tttapa, 2018-01-17:
I have been working on a second library that supports MIDI input as well. However, it's not documented yet.
[https://github.com/tttapa/Control-Surface](https://github.com/tttapa/Control-Surface)

devwmbr, 2018-01-18:
Will it be possible to receive bank name or preset name from guitar Amp simulator, e.g. GuitarRig, Overloud TH3 or similar?

Jorgen-VikingGod, 2018-01-18:
great to see that some progress is coming.

mungewell, 2018-06-25:
Did you consider SysEx messages to 'just' set bytes within a memory?

I'm certainly not an expert, but in reversing a Akai MPD218 I can pretty much drive/control the device via SysEx.

closed 2019-07-31 by tttapa