tttapa/MIDI_controller #19: errors with installation

Skytrobb, 2018-01-15:
Hey there, I followed every step in your install instruction and I get an error when
trying to update the drivers for ATmega16U2 which prevents me from updating the firmware.
I get "windows could not find the drivers for the device" every time I specify the path
c:/Windows/program files(x86)/ATMEL/FLIP 2.4.6/usb. I'm using Windows 10 64bit, latest MIDI version and an Arduino UNO.
I noticed your instructions use flip 3.4.7, I thought maybe I got the wrong version but as far
as I can tell, 2.4.6 is the latest flip version for Windows. I did lots of research and I can't seem
to find anyone else with this issue. Any ideas?

tttapa, 2018-01-17:
Microchip recently changed the download page, and version 3.4.7 is nowhere to be found ...  
I'm a Linux user myself, and I still have version 3.4.7 (+drivers) installed on my Windows machines, so I can't really help you, I'm afraid.  
You could try dfu-programmer instead of Flip, that's what I use on Linux, and they seem to have a Windows version as well.

Skytrobb, 2018-01-17:
I did some pretty extensive research on this and it looks like flip hasn't
been successfully tested on windows 10 so as far as I can tell there is no
way for windows 10 users to install this, I had to use the dfu-programmer
on a virtual machine to get it working on the windows side, everything
seems to be working fine now.

closed 2018-02-21 by tttapa