tttapa/MIDI_controller #21: appleMIDI Ethernet

FoxPerdana, 2018-02-04:
How to using this library with AppleMIDI Library? Im not to good In arduino stuff, I hope can help me trough it.

I dont know where to put the
 "const uint8_t velocity = 0b1111111; 
   const uint8_t C4 = 60;  "

and

"Digital button(2, C4, 1, velocity);"

and

"MIDI_Controller.refresh();"

Im using Arduino Uno with Ethernet Shield

Library version 3.0.1 and AppleMIDI

Please dont judge or laugh because my stupid code I made

#include <MIDI_Controller.h>

#include <Ethernet.h>

#include "AppleMidi.h"

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};

unsigned long t0 = millis();
bool isConnected = false;

APPLEMIDI_CREATE_DEFAULT_INSTANCE(); // see definition in AppleMidi_Defs.h

// Code size:
//
// IDE 1.8.3
//
// Arduino Ethernet / Uno
// Sketch uses 23976 bytes (74%) of program storage space. Maximum is 32256 bytes.
// Global variables use 969 bytes (47%) of dynamic memory, leaving 1079 bytes for local variables. Maximum is 2048 bytes.
const uint8_t velocity = 0b1111111; // Maximum velocity (0b1111111 = 0x7F = 127)
const uint8_t C4 = 60;              // Note number 60 is defined as middle C in the MIDI specification
Digital button(2, C4, 1, velocity);
// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void setup()
{
  // Serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  Serial.print(F("Getting IP address..."));

  if (Ethernet.begin(mac) == 0) {
    Serial.println();
    Serial.println(F("Failed DHCP, check network cable & reboot"));
    for (;;)
      ;
  }

  Serial.println();
  Serial.print(F("IP address is "));
  Serial.println(Ethernet.localIP());

  Serial.println(F("OK, now make sure you an rtpMIDI session that is Enabled"));
  Serial.print(F("Add device named Arduino with Host/Port "));
  Serial.print(Ethernet.localIP());
  Serial.println(F(":5004"));
  Serial.println(F("Then press the Connect button"));
  Serial.println(F("Then open a MIDI listener (eg MIDI-OX) and monitor incoming notes"));

  // Create a session and wait for a remote host to connect to us
  AppleMIDI.begin("test");

Digital button(2, C4, 1, velocity);

}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void loop()
{
  // Listen to incoming notes
  AppleMIDI.run();

  MIDI_Controller.refresh();

Thank You

tttapa, 2018-02-04:
RTP MIDI is currently not supported. Although, I'm planning on adding it in the near future.  
It should be as simple as creating a new class that derives from the [MIDI_Interface](https://github.com/tttapa/MIDI_controller/blob/master/src/MIDI_Interfaces/MIDI_Interface.h) class. You have to implement two virtual functions: `sendImpl` and `refresh`.

```cpp
class RTPMIDI_Interface : public MIDI_Interface
{
  public:
    RTPMIDI_Interface() {
        // Constructor (initialize RTP MIDI etc.)
    }

    bool refresh() {
        // Ignore MIDI input:
        // Empty the incomming MIDI buffer, return false if it's empty, return true otherwise
    }

  protected:
    void sendImpl(uint8_t m, uint8_t c, uint8_t d1, uint8_t d2) {
       // Send a three-byte MIDI message
    }
    void sendImpl(uint8_t m, uint8_t c, uint8_t d1) {
        // Send a two-byte MIDI message
    }
};
```

FoxPerdana, 2018-02-04:
Sorry I'm confuse. Should I do Replace the original MIDI_Interface.h or create new class? like RTPMIDI_Interface.h?

FoxPerdana, 2018-02-04:
Please tell me If I do wrong.  I'm create a new class called RTPMIDI-Interface.h like this ; 
//--------------------------------------------------------------------------------------------------------
#ifndef RTPMIDI_INTERFACE_H_
#define RTPMIDI_INTERFACE_H_

#include "Arduino.h"

#include "Settings/Settings.h"

#define MIDI_BAUD 31250

// #define IGNORE_SYSEX

const uint8_t NOTE_OFF = 0x80;
const uint8_t NOTE_ON = 0x90;
const uint8_t KEY_PRESSURE = 0xA0;
const uint8_t CC = 0xB0, CONTROL_CHANGE = CC;
const uint8_t PROGRAM_CHANGE = 0xC0;
const uint8_t CHANNEL_PRESSURE = 0xD0;
const uint8_t PITCH_BEND = 0xE0;

const uint8_t SysExStart = 0xF0;
const uint8_t SysExEnd = 0xF7;

class RTPMIDI_Interface : public MIDI_Interface
{
  public:
    RTPMIDI_Interface() {
        // Constructor (initialize RTP MIDI etc.)
    }

    bool refresh() {
        // Ignore MIDI input:
        // Empty the incomming MIDI buffer, return false if it's empty, return true otherwise
    }

  protected:
    void sendImpl(uint8_t m, uint8_t c, uint8_t d1, uint8_t d2) {
        // Send a three-byte MIDI message
    }
    void sendImpl(uint8_t m, uint8_t c, uint8_t d1) {
        // Send a two-byte MIDI message
    }
};

#endif // RTPMIDI_INTERFACE_H_
-----------------------------------------------------------------------------------------------------//
and add string on MIDI_Controller.h

//--------------------------------------------------------------------------------------------------------
#include "./MIDI_Interfaces/RTPMIDI_Interface.h"
-----------------------------------------------------------------------------------------------------//

I get this error
//----------------------------------------------------------------------------------------------
In file included from C:\Users\......\Documents\Arduino\libraries\MIDI_controller-master\src/./MIDI_Controller/../MIDI_Interfaces/USBMIDI_Interface.h:4:0,

                 from C:\Users\......\Documents\Arduino\libraries\MIDI_controller-master\src/./MIDI_Controller/MIDI_Controller_Class.h:4,

                 from C:\Users\.......\Documents\Arduino\libraries\MIDI_controller-master\src/MIDI_Controller.h:10,

                 from C:\Users\.......\Documents\Arduino\tappa_midi_ethernet\tappa_midi_ethernet.ino:1:

C:\Users\......\Documents\Arduino\libraries\MIDI_controller-master\src/./MIDI_Controller/../MIDI_Interfaces/MIDI_Interface.h:21:15: note: 'const uint8_t SysExEnd' previously defined here

 const uint8_t SysExEnd = 0xF7;
---------------------------------------------------------------------------------------------------//

I'm pretty sure I do a mistake. but I'm confused, so where I did wrong?

FoxPerdana, 2018-02-08:
Well. I think Its better to wait the Update Version of MIDI Controller Library that support rtp stuff (AplleMIDI, etc). You can close this issue mr Tttapa

closed 2018-02-08 by FoxPerdana