tttapa/MIDI_controller #22: Please help, can't compile at all

iDaemo, 2018-03-04:
## Description of the problem or question

First of all I'm noob here, Need advice or suggestion that I can't compile code use this library at all even Example code it just show this warning (testing with example code)

`
In file included from /Users/nupuan/Documents/Arduino/libraries/MIDI_controller-3.0.1/examples/MIDI_Controller_finished_example/MIDI_Controller_finished_example.ino:20:0:
/Users/nupuan/Documents/Arduino/libraries/MIDI_controller-3.0.1/src/MIDI_Controller.h:7:2: error: #error Please select a 'Serial + MIDI' option in the 'Tools > USB Type' menu.
 #error Please select a 'Serial + MIDI' option in the 'Tools > USB Type' menu.
  ^
In file included from /Users/nupuan/Documents/Arduino/libraries/MIDI_controller-3.0.1/src/./MIDI_Controller/MIDI_Controller_Class.h:4:0,
                 from /Users/nupuan/Documents/Arduino/libraries/MIDI_controller-3.0.1/src/MIDI_Controller.h:10,
                 from /Users/nupuan/Documents/Arduino/libraries/MIDI_controller-3.0.1/examples/MIDI_Controller_finished_example/MIDI_Controller_finished_example.ino:20:
/Users/nupuan/Documents/Arduino/libraries/MIDI_controller-3.0.1/src/./MIDI_Controller/../MIDI_Interfaces/USBMIDI_Interface.h:7:21: fatal error: usb_dev.h: No such file or directory
 #include <usb_dev.h>
                     ^
compilation terminated.
Error compiling for board Teensy 2.0.`

it's no MIDI+Serial option there. And I think I’m already install all required.

### Hardware
Teensy 2.0

### Software versions:

**MIDI Controller library**: ? 3.0.0 ?  3.0.1
**Arduino IDE**: ? 1.8.4 ?  1.8.5
**Operating System**: ? Windows, OSX, Ubuntu, ... ?  OSX
**Operating System version**: ? 10, 16.04, ... ?  10.13.3
(**Teensyduino**): ? 1.39 ? 1.41
(**Encoder library**): ? 1.4.1 ?  1.4.1
(**MIDIUSB library**): ? 1.0.3 ? 1.0.3

DH1KLM, 2018-03-06:
Habe you tried the Teensy Branch?
https://github.com/tttapa/MIDI_controller/tree/teensy++2.0
I dont know if it works with Teensy 2.0, but it works for Teensy++ 2.0

You could also read https://github.com/tttapa/MIDI_controller/issues/20

iDaemo, 2018-03-06:
Thanks I will give it a try.

iDaemo, 2018-03-07:
It's work, Thanks DH1KLM

closed 2018-03-07 by tttapa