tttapa/MIDI_controller #23: Play/Pause PushButton like old SendNoteOn-SendNoteOff

hachimbala, 2018-03-07:
Hello, I am developing a MIDI controller that requires some different functions, I have all done but the only thing that I haven't is the Play/Pause assigned to one button, I want to know if there is some way to do it like in the older version with SendNoteOn and SendNoteOff.
I have been looking the switch example but it say that has a limited duration, and I want an undefined duration, that when you press the pushbutton one time it sends a NoteOn, and when you repress the button it sends the NoteOff command.

Thank you so much

### Software versions:

**MIDI Controller library**: 3.0.1
**Arduino IDE**:  1.8.5

tttapa, 2018-03-08:
Have you tried the [button example](https://github.com/tttapa/MIDI_controller/blob/master/examples/Ex.02.Button/Ex.02.Button.ino)?   
It works with all software I've come across so far. Latching should be done in the DAW software, not on the controller.

hachimbala, 2018-03-08:
Yes but I will use it with Traktor so I need the same button to do note on when pulsed and note off when pulsed again no when released

tttapa, 2018-03-08:
I don't think that's necessary. Please read §13.22.6 of the [Traktor manual](https://www.native-instruments.com/fileadmin/ni_media/downloads/manuals/TRAKTOR_2.11_Manual_English_0317.pdf).

hachimbala, 2018-03-08:
Thank you so much!!

closed 2018-03-11 by tttapa

guillecip, 2019-03-12:
Did you know how can I put a toogle button in Ableton?