tttapa/MIDI_controller #24: MIDI read

hachimbala, 2018-03-11:
Hello,

I was eondering how I can do a MIDI.read() (or something similar) to read the MIDI messages that Traktor sends to the Arduino and then use it.

In my void setup I inicialized with MIDI.begin();
So now only with MIDI.read() can I get the MIDI messages that Traktor send to the Arduino?

Thank you so much!

tttapa, 2018-03-14:
What board are you using?  
MIDI input is currently not supported.

hachimbala, 2018-03-14:
Mega

tttapa, 2018-03-14:
MIDI input is discarded by the library. You can just change this function to allow for MIDI input, and then use the MIDI library.
[https://github.com/tttapa/MIDI_controller/blob/master/src/MIDI_Interfaces/SerialMIDI_Interface.h#L11-L18](https://github.com/tttapa/MIDI_controller/blob/master/src/MIDI_Interfaces/SerialMIDI_Interface.h#L11-L18)

hachimbala, 2018-03-14:
Ok, but how I can open the incoming data communication, what should I have to change there?

tttapa, 2018-03-15:
Just comment out the entire function. Then in your main .ino sketch, add code for MIDI input using your MIDI library of choice.

hachimbala, 2018-03-15:
So I uncomment the whole function and then for the MIDI input I use other MIDI library,  right?

tttapa, 2018-03-15:
Yes.

hachimbala, 2018-03-15:
Ok, it is working! Thank you so much!

closed 2018-03-16 by tttapa