tttapa/MIDI_controller #25: buttons multiplexer

skizzzzz, 2018-03-19:
Hello,

first off all thanks for this wonderful libary! I'im a noob in programming and this helps a lot!!!

I was wondering if we can simply use multiplexer with buttons..?

I'm trying to modify the exemple:   	Ex.12.AnalogMultiplex

to make it work with buttons...without success.

Is it possible?

Thanks

tttapa, 2018-03-20:
Yes, it should be possible.  
What exactly did you try? Please post your complete code and schematic.

skizzzzz, 2018-03-21:
Great !
I will take the time this week to to explain it to you clearly!  (with fritzing that's i'm discovering).

skizzzzz, 2018-03-21:
Here it is

[image]

So I use an arduino 'pro micro' which correspond emulate a leonardo. (on my shema it's a micro..)

I tried differents pins of the arduino to connect my multiplexer like A3, '5' etc...

and use try several things like this code 
``

`
#include <MIDI_Controller.h> // Include the library

const uint8_t velocity = 0b1111111; // Maximum velocity (0b1111111 = 0x7F = 127)
const uint8_t C4 = 60;              // Note number 60 is defined as middle C in the MIDI specification

// Create an instance of 'AnalogMultiplex' with the output pin of the multiplexer connected to
// analog input pin A0 and the address pins connected to pins 2, 3 and 4.
AnalogMultiplex multiplexer(5, { 2, 3, 4 } );

// Create 8 new instances of the class 'Analog', on the 8 pins of the multiplexer,
// that send MIDI messages with controller 7 (channel volume) on channels 1 - 8

Digital button(multiplexer.pin(0), C4, 1, velocity);

void setup() {}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}`

tttapa, 2018-03-22:
Should be fixed in https://github.com/tttapa/MIDI_controller/commit/90b8796f8365faf28bcfa15eab8f70336d9381d6.

closed 2018-03-22 by tttapa

skizzzzz, 2018-03-22:
Woow. It works really well now. What a quick update. THANK YOU !