tttapa/MIDI_controller #28: Ca

userdsp, 2018-04-01:
Can you add support for CD74HC4067 please! I want to use banksellector with this mux like i can do it now with CD74HC4051

tttapa, 2018-04-01:
It should be supported already, what makes you think that it doesn't work?  

A digital multiplexer is kind of useles, since there is no difference in the implementation. If you really want to, you can use the AnalogMultiplex class with digital inputs.  
That's not really a good idea, because you can address multiple buttons much more efficiently using a button matrix or using shift registers.

userdsp, 2018-04-01:
"It should be supported already, what makes you think that it doesn't work?"
I guest you mean 4067? I have tested it and after I add a line   
{multiplexer.pin(8), MIDI_CC::Channel_Volume, 9},
whole program freezes.

userdsp, 2018-04-01:
My mistake! IDK why but now all 16 pins work but arduino freezes if I add 
pots.add(potentiometers, Bank::CHANGE_ADDRESS);

tttapa, 2018-04-01:
That's strange. What do you mean "freezes"? Does it stop running, or does it just stop sending MIDI messages? 
Try adding a blink without delay to the code:

```cpp
void setup() {
  // ...
  pinMode(LED_BUILTIN, OUTPUT);
}

void blink() {
  static unsigned long previousBlinkTime = millis();
  if (millis() - previousBlinkTime > 250) {
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
    previousBlinkTime += 250;
  }
}

void loop() {
  blink();
  // ...
}
```
If it stops blinking, you know it has entered an infinite loop (i.e. bad stuff happened).

userdsp, 2018-04-01:
If i add pots.add(potentiometers, Bank::CHANGE_ADDRESS); it stop sending any midi messages

tttapa, 2018-04-01:
>And what about Encoders thru mux?

Encoders through a mux are a little harder, since they require hardware interrupts.

userdsp, 2018-04-01:
i see that is not so big deal. It was just interesting for me. And i did add this code ... led stop blinks if pots.add(potentiometers, Bank::CHANGE_ADDRESS); is active

tttapa, 2018-04-01:
I'll see what I can do.

userdsp, 2018-04-01:
Sorry about your time! I start a fresh Scetch and everything its ok. maybe it was a conflict from another function witch was used for reading the first mux! Have a nice day and everything nice!

tttapa, 2018-04-01:
You posted this code:
```cpp
#include <MIDI.h>
#include "Controller.h"
#include <Shifty.h>
#include <MIDI_Controller.h>
//const uint8_t Channel = 10; // MIDI channel 1
//const uint8_t LController = 0x14; // MIDI Control Change controller number
//const uint8_t RController = 0x15; // MIDI Control Change controller number
const int speedMultiply = 1; // If the jog wheels or other encoders are too slow in your software, increase this value
// (it will be multiplied with the actual speed of the encoder, as the name implies.) Default is 1.

MIDI_CREATE_DEFAULT_INSTANCE();

Shifty myreg;

// Create a new instance of the class 'RotaryEncoder', called 'enc', on pin 2 and 3, controller number 0x14, on channel1,
// no change in speed (speed is multiplied by 1), it's used as a jog wheel, and the sign mode is set to two's complement.

RotaryEncoder encoderL(7, 9, 0x14, 6, speedMultiply, JOG, TWOS_COMPLEMENT);
RotaryEncoder encoderR(A4, A5, 0x15, 6, speedMultiply, JOG, TWOS_COMPLEMENT);
Bank pots1(16); // A bank with two address
BankSelector bankselector(pots1, 10, LED_BUILTIN, BankSelector::MOMENTARY); // A bank selector with a single toggle switch on pin 11 and an LED for feedback on pin 13

//int ledPin = 13;
//void OnNoteOn(byte channel, byte note, byte velocity) {
// digitalWrite(ledPin, HIGH); // Any Note-On turns on LED
//}
//void OnNoteOff(byte channel, byte note, byte velocity) {
// digitalWrite(ledPin, LOW); // Any Note-Off turns off LED
//}

AnalogMultiplex multiplexer1(A0, { 2, 3, 4, 5 } );

// Create 8 new instances of the class 'Analog', on the 8 pins of the multiplexer,
// that send MIDI messages with controller 7 (channel volume) on channels 1 - 8
Analog potentiometers1[] = {
{multiplexer1.pin(0), MIDI_CC::Channel_Volume, 1},
{multiplexer1.pin(1), MIDI_CC::Channel_Volume, 2},
{multiplexer1.pin(2), MIDI_CC::Channel_Volume, 3},
{multiplexer1.pin(3), MIDI_CC::Channel_Volume, 4},
{multiplexer1.pin(4), MIDI_CC::Channel_Volume, 5},
{multiplexer1.pin(5), MIDI_CC::Channel_Volume, 6},
{multiplexer1.pin(6), MIDI_CC::Channel_Volume, 7},
{multiplexer1.pin(7), MIDI_CC::Channel_Volume, 8},
{multiplexer1.pin(8), MIDI_CC::Channel_Volume, 9},
{multiplexer1.pin(9), MIDI_CC::Channel_Volume, 10},
{multiplexer1.pin(10), MIDI_CC::Channel_Volume, 11},
{multiplexer1.pin(11), MIDI_CC::Channel_Volume, 12},
{multiplexer1.pin(12), MIDI_CC::Channel_Volume, 13},
{multiplexer1.pin(13), MIDI_CC::Channel_Volume, 14},
{multiplexer1.pin(14), MIDI_CC::Channel_Volume, 15},
{multiplexer1.pin(15), MIDI_CC::Channel_Volume, 16}
};

//AnalogMultiplex multiplexer2(A1, { 2, 3, 4, 5 } );
//
// Create 8 new instances of the class 'Analog', on the 8 pins of the multiplexer,
// that send MIDI messages with controller 7 (channel volume) on channels 1 - 8
//Analog potentiometers2[] = {
// {multiplexer2.pin(0), MIDI_CC::Channel_Volume, 1},
// {multiplexer2.pin(1), MIDI_CC::Channel_Volume, 2},
// {multiplexer2.pin(2), MIDI_CC::Channel_Volume, 3},
// {multiplexer2.pin(3), MIDI_CC::Channel_Volume, 4},
// {multiplexer2.pin(4), MIDI_CC::Channel_Volume, 5},
// {multiplexer2.pin(5), MIDI_CC::Channel_Volume, 6},
// {multiplexer2.pin(6), MIDI_CC::Channel_Volume, 7},
// {multiplexer2.pin(7), MIDI_CC::Channel_Volume, 8},
// {multiplexer2.pin(8), MIDI_CC::Channel_Volume, 9},
// {multiplexer2.pin(9), MIDI_CC::Channel_Volume, 10},
[... 76 lines cut ...]
if (value >= 1) {
myreg.writeBit(1, HIGH);
break;
}
else {
myreg.writeBit(1, LOW);
break;
}
case 2:
if (value >= 1) {
myreg.writeBit(2, HIGH);
break;
}
else {
myreg.writeBit(2, LOW);
break;
}
}
}
```
A couple of remarks: including and initializing the MIDI library will probably cause conflicts. The MIDI_Controller library uses the same interface to write to.  
The MIDI Controller library includes a [library for driving shift registers](https://github.com/tttapa/MIDI_controller/wiki/Library-documentation#shift-registers), so there's no need for Shifty.h.

You can use a mux to make buttons. Just use `mux.pin(#)` as the pin number.  
If you have to read or write to the pin manually, you can use `ExtIO::digitalRead(pin)`, `ExtIO::digitalWrite(pin, state)`, etc.

tttapa, 2018-04-01:
Do you have any idea what caused it to crash before?

userdsp, 2018-04-01:
Thank you very much again! . Crash was causing a loop in MyCCPWMFunction

tttapa, 2018-04-02:
>The crash was causing a loop in MyCCPWMFunction

I would be interested in you original code that caused the crash. It could reveil a bug in the library.

I'm not sure what you mean by the command parameter? 
If you want to use normal momentary push buttons, you can use the Digital class to send Note events, or the [DigitalCC](https://github.com/tttapa/MIDI_controller/wiki/Custom-MIDI-controls#digitalcch) class to send Control Change events. If you have toggle switches, you can use the DigitalLatch class for Note events, or you could create your own control element for DigitalLatchCC (see wiki on how to do that).

userdsp, 2018-04-02:
I want to use push buttons with will act like toggle buttons. 
First press = HIGH
Second press = LOW

tttapa, 2018-04-02:
What software do you plan to use it with? All music programs I've come across do this toggling on the computer, not on the microcontroller.

The implementation is trivial, but I'm wondering if you really need it.  
I'm currently finishing some other parts of the library, but reworking the Digital and DigitalLatch classes is on my list as well, I'll see what I can do.

Would you be willing to send me the code that caused the crash before?

userdsp, 2018-04-03:
Ableton. I will post the code soon at possible, because i'm not at my home, and my PC is power OFF

closed 2018-04-04 by userdsp