tttapa/MIDI_controller #30: Problem with DigitalCC

TheProgWalrus, 2018-04-05:
Hi, 
I cannot manage to install the DigitalCC class in order to use the switches as MIDI_CC controls. Is there a release of the MIDI_CONTROLLER library with this class already implemented? Or is there a tutorial that can explain me how to do this? Many thanks.

tttapa, 2018-04-05:
Just create two new files (CTRL+SHIFT+N in the IDE): `DigitalCC.h` and DigitalCC.cpp`. Then just paste the code from the [wiki](https://github.com/tttapa/MIDI_controller/wiki/Custom-MIDI-controls) into the respective files.  
Finally, include the new file by using `#include "DigitalCC.h"` into your main sketch.  In this sketch, you can then instantiate DigitalCC objects.

TheProgWalrus, 2018-04-05:
Mamy thanks. And is  the switch  with this instruction?

TheProgWalrus, 2018-04-05:
*is the switch latched

tttapa, 2018-04-05:
No, it's a normal momentary pushbutton.

TheProgWalrus, 2018-04-05:
Thank you so much

closed 2018-04-05 by TheProgWalrus