tttapa/MIDI_controller #31: 10+1pcs CD74HC4067 multiplexers

mkothencz, 2018-04-06:
Hello,

I would like to build a MIDI Controller with 150 buttons, 16 faders, and 8 encoders.

I would like to multiplex the 150 buttons with 10+1pcs  CD74HC4067 multiplexers.
I use Arduino Leonardo.

I have problem:
My code is:

#include <MIDI_Controller.h>
const int speedMultiply = 1;

//BUTTONS
AnalogMultiplex multiplexer(8, { 9, 10, 11, 12, A2, A3, A4, A5 } );
Analog potentiometers[] = {
  {multiplexer.pin(0), 0x1, 1},
  {multiplexer.pin(1), 0x2, 1},
  {multiplexer.pin(2), 0x3, 1},
  {multiplexer.pin(3), 0x4, 1},
  {multiplexer.pin(4), 0x5, 1},
  {multiplexer.pin(5), 0x6, 1},
  {multiplexer.pin(6), 0x7, 1},
  {multiplexer.pin(7), 0x8, 1},
  {multiplexer.pin(8), 0x9, 1},
  {multiplexer.pin(9), 0xA, 1},
  {multiplexer.pin(10), 0xB, 1},
  {multiplexer.pin(11), 0xC, 1},
  {multiplexer.pin(12), 0xD, 1},
  {multiplexer.pin(13), 0xE, 1},
  {multiplexer.pin(14), 0xF, 1},
  {multiplexer.pin(15), 0x10, 1},
  {multiplexer.pin(16), 0x11, 1},
  {multiplexer.pin(17), 0x12, 1},
  {multiplexer.pin(18), 0x13, 1},
  {multiplexer.pin(19), 0x14, 1},
//  {multiplexer.pin(20), 0x15, 1},
  {multiplexer.pin(20), 0x16, 1},
  {multiplexer.pin(21), 0x17, 1},
  {multiplexer.pin(22), 0x18, 1},
  {multiplexer.pin(23), 0x19, 1},
  {multiplexer.pin(24), 0x1A, 1},
  {multiplexer.pin(25), 0x1B, 1},
  {multiplexer.pin(26), 0x1C, 1},
  {multiplexer.pin(27), 0x1D, 1},
  {multiplexer.pin(28), 0x1E, 1},
  {multiplexer.pin(29), 0x1F, 1},
  {multiplexer.pin(30), 0x20, 1},
  {multiplexer.pin(31), 0x21, 1},
  {multiplexer.pin(32), 0x22, 1},
  {multiplexer.pin(33), 0x23, 1},
  {multiplexer.pin(34), 0x24, 1},
  {multiplexer.pin(35), 0x25, 1},
  {multiplexer.pin(36), 0x26, 1},
  {multiplexer.pin(37), 0x27, 1},
  {multiplexer.pin(38), 0x28, 1},
  {multiplexer.pin(39), 0x29, 1},
  {multiplexer.pin(40), 0x2A, 1},
  {multiplexer.pin(41), 0x2B, 1},
  {multiplexer.pin(42), 0x2C, 1},
  {multiplexer.pin(43), 0x2D, 1},
  {multiplexer.pin(44), 0x2E, 1},
  {multiplexer.pin(45), 0x2F, 1},
  {multiplexer.pin(46), 0x30, 1},
  {multiplexer.pin(47), 0x31, 1},
  {multiplexer.pin(48), 0x32, 1},
  {multiplexer.pin(49), 0x33, 1},
  {multiplexer.pin(50), 0x34, 1},
  {multiplexer.pin(51), 0x35, 1},
  {multiplexer.pin(52), 0x36, 1},
  {multiplexer.pin(53), 0x37, 1},
  {multiplexer.pin(54), 0x38, 1},
  {multiplexer.pin(55), 0x39, 1},
  {multiplexer.pin(56), 0x3A, 1},
  {multiplexer.pin(57), 0x3B, 1},
  {multiplexer.pin(58), 0x3C, 1},
  {multiplexer.pin(59), 0x3D, 1},
  {multiplexer.pin(60), 0x3E, 1},
  {multiplexer.pin(61), 0x3F, 1},
  {multiplexer.pin(62), 0x40, 1}, 
  {multiplexer.pin(63), 0x41, 1},
  {multiplexer.pin(64), 0x42, 1},
  {multiplexer.pin(65), 0x43, 1},
  {multiplexer.pin(66), 0x44, 1},
  {multiplexer.pin(67), 0x45, 1},
  {multiplexer.pin(68), 0x46, 1},
  {multiplexer.pin(69), 0x47, 1},
  {multiplexer.pin(70), 0x48, 1},
  {multiplexer.pin(71), 0x49, 1},
  {multiplexer.pin(72), 0x4A, 1},
  {multiplexer.pin(73), 0x4B, 1},
  {multiplexer.pin(74), 0x4C, 1},
  {multiplexer.pin(75), 0x4D, 1},
  {multiplexer.pin(76), 0x4E, 1},
  {multiplexer.pin(77), 0x4F, 1},
  {multiplexer.pin(78), 0x50, 1},
  {multiplexer.pin(79), 0x51, 1},
  {multiplexer.pin(80), 0x52, 1},
  {multiplexer.pin(81), 0x53, 1},
  {multiplexer.pin(82), 0x54, 1},
  {multiplexer.pin(83), 0x55, 1},
  {multiplexer.pin(84), 0x56, 1},
  {multiplexer.pin(85), 0x57, 1},
  {multiplexer.pin(86), 0x58, 1},
  {multiplexer.pin(87), 0x59, 1},
  {multiplexer.pin(88), 0x5A, 1},
  {multiplexer.pin(89), 0x5B, 1},
  {multiplexer.pin(90), 0x5C, 1},
  {multiplexer.pin(91), 0x5D, 1},
  {multiplexer.pin(92), 0x5E, 1},
  {multiplexer.pin(93), 0x5F, 1},
  {multiplexer.pin(94), 0x60, 1},
  {multiplexer.pin(95), 0x61, 1},
  {multiplexer.pin(96), 0x62, 1},
  {multiplexer.pin(97), 0x63, 1},
  {multiplexer.pin(98), 0x64, 1},
  {multiplexer.pin(99), 0x65, 1},
  {multiplexer.pin(100), 0x66, 1},
  {multiplexer.pin(101), 0x67, 1},
  {multiplexer.pin(102), 0x68, 1},
  {multiplexer.pin(103), 0x69, 1},
  {multiplexer.pin(104), 0x6A, 1},
  {multiplexer.pin(105), 0x6B, 1},
  {multiplexer.pin(106), 0x6C, 1},
  {multiplexer.pin(107), 0x6D, 1},
  {multiplexer.pin(108), 0x6E, 1},
  {multiplexer.pin(109), 0x6F, 1},
  {multiplexer.pin(110), 0x70, 1},
  {multiplexer.pin(111), 0x71, 1},
  {multiplexer.pin(112), 0x72, 1},
  {multiplexer.pin(113), 0x73, 1},
  {multiplexer.pin(114), 0x74, 1},
  {multiplexer.pin(115), 0x75, 1},
  {multiplexer.pin(116), 0x76, 1},
  {multiplexer.pin(117), 0x77, 1},
  {multiplexer.pin(118), 0x78, 1},
  {multiplexer.pin(119), 0x79, 1},
  {multiplexer.pin(120), 0x7A, 1},
  {multiplexer.pin(121), 0x7B, 1},
  {multiplexer.pin(122), 0x7C, 1},
  {multiplexer.pin(123), 0x7D, 1},
  {multiplexer.pin(124), 0x7E, 1},
  {multiplexer.pin(125), 0x7F, 1},
  {multiplexer.pin(126), 0x1, 2},
  {multiplexer.pin(127), 0x2, 2},
  {multiplexer.pin(128), 0x3, 2},
  {multiplexer.pin(129), 0x4, 2},
  {multiplexer.pin(130), 0x5, 2},
  {multiplexer.pin(131), 0x6, 2},
  {multiplexer.pin(132), 0x7, 2},
  {multiplexer.pin(133), 0x8, 2},
  {multiplexer.pin(134), 0x9, 2},
  {multiplexer.pin(135), 0xA, 2},
  {multiplexer.pin(136), 0xB, 2},
  {multiplexer.pin(137), 0xC, 2},
  {multiplexer.pin(138), 0xD, 2},
  {multiplexer.pin(139), 0xE, 2},
  {multiplexer.pin(140), 0xF, 2},
  {multiplexer.pin(141), 0x10, 2},
  {multiplexer.pin(142), 0x11, 2},
  {multiplexer.pin(143), 0x12, 2},
  {multiplexer.pin(144), 0x13, 2},
  {multiplexer.pin(145), 0x14, 2},
  {multiplexer.pin(146), 0x15, 2},
  {multiplexer.pin(147), 0x16, 2},
  {multiplexer.pin(148), 0x17, 2},
  {multiplexer.pin(149), 0x18, 2},
  {multiplexer.pin(150), 0x19, 2},
};
//FADERS
AnalogMultiplex multiplexerF(A0, { 9, 10, 11, 12,} );
Analog potentiometersF[] = {
  {multiplexerF.pin(0), 0x1, 3},
  {multiplexerF.pin(1), 0x2, 3},
  {multiplexerF.pin(3), 0x4, 3},
  {multiplexerF.pin(4), 0x5, 3},
  {multiplexerF.pin(5), 0x6, 3},
  {multiplexerF.pin(6), 0x7, 3},
  {multiplexerF.pin(7), 0x8, 3},
  {multiplexerF.pin(8), 0x9, 3},
  {multiplexerF.pin(9), 0xA, 3},
  {multiplexerF.pin(10), 0xB, 3},
  {multiplexerF.pin(11), 0xC, 3},
  {multiplexerF.pin(12), 0xD, 3},
  {multiplexerF.pin(13), 0xE, 3},
  {multiplexerF.pin(14), 0xF, 3},
  {multiplexerF.pin(15), 0x10, 3},
};

  
//ENCODERS
RotaryEncoder enc0 = {1, 0, 0x1, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
RotaryEncoder enc1 = {2, 1, 0x2, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
RotaryEncoder enc2 = {3, 2, 0x3, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
RotaryEncoder enc3 = {4, 3, 0x4, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
RotaryEncoder enc4 = {5, 4, 0x5, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
RotaryEncoder enc5 = {6, 5, 0x6, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
RotaryEncoder enc6 = {7, 6, 0x7, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
RotaryEncoder enc7 = {0, 7, 0x8, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};

void setup() {}
void loop() {MIDI_Controller.refresh();}

I got this message when I tried to Verify:
,,Sketch uses 13276 bytes (46%) of program storage space. Maximum is 28672 bytes.
Global variables use 5392 bytes (210%) of dynamic memory, leaving -2832 bytes for local variables. Maximum is 2560 bytes.
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
Error compiling for board Arduino Leonardo."

What could I do?   Do I have to buy a Teensy board? For example Teensy 3.2? Which would be the best?

tttapa, 2018-04-08:
An advantage of the MIDI Controller library is that it's easy to use and very flexible. This comes at a cost, of course: it uses more memory.

You can either write your own code from scratch, or you could use a board that has more RAM, like a Teensy.

userdsp, 2018-04-08:
I love teensy 3.6 <3

Archiving built core (caching) in: C:\Users\*\AppData\Local\Temp\arduino_cache_336804\core\core_teensy_avr_teensy36_usb_serialmidi,speed_180,opt_o2std,keys_en-us_4939f5412a75a8560acb0f9e98268f02.a
Opening Teensy Loader...

Sketch uses 34528 bytes (3%) of program storage space. Maximum is 1048576 bytes.
Global variables use 15764 bytes (6%) of dynamic memory, leaving 246380 bytes for local variables. Maximum is 262144 bytes.

tttapa, 2018-04-09:
Note that your code has several errors:
- Your multiplexers have only 4 address pins, so use multiple `AnalogMultiplex` objects .
- You can't use the same pins for rotary encoders.

```cpp
#include <MIDI_Controller.h>

const int speedMultiply = 1;

//BUTTONS
AnalogMultiplex multiplexers[] = {
  {A0, {10, 11, 12, 13}},
  {A1, {10, 11, 12, 13}},
  {A2, {10, 11, 12, 13}},
  // ...
};

Digital buttons[] = {
  { multiplexers[0].pin(0), 0x1, 1 },
  { multiplexers[0].pin(1), 0x2, 1 },
  { multiplexers[0].pin(2), 0x3, 1 },
  // ...
  { multiplexers[1].pin(0), 0x11, 1 },
  { multiplexers[1].pin(1), 0x12, 1 },
  // ...
  { multiplexers[2].pin(0), 0x21, 1 },
  // ...
};
//FADERS
AnalogMultiplex multiplexerF(A3, {10, 11, 12, 13} );
Analog potentiometersF[] = {
  { multiplexerF.pin(0), 0x1, 3 },
  { multiplexerF.pin(1), 0x2, 3 },
  { multiplexerF.pin(3), 0x4, 3 },
  { multiplexerF.pin(4), 0x5, 3 },
  { multiplexerF.pin(5), 0x6, 3 },
  { multiplexerF.pin(6), 0x7, 3 },
  { multiplexerF.pin(7), 0x8, 3 },
  { multiplexerF.pin(8), 0x9, 3 },
  { multiplexerF.pin(9), 0xA, 3 },
  { multiplexerF.pin(10), 0xB, 3 },
  { multiplexerF.pin(11), 0xC, 3 },
  { multiplexerF.pin(12), 0xD, 3 },
  { multiplexerF.pin(13), 0xE, 3 },
  { multiplexerF.pin(14), 0xF, 3 },
  { multiplexerF.pin(15), 0x10, 3 },
};

//ENCODERS
RotaryEncoder enc0 = {0, 1, 0x1, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
RotaryEncoder enc1 = {2, 3, 0x2, 4, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT};
// ...

void setup() {}
void loop() {
  MIDI_Controller.refresh();
}
```