tttapa/MIDI_controller #32: Hairless Midi isn't able to read data

mr-dgidgi, 2018-04-11:
## Description of the problem or question
When I use Hairless MIDI Serial to test my circuit it is not able to read midi data and I have strange messages like bellow : 
```

 +233.822 - Serial In: System Message #14
 +233.826 - Serial In: System Message #14
 +233.899 - Warning: got a status byte when we were expecting 3 more data bytes, sending possibly incomplete MIDI message 0xff
 +233.899 - Serial Says: 
 +233.899 - Warning: got a status byte when we were expecting 3 more data bytes, sending possibly incomplete MIDI message 0xff
```

This only append with Hairless. When I disable `HairlessMIDI_Interface` and activate `USBDebugMIDI_Interface` I'm able to see correct midi data.

I'm doing it to repare a Gemini G4v controller with  an arduino, my code is huge but I also tried with only one potentiometer and the issue is the same.

## Steps to reproduce the problem
Connect few potentiometers to a mux use HairlessMIDI_Interface. Hairless midi serial is set to 9600baud
### Hardware

**Arduino board**:  ? Arduino MEGA 2560  

### Software versions:

**MIDI Controller library**: ? 3.0.1  
**Arduino IDE**: ? 1.8.5
**Operating System**: ? Windows  
**Operating System version**: ? 10 

### Full code 

```cpp
#include <MIDI_Controller.h>

/*********************************************************

Arduino midi controller
G4v hack

*********************************************************/

//serial debug
//USBDebugMIDI_Interface midiInterface(9600);
//We instantiate an hairlessmidi interface to send midi over serialUSB 
HairlessMIDI_Interface HairlessMidi;

//AnalogMultiplex NAME(pin_t analogPin, { pin_t addressPin1, addressPin2, ... addressPinN } )
AnalogMultiplex MuxD1(A0, {2,3,4,5});
AnalogMultiplex MuxD2(A1, {2,3,4,5});
AnalogMultiplex MuxE1(A2, {2,3,4,5});
AnalogMultiplex MuxE2(A3, {2,3,4,5});
AnalogMultiplex MuxF1(A4, {2,3,4,5});
AnalogMultiplex MuxF2(A5, {2,3,4,5});

//we create an array of multiplexer to convert button value later.
AnalogMultiplex MidiMux[] = {
  MuxD1,
  MuxD2,
  MuxE1,
  MuxE2,
  MuxF1,
  MuxF2
};

/*
 * We define only MUX D, E & F but we define them like a "virtual mux" in order to contain the secondary mux (A,B,C,G,H)
pin 2 & 3 used for the main mux (D, E, F)
pin 4 & 5 used by the secondary mux (A, B, C, G, H)
*/

//create an array for each Mux containing potentiometer
//channel 3 will be used to read button values
Analog D1pot[] = {
  {MuxD1.pin(0), 1, 3},
  {MuxD1.pin(1), 3, 3},
  {MuxD1.pin(2), 4, 3},
  {MuxD1.pin(3), 8, 3},
  {MuxD1.pin(4), 1, 3},
  {MuxD1.pin(5), 3, 3},
  {MuxD1.pin(6), 5, 3},
  {MuxD1.pin(7), 23, 4},
  {MuxD1.pin(8), 1, 3},
  {MuxD1.pin(9), 3, 3},
  {MuxD1.pin(10), 4, 1},
  {MuxD1.pin(11), 24, 4},
  {MuxD1.pin(12), 1, 3},
  {MuxD1.pin(3), 3, 3},
  {MuxD1.pin(14), 5, 1},
  {MuxD1.pin(15), 2, 4}
};

Analog D2pot[] = {
  {MuxD2.pin(0), 2, 3},
  {MuxD2.pin(1), 25, 4},
  {MuxD2.pin(2), 6, 3},
  {MuxD2.pin(3), 9, 3},
  {MuxD2.pin(4), 2, 3},
  {MuxD2.pin(5), 25, 4},
  {MuxD2.pin(6), 7, 3},
  {MuxD2.pin(7), 21, 4},
  {MuxD2.pin(8), 2, 3},
  {MuxD2.pin(9), 25, 4},
  {MuxD2.pin(10), 4, 2},
  {MuxD2.pin(11), 22, 4},
  {MuxD2.pin(12), 2, 3},
  {MuxD2.pin(13), 25, 4},
  {MuxD2.pin(14), 5, 2},
  {MuxD2.pin(15), 1, 1}
};

Analog E1pot[] = {
  {MuxE1.pin(0), 10, 3},
  {MuxE1.pin(1), 1, 2},
  {MuxE1.pin(2), 17, 4},
  {MuxE1.pin(3), 28, 4},
  {MuxE1.pin(4), 10, 3},
  {MuxE1.pin(5), 1, 2},
  {MuxE1.pin(6), 18, 4},
  {MuxE1.pin(7), 28, 4},
  {MuxE1.pin(8), 10, 3},
  {MuxE1.pin(9), 1, 2},
  {MuxE1.pin(10), 19, 4},
  {MuxE1.pin(11), 28, 4},
  {MuxE1.pin(12), 10, 3},
  {MuxE1.pin(13), 1, 2}, 
  {MuxE1.pin(14), 20, 4},
  {MuxE1.pin(15), 28, 4}
};

Analog E2pot[] = {
  {MuxE2.pin(0), 11, 3},
  {MuxE2.pin(1), 1, 1},
  {MuxE2.pin(2), 13, 4},
  {MuxE2.pin(3), 27, 4},
  {MuxE2.pin(4), 11, 3},
  {MuxE2.pin(5), 1, 1},
  {MuxE2.pin(6), 14, 4},
  {MuxE2.pin(7), 27, 4},
  {MuxE2.pin(8), 11, 3},
  {MuxE2.pin(9), 1, 1},
  {MuxE2.pin(10), 15, 4},
  {MuxE2.pin(11), 27, 4},
  {MuxE2.pin(12), 11, 3},
  {MuxE2.pin(13), 1, 1}, 
  {MuxE2.pin(14), 16, 4},
  {MuxE2.pin(15), 27, 4}
};

Analog F1pot[] = {
  {MuxF1.pin(0), 12, 3},
  {MuxF1.pin(1), 2, 2},
  {MuxF1.pin(2), 26, 4},
  {MuxF1.pin(3), 5, 4},
  {MuxF1.pin(4), 12, 3},
  {MuxF1.pin(5), 2, 2},
  {MuxF1.pin(6), 3, 2},
  {MuxF1.pin(7), 6, 4},
  {MuxF1.pin(8), 12, 3},
  {MuxF1.pin(9), 2, 2},
  {MuxF1.pin(11), 7, 4},
  {MuxF1.pin(12), 12, 3},
  {MuxF1.pin(13), 2, 2},
  {MuxF1.pin(14), 4, 4},
  {MuxF1.pin(15), 8, 4}
};

Analog F2pot[] = {
  {MuxF2.pin(0), 13, 3},
  {MuxF2.pin(1), 2, 1},
  {MuxF2.pin(2), 14, 3},
  {MuxF2.pin(3), 9, 4},
  {MuxF2.pin(4), 13, 3},
  {MuxF2.pin(5), 2, 1},
  {MuxF2.pin(6), 3, 1},
  {MuxF2.pin(7), 10, 4},
  {MuxF2.pin(8), 13, 3},
  {MuxF2.pin(9), 2, 1},
  {MuxF2.pin(11), 11, 4},
  {MuxF2.pin(12), 13, 3},
  {MuxF2.pin(13), 2, 1},
  {MuxF2.pin(14), 3, 4},
  {MuxF2.pin(15), 12, 4}
};

void setup() {
  // put your setup code here, to run once:

}

void loop() {

  MIDI_Controller.refresh();

  
}

```

## Steps taken to try to diagnose or solve the problem
Work properly with midi debug and serial terminal.

## The goal of your project and aditional information
I'm fixing a Gemini G4v which has been connected to 12v instead of 5v. Microcontrollers are dead but button, potentiometer, mux, shift register and led are still alive. The goal is to be able to read the data from every pot/buttons of this controller and send the correct midi data to the computer.

The mux configuration is strange because some mux data output pin are connected to other mux data input in order to use less analog pin on the original configuration.

tttapa, 2018-04-11:
The default Hairless baud rate is 115200 symbols/second.

(You could change it in the settings, but I recommend just changing it in Hairless, as I think it is the default.)
https://github.com/tttapa/MIDI_controller/blob/90b8796f8365faf28bcfa15eab8f70336d9381d6/src/Settings/Settings.h#L10

tttapa, 2018-04-11:
Note that `HairlessMIDI_Interface` is just a wrapper for `USBSerialMIDI_Interface`:

https://github.com/tttapa/MIDI_controller/blob/90b8796f8365faf28bcfa15eab8f70336d9381d6/src/MIDI_Interfaces/SerialMIDI_Interface.h#L89-L94

So if you want, you can just use `USBSerialMIDI_Interface` with any baud rate you wish, without having to dive into the settings.

mr-dgidgi, 2018-04-11:
Thanks for your quick feedback. I tried 9600 to 38400 baud but as only 9600baud was not crashing Hairless I thought it was the good baud rate. I tried 115200 and it work perfectly. Thx for the information about `USBSerialMIDI_Interface`.

closed 2018-04-11 by mr-dgidgi