tttapa/MIDI_controller #33: convert analog data to button activation

mr-dgidgi, 2018-04-12:
## Description of the problem or question
I'm repairing a midi dj controler with Arduino board. On the circuit board, buttons are connected to analog mux in serial with resistor between each buttons.  On the same mux there is also some potentiometer. I use channel 3 for each mux pin connected to buttons and one cc per mux pin.

I'm looking for a way to read analog mux pin value and convert it to button depending of the value or to do the same thing from each cc in channel 3.

Fox example, I'v got a mux with pin 1 connected to buttons. I set it to CC 1 Channel 3. When I press the first button the value pass from 127 to 2, when I release it the value go back to 127. If I press the second button the value of CC 1 pass from 127 to 12, when I release it the value go back to 127. 

I want to be able to convert cc value to button activation like : 

```
If CC1 Chan3 < 10 send note C1 pressure 127
else if CC1 Chan3 >10 & <20 send note D1 pressure 127
else if CC1 Chan3 >20 & <30 send note E1 pressure 127
etc...
```
I don't know how I can read the CC value or mux pin value on my code in order to convert it to button.

### Hardware

**Arduino board**:  ? Arduino MEGA 2560

**Schematic**: ?
[schema](http://www.casimages.com/i/180412025239110746.jpg.html)

### Software versions:

**MIDI Controller library**: ? 3.0.1 
**Arduino IDE**: ? 1.8.5
**Operating System**: ? Windows  
**Operating System version**: ? 10   

## The goal of your project and aditional information

I'm fixing a Gemini G4v which has been connected to 12v instead of 5v. Microcontrollers are dead but button, potentiometer, mux, shift register and led are still alive. The goal is to be able to read the data from every pot/buttons of this controller and send the correct midi data to the computer.

The mux configuration is strange because some mux data output pin are connected to other mux data input in order to use less analog pin on the original configuration. Analog mux are also used to connect multiples buttons

tttapa, 2018-04-29:
This is currently not supported. You could easily add it to the library, though, by inheriting from the MIDI_Control_Element class.

cferrarini, 2018-07-22:
Hello. I had the same setup of an array of buttons with resistors on the same analog port, but it proved too complicated to debounce with code (read the state without interference). What I did is to use the digital pins instead, the present code of this library debounces perfect digital pins, with a switch to ground. You can convert/map analog ports also to digital in arduino pro micro, instead of a0 a1 a2... map to digital 19 20 21... You can always use a multiplexing chip id there's not enough pins for your buttons. I found out that you can create many banks and bank selectors if you have different number of buttons or knobs per bank.