tttapa/MIDI_controller #35: Input with VCC not GND, how to do it?

brian12112, 2018-04-29:
There is any solution for triggering the pin with vcc (Debounce) and not with gnd? Thank you Mr. tttapa

tttapa, 2018-04-29:
Normally, you don't need this, usually, buttons are wired between ground and an input.

If you really need it, you can use `Digital::invert()`.

brian12112, 2018-04-29:
Wow, thank you mr tttapa, i will try it.

brian12112, 2018-04-29:
Can you give a example where to put it ?

I try `
Digital::invert button(2, C4, 1, velocity);
`
But I cant compile it. Thanks

tttapa, 2018-04-29:
No, `Digital::invert()` just means that invert is a method of the Digital class. 

```cpp
Digital button(2, C4, 1, velocity);

void setup() {
  button.invert();
}
```

brian12112, 2018-04-29:
Awesome!. It works Great. Thanks again Mr. tttapa.

 Just last questions. Can I do the same for Multiplexer?
`Digital buttons[] = {
  {multiplexer.pin(0), 0x10, 1, velocity},
  {multiplexer.pin(1), 0x11, 1, velocity},
  {multiplexer.pin(2), 0x12, 1, velocity},
  {multiplexer.pin(3), 0x13, 1, velocity},
  {multiplexer.pin(4), 0x14, 1, velocity},
  {multiplexer.pin(5), 0x15, 1, velocity},
  {multiplexer.pin(6), 0x16, 1, velocity},
  {multiplexer.pin(7), 0x17, 1, velocity},`
 or I must use another IC? like shift register or else?

brian12112, 2018-04-29:
Awesome!. It works Great. Thanks again Mr. tttapa.

Just last questions. Can I do the same for Multiplexer?
`
Digital buttons[] = { 
{multiplexer.pin(0), 0x10, 1, velocity}, 
{multiplexer.pin(1), 0x11, 1, velocity}, 
{multiplexer.pin(2), 0x12, 1, velocity}, 
{multiplexer.pin(3), 0x13, 1, velocity}, 
{multiplexer.pin(4), 0x14, 1, velocity}, 
{multiplexer.pin(5), 0x15, 1, velocity}, 
{multiplexer.pin(6), 0x16, 1, velocity}, 
{multiplexer.pin(7), 0x17, 1, velocity},
`
or I must use another IC? like shift register or else?

tttapa, 2018-04-29:
Yes. But you might have to change the pin mode (to disable the internal pull-up resistor). 
`ExtIO::pinMode(multiplexer.pin(0));` (You only have to do it for one pin, it will change it for the entire multiplexer.)

brian12112, 2018-05-01:
Yes,  Mr. tttapa. Its working great! Thank you for your Awesome Library.

closed 2018-05-01 by brian12112