tttapa/MIDI_controller #39: No Midi Messages in Mixx or elsewhere, unable to re-establish connection with Leonardo.

rabbiccu, 2018-06-06:
## Description of the problem or question
Arduino shows up as a MIDI Contoller, but no MIDI messages being passed into Mixx program during learning. Configured the faders and pots with numerous wiring pinout and still nothing.

### Hardware
Arduino Leonardo. B10k pots, B103 faders, breadboard. miniUSB-->Computer

### Software versions:

**MIDI Controller library**:  3.0.1  
**Arduino IDE**: 1.6.8  
**Operating System**: Windows 10  
**Operating System version**: ? 10, 16.04, ... ?  
(**Teensyduino**):  latest
(**Encoder library**):  latest  
(**MIDIUSB library**): latest 
(THESE LAST THREE ARE SET TO LATEST BECAUSE ANOTHER ISSUE. AFTER UPLOADING THE FIRST TIME I HAVE BEEN UNABLE TO UPLOAD AGAIN AS THE PORT MENU IN ARDUINO IDE IS GREYED OUT. I GET A ERROR MESSAGE ABOUT THE PORT NOT BEING THERE )
WHEN I TRY TO REUPLOAD TO ENSURE LIBRARY VERSIONS ETC I GET:
```
`Arduino: 1.6.8 (Windows 10), Board: "TeeOnArdu, MIDI, 16 MHz, US English"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\admin\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\admin\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\admin\Documents\Arduino\libraries" -fqbn=TeeOnArdu:avr:TeeOnArdu:usb=midi,speed=16,keys=en-us -ide-version=10608 -build-path "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\admin\Desktop\MIDI_Controller_Leonardo_-_Analog\MIDI_Controller_Leonardo_-_Analog.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\admin\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\admin\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\admin\Documents\Arduino\libraries" -fqbn=TeeOnArdu:avr:TeeOnArdu:usb=midi,speed=16,keys=en-us -ide-version=10608 -build-path "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\admin\Desktop\MIDI_Controller_Leonardo_-_Analog\MIDI_Controller_Leonardo_-_Analog.ino"
WARNING: Spurious .github folder in 'MIDI controller' library
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Arduino"' '-DUSB_PRODUCT="TeeOnArdu"' -DUSB_MIDI -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\cores\TeeOnArdu" "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\variants\leonardo" "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\sketch\MIDI_Controller_Leonardo_-_Analog.ino.cpp" -o "nul"
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Arduino"' '-DUSB_PRODUCT="TeeOnArdu"' -DUSB_MIDI -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\cores\TeeOnArdu" "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\variants\leonardo" "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\sketch\MIDI_Controller_Leonardo_-_Analog.ino.cpp" -o "nul"
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Arduino"' '-DUSB_PRODUCT="TeeOnArdu"' -DUSB_MIDI -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\cores\TeeOnArdu" "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\variants\leonardo" "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\sketch\MIDI_Controller_Leonardo_-_Analog.ino.cpp" -o "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Arduino"' '-DUSB_PRODUCT="TeeOnArdu"' -DUSB_MIDI -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\cores\TeeOnArdu" "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\variants\leonardo" "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\sketch\MIDI_Controller_Leonardo_-_Analog.ino.cpp" -o "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\sketch\MIDI_Controller_Leonardo_-_Analog.ino.cpp.o"
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Arduino"' '-DUSB_PRODUCT="TeeOnArdu"' -DUSB_MIDI -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\cores\TeeOnArdu" "-IC:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\variants\leonardo" "C:\Program Files (x86)\Arduino\hardware\TeeOnArdu\avr\cores\TeeOnArdu\wiring_pulse.S" -o "C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\wiring_pulse.S.o"
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\WInterrupts.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\hooks.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\keylayouts.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\teensyduino.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\usb.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\wiring.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\wiring_analog.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\wiring_digital.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\wiring_pulse.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\wiring_shift.c.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\HardwareSerial.cpp.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\HardwareSerial0.cpp.o
Using previously compiled file: C:\Users\admin\AppData\Local\Temp\build05e956ed821e7a998eb94654329edf25.tmp\core\HardwareSerial1.cpp.o
[... cut ...]
I have been trying to re-establish connection with the Arduino so I can upload some new code/libraries thinking there is probably an error somewhere in the current code. The Arduino does show up as a MIDI controller and has the ANALOG Example uploaded to it. Being locked out of the Arduino has hindered my ability to troubleshoot.

## The goal of your project and aditional information
I want 9 Faders, 2 Knobs and an external TRS female jack for an expression pedal. I want this to run an organ VST's drawbars, overdrive, and expression pedal.

tttapa, 2018-06-06:
TeeOnArdu is no longer supported. Use the examples from GitHub instead.

To fix your uploading problem, just press the reset button right after hitting upload.  
If you're unable to select the port, just press the reset button right before opening the Tools menu.

rabbiccu, 2018-06-06:
Thank you so much ttapa.

So, I have switched the board to Arduino Leonardo and am attempting to use the example code from GitHub here.  I wanted to add a couple pots for testing.

##CODE
```
/*
This is an example of the "Analog" class of the MIDI_controller library.
Connect a potentiometer to analog pin A0. This will be the MIDI channel volume of channel 1.
Map it in your DAW or DJ software.
Written by Pieter P, 08-09-2017
https://github.com/tttapa/MIDI_controller
*/

#include <MIDI_Controller.h> // Include the library

// Create a new instance of the class 'Analog', called 'potentiometer', on pin A0, 
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog potentiometer_A(A0, MIDI_CC::Channel_Volume, 1);
Analog potentiometer_B(A1, MIDI_CC::Channel_Volume, 2);
Analog potentiometer_C(A2, MIDI_CC::Channel_Volume, 3);
Analog potentiometer_D(A3, MIDI_CC::Channel_Volume, 4);
Analog potentiometer_E(A4, MIDI_CC::Channel_Volume, 5);

void setup() {}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}
```

tttapa, 2018-06-06:
That's really strange. There's a compilation error in MIDIUSB.cpp (which is not part of the MIDI Controller library). I've never seen that one before.

tttapa, 2018-06-06:
I would try installing the latest version of the Arduino IDE (≥1.8.x), after removing the old versions. It seems to be a problem between the MIDIUSB library and your version of the Arduino Core.

rabbiccu, 2018-06-06:
I did so many things trying to connect to the Arduino.  UninstallIng and reinstalling.. I'm going to try to completely remove all traces of arduino and start over.

rabbiccu, 2018-06-06:
Ok everything deleted and reinstalled, it compiles just fine for the Leonardo board. Still not able to upload and the port is greyed out at all times. The moment I release the reset button the Arduino mounts as a MIDI device still. I cant help but think that somehow I have locked myself out of the board by making the USB connection a MIDI connection? Totally lost but the compilation issue is overcome! 

Any thoughts on greyed out port menu? What is "Upload using Programmer"?

### Code
```
/*
This is an example of the "Analog" class of the MIDI_controller library.
Connect a potentiometer to analog pin A0. This will be the MIDI channel volume of channel 1.
Map it in your DAW or DJ software.
Written by Pieter P, 08-09-2017
https://github.com/tttapa/MIDI_controller
*/

#include <MIDI_Controller.h> // Include the library

// Create a new instance of the class 'Analog', called 'potentiometer', on pin A0, 
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog potentiometer_A(A0, MIDI_CC::Channel_Volume, 1);
Analog potentiometer_B(A1, MIDI_CC::Channel_Volume, 2);
Analog potentiometer_C(A2, MIDI_CC::Channel_Volume, 3);
Analog potentiometer_D(A3, MIDI_CC::Channel_Volume, 4);
Analog potentiometer_E(A4, MIDI_CC::Channel_Volume, 5);

void setup() {}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}
```

### Error
`Couldn't find a Board on the selected port. Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.
`

rabbiccu, 2018-06-06:
When verbose output message is turned on for output I get a string of:

```
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
Couldn't find a Board on the selected port. Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.
```

rabbiccu, 2018-06-06:
Ok I got super frustrated and just started jamming the reset button erratically during upload and it caught and uploaded.

My port menu is back and the new clean code is uploaded. Now to MIDI testing...

tttapa, 2018-06-06:
Great, glad to hear!

closed 2018-06-06 by rabbiccu