tttapa/MIDI_controller #43: cd4067be multiplexer and others

fredmorais, 2018-06-09:
Hello,

I would like to ask multiple questions:

-Is cd4067be supported? If so, what should be the reason I am unable to connect it?
-Can I use encoders like this?
`AnalogMultiplex multiplexers[] = {
  {A0, {10, 11, 12, 13}},
};

RotaryEncoder encoder[] = {
  {multiplexers[0].pin(1), multiplexer[0].pin(2), Controller, Channel, speedMultiply, JOG, TWOS_COMPLEMENT},
};
`
-Can I connect buttons to a AnalogMultiplex ? If so, should it be connected to an analog pin as well?

Thank you so much for your time!!! You are amazing

tttapa, 2018-06-10:
> Is cd4067be supported? If so, what should be the reason I am unable to connect it?  

Yes, it is supported. I don't know, please post your full and exact code and a schematic.

>  I have potentiometers connected with one of those mux and as soon as i turn the pot on it goes to 100%

That's very strange. Please post your full and exact code and a schematic.

> Can I use encoders like this?

No, encoders require interrupt-capable pins. External pins (e.g. multiplexed pins) are not supported.

> Can I connect buttons to a AnalogMultiplex ? If so, should it be connected to an analog pin as well?

Yes, you can, but in that case, it would be more efficient to use a button matrix. If you only connect buttons to that multiplexer, you can use a digital pin.

> Is there any way to convert a log pot into a linear pot? I could really use this functionality!

Yes, you can use the map function to apply the reverse of the log taper before it is further processed by the library.
[Documentation](https://github.com/tttapa/MIDI_controller/wiki/Library-documentation#8-set-analog-map-functions-and-invert-buttons-optional)
Keep in mind that a logarithmic pot doesn't have a true logarithmic taper, it's just a piecewise linear taper:
[image]

fredmorais, 2018-06-10:
Hi, 

First of all thank you for your time!

I am using the following code and schematics, but just to let you know I am using a breadboard and the multiplexer on the image is not a cd4067be because there wasn't one on Fritzing library:

[image]

`#include <MIDI_Controller.h> // Include the library

AnalogMultiplex multiplexer(A7, { 40, 41, 42, 43 } );

Analog potentiometers[] = {
  {multiplexer.pin(7), MIDI_CC::Channel_Volume, 1},
};

void setup() {}

void loop() {

  MIDI_Controller.refresh();
}`

I am using a dual gang pot, can I take advantage of that to make it linear? I am sorry but I searched and couldn't figure out how I should make my function to apply the reverse log.

tttapa, 2018-06-10:
Have you tried a normal sketch to get the multiplexer to work? That could narrow it down significantly.

fredmorais, 2018-06-10:
I'm sorry but I don't know much about this. What's a normal sketch?

tttapa, 2018-06-10:
Without the MIDI_Controller library, just setting the address lines to '7' and then just printing out the analog reading to the Serial Monitor (see File > Examples > Basics > AnalogReadSerial).

fredmorais, 2018-06-10:
I just tried it with the schematics I sent before and this code: 
/*
  AnalogReadSerial

  Reads an analog input on pin 0, prints the result to the Serial Monitor.
  Graphical representation is available using Serial Plotter (Tools > Serial Plotter menu).
  Attach the center pin of a potentiometer to pin A0, and the outside pins to +5V and ground.

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/AnalogReadSerial
*/

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A7);
  // print out the value you read:
  Serial.println(sensorValue);
  delay(1);        // delay in between reads for stability
}

And changing the potentiometer does nothing and the values go crazy between ~980 and ~1010

tttapa, 2018-06-10:
Mathematically, the inverse of the blue curve would be:
```cpp
value = (value < y_knee) ? (value * x_knee / y_knee) : ((value - y_knee) * (1023 - x_knee) / (1023 - y_knee) + x_knee); 
```
(Where `x_knee` is the x-coordinate of the knee point, and `y_knee` the y-coordinate, e.g. x_knee = 512 and y_knee = 102.3 in the picture, but it depends on your potentiometer, so you'll have to measure it.)

But you have to keep in mind that `/` is an integer division in C++, so you'll get rounding errors.

tttapa, 2018-06-10:
You have to set the correct value on the address lines. I recommend you check out some tutorials on how to use a multiplexer with Arduino.

fredmorais, 2018-06-10:
I'll try that to make the pot linear. About the multiplexer I think I'll just go and buy the 74hc4067. It is supposed to work with the schematics and code I made, isn't it?

Thank you for your time. How can I make a donation for support?

tttapa, 2018-06-10:
I don't think the chip is the problem. The CD4067B(e) should work just as well as the 74hc4067. They're essentially the same chip.

I'd try getting it to work without the MIDI Controller library first, to make sure that the wiring is 100% correct. Try swapping the Address lines, for example.

Try this code to check your wiring:
```cpp
const uint8_t analogPin = A7;

const uint8_t addressPin1 = 40;
const uint8_t addressPin2 = 41;
const uint8_t addressPin3 = 42;
const uint8_t addressPin4 = 43;

void setup() {
  Serial.begin(9600);
  
  pinMode(addressPin1, OUTPUT);
  pinMode(addressPin2, OUTPUT);
  pinMode(addressPin3, OUTPUT);
  pinMode(addressPin4, OUTPUT);

  // B0111 == 7
  digitalWrite(addressPin1, 0);
  digitalWrite(addressPin2, 1);
  digitalWrite(addressPin3, 1);
  digitalWrite(addressPin4, 1);
}

void loop() {
  int sensorValue = analogRead(analogPin);
  Serial.println(sensorValue);
  delay(1);
}
```
I don't have anything for donations yet, but I should definitely look into it, you're not the first to ask :)

fredmorais, 2018-06-10:
Hey,

I did that and the values on Serial Monitor are around 250-270 sometimes getting values like 300 but without any pot connected. What should I be getting? Is this ok?

tttapa, 2018-06-10:
That doesn't tell you very much, you have to connect a pot to input 7. The numbers in the serial monitor should vary from 0 to 1023 as you move the potentiometer from 0% to 100%.

fredmorais, 2018-06-10:
It was input 14. Actually was very lucky to remember to check that ahaha. It is working perfectly now, thank you so much. Even tried a button one and it works no problem. Just one detail: I have this working on a lighting program (GrandMA2 if you are curious). Some buttons have the function to be active while I press the button only, but it doesn't work like that. Any reason why this doesn't work? It's just a small detail but if you had something in mind...

Go work on the donation part, you have really been amazing!!

tttapa, 2018-06-10:
So swapping the address lines should have worked then? (7 = B0111, and 14 = B1110)  
Great to hear! Do you know think that I should swap the pins in the library?

Normal buttons (i.e. `Digital`) send a _Note On_ event when pressed down, and a _Note Off_ event when they're released. Latched switches (i.e. `DigitalLatch`) send _Note On + Note Off_ when pressed, and then once again _Note On + Note Off_ when released. You could try the latter.

I'll look at the donation part once my exams are over (early July).

guillecip, 2019-03-14:
Can I use DigitalLatch in a Multiplexer? Can I use half of them with Digital and the other half with DigitalLatch?
Fredmorais, did you tried to modificate the code in order to put a Rotary encoder in a multiplexer?

guillecip, 2019-03-14:
and other thing, when you use digital in a multiplexer I saw that the parameters in the button are only 2 and not 3 like they are in the example button. Why?

tttapa, 2019-03-14:
> Can I use DigitalLatch in a Multiplexer? Can I use half of them with Digital and the other half with DigitalLatch?

Yes.

> and other thing, when you use digital in a multiplexer I saw that the parameters in the button are only 2 and not 3 like they are in the example button. Why?

What do you mean? The constructor of the buttons is the same, no matter if you're using a multiplexer or not:
https://github.com/tttapa/MIDI_controller/blob/815c86950e0a7323a2b89e123e59471158b88e04/src/MIDI_Outputs/Digital.h#L12

guillecip, 2019-03-14:
Can you give me an example with both DigitalLatch and Digital in the same multiplexer?

I´m asking you these because DigitalLatch have one more parameter (duration)

DigitalLatch toggleSwitch(pin_t pin, uint8_t note, uint8_t channel, uint8_t velocity, duration);

Xicury, 2019-08-20:
Greetings I have a question friends, I try using the MIDI_Controller library to use the digital multiplexer to connect 16 potentiometers in a 16-output analog multiplexer, I see that the library codes only have 1 analog and 3 digital pin for communication, in my In case my multiplexer has 4 digital for communication with the 16 channels, my question is, could the multriplexer work with the codes of the library just by increasing that pin to the code of the library?

 ----------------------------------------------ORIGINAL CODE-----------------------------------------

#include <MIDI_Controller.h> // Include the library

// Create an instance of 'AnalogMultiplex' with the output pin of the multiplexer connected to
// analog input pin A0 and the address pins connected to pins 2, 3 and 4.
AnalogMultiplex multiplexer(A0, { 2, 3, 4 } );

// Create 8 new instances of the class 'Analog', on the 8 pins of the multiplexer,
// that send MIDI messages with controller 7 (channel volume) on channels 1 - 8
Analog potentiometers[] = {
  {multiplexer.pin(0), MIDI_CC::Channel_Volume, 1},
  {multiplexer.pin(1), MIDI_CC::Channel_Volume, 2},
  {multiplexer.pin(2), MIDI_CC::Channel_Volume, 3},
  {multiplexer.pin(3), MIDI_CC::Channel_Volume, 4},
  {multiplexer.pin(4), MIDI_CC::Channel_Volume, 5},
  {multiplexer.pin(5), MIDI_CC::Channel_Volume, 6},
  {multiplexer.pin(6), MIDI_CC::Channel_Volume, 7},
  {multiplexer.pin(7), MIDI_CC::Channel_Volume, 8}
};

void setup() {}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}

------------------------------------------16 analog CODE---------------------------------------------

#include <MIDI_Controller.h> // Include the library

// Create an instance of 'AnalogMultiplex' with the output pin of the multiplexer connected to
// analog input pin A0 and the address pins connected to pins 2, 3 and 4.
AnalogMultiplex multiplexer(A0, { 2, 3, 4, 5 } );

// Create 8 new instances of the class 'Analog', on the 8 pins of the multiplexer,
// that send MIDI messages with controller 7 (channel volume) on channels 1 - 8
Analog potentiometers[] = {
  {multiplexer.pin(0), MIDI_CC::Channel_Volume, 1},
  {multiplexer.pin(1), MIDI_CC::Channel_Volume, 2},
  {multiplexer.pin(2), MIDI_CC::Channel_Volume, 3},
  {multiplexer.pin(3), MIDI_CC::Channel_Volume, 4},
  {multiplexer.pin(4), MIDI_CC::Channel_Volume, 5},
  {multiplexer.pin(5), MIDI_CC::Channel_Volume, 6},
  {multiplexer.pin(6), MIDI_CC::Channel_Volume, 7},
  {multiplexer.pin(7), MIDI_CC::Channel_Volume, 8}
  {multiplexer.pin(8), MIDI_CC::Channel_Volume, 9},
  {multiplexer.pin(9), MIDI_CC::Channel_Volume, 10 },
  {multiplexer.pin(10), MIDI_CC::Channel_Volume, 11},
  {multiplexer.pin(11), MIDI_CC::Channel_Volume, 12},
  {multiplexer.pin(12), MIDI_CC::Channel_Volume, 13},
  {multiplexer.pin(13), MIDI_CC::Channel_Volume, 14},
  {multiplexer.pin(14), MIDI_CC::Channel_Volume, 15},
  {multiplexer.pin(15), MIDI_CC::Channel_Volume, 16}
};

void setup() {}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}