tttapa/MIDI_controller #46: Errors uploading with multiple inputs and logarithm/linear adjustments and the main-branch library

rabbiccu, 2018-06-13:
I was having strange crosstalk and sporadic functionality on A0-A4 so I decided to load the new library from the main branch. I also noticed the pots were functioning logarithmic, so I found your calibration example and implemented that into my sketch. I am using:

MIDI CONTOLLER: main branch
Arduino IDE 1.8.5
Arduino Leonardo

###CODE
```/*
This is an example of the "Analog" class of the MIDI_controller library.
Connect a potentiometer to analog pin A0. This will be the MIDI channel volume of channel 1.
Map it in your DAW or DJ software.
Set 'minimumValue' to the analog value read when the potentiometer is in the minimum position.
Set 'maximumValue' to the analog value read when the potentiometer is in the maximum position.
Written by Pieter P, 20-09-2017
https://github.com/tttapa/MIDI_controller
*/

#include <MIDI_Controller.h> // Include the library

// Create a new instance of the class 'Analog', called 'potentiometer', on pin A0, 
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog potentiometer_A(A0, MIDI_CC::Sound_Controller_1, 1);
Analog potentiometer_B(A1, MIDI_CC::Sound_Controller_2, 1);
Analog potentiometer_C(A2, MIDI_CC::Sound_Controller_3, 1);
Analog potentiometer_D(A3, MIDI_CC::Sound_Controller_4, 1);
Analog potentiometer_E(A4, MIDI_CC::Sound_Controller_5, 1);

const int minimumValue = 3;    // the analog value read when the potentiometer is in the minimum position.
const int maximumValue = 1020; // the analog value read when the potentiometer is in the maximum position.

int mapCalibrated(int value) {
  value = constrain(value, minimumValue, maximumValue); // make sure that the analog value is between the minimum and maximum value
  return map(value, minimumValue, maximumValue, 0, 1023); // map the value from [minimumValue, maximumValue] to [0, 1023]
}

void setup() {
  potentiometer_A.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer_B.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer_C.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer_D.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer_E.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}
```

###ERRORS
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\admin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\admin\Documents\Arduino\libraries -fqbn=arduino:avr:leonardo -ide-version=10805 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_843959 -warnings=none -build-cache C:\Users\admin\AppData\Local\Temp\arduino_cache_265514 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\admin\Documents\Arduino\61318\61318.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\admin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\admin\Documents\Arduino\libraries -fqbn=arduino:avr:leonardo -ide-version=10805 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_843959 -warnings=none -build-cache C:\Users\admin\AppData\Local\Temp\arduino_cache_265514 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\admin\Documents\Arduino\61318\61318.ino
Using board 'leonardo' from platform in folder: C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Using core 'arduino' from platform in folder: C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Detecting libraries used...
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\leonardo" "C:\Users\admin\AppData\Local\Temp\arduino_build_843959\sketch\61318.ino.cpp" -o "nul"
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\leonardo" "-IC:\Users\admin\Documents\Arduino\libraries\MIDI_controller-master\src" "C:\Users\admin\AppData\Local\Temp\arduino_build_843959\sketch\61318.ino.cpp" -o "nul"
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\leonardo" "-IC:\Users\admin\Documents\Arduino\libraries\MIDI_controller-master\src" "-IC:\Users\admin\Documents\Arduino\libraries\MIDIUSB\src" "C:\Users\admin\AppData\Local\Temp\arduino_build_843959\sketch\61318.ino.cpp" -o "nul"
"C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\leonardo" "-IC:\Users\admin\Documents\Arduino\libraries\MIDI_controller-master\src" "-IC:\Users\admin\Documents\Arduino\libraries\MIDIUSB\src" "-IC:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SoftwareSerial\src" "C:\Users\admin\AppData\Local\Temp\arduino_build_843959\sketch\61318.ino.cpp" -o "nul"
[... cut ...]
Caused by: processing.app.SerialException: Error touching serial port 'COM4'.
	at processing.app.Serial.touchForCDCReset(Serial.java:107)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:144)
	... 5 more
Caused by: jssc.SerialPortException: Port name - COM4; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.touchForCDCReset(Serial.java:101)
	... 6 more
```

I am enjoying this, again thanks for input.

tttapa, 2018-06-14:
That's an error during upload, not a compilation error. The code is fine.
Try resetting the board during upload, and make sure you select the correct serial port.

rabbiccu, 2018-06-15:
Done, again it was an issue with my uploading and the reset button needs to be mashed in just the right way.

closed 2018-06-15 by rabbiccu