tttapa/MIDI_controller #47: MIDI signals non-stop

rabbiccu, 2018-06-16:
I have sporadic behavior when I go to tinker daily on this project. My faders (10k fader pots) have stopped working all together, although at one time I was able to get a logarithmic response from them. 

I have ordered a multimeter and will be testing my hardware, but is there any reason for this that could be software related? The following screenshot shows the flow of midi signals, never stopping.

[image]

I am using code mentioned in other issues I've opened:

```

/*
This is an example of the "Analog" class of the MIDI_controller library.
Connect a potentiometer to analog pin A0. This will be the MIDI channel volume of channel 1.
Map it in your DAW or DJ software.
Set 'minimumValue' to the analog value read when the potentiometer is in the minimum position.
Set 'maximumValue' to the analog value read when the potentiometer is in the maximum position.
Written by Pieter P, 20-09-2017
https://github.com/tttapa/MIDI_controller
*/

#include <MIDI_Controller.h> // Include the library

// Create a new instance of the class 'Analog', called 'potentiometer', on pin A0, 
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog potentiometer_A(A0, MIDI_CC::Sound_Controller_1, 1);
Analog potentiometer_B(A1, MIDI_CC::Sound_Controller_2, 1);
Analog potentiometer_C(A2, MIDI_CC::Sound_Controller_3, 1);
Analog potentiometer_D(A3, MIDI_CC::Sound_Controller_4, 1);
Analog potentiometer_E(A4, MIDI_CC::Sound_Controller_5, 1);

void setup() {

}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}
```

tttapa, 2018-06-16:
All inputs are independent objects in software, so that shouldn't be an issue.  
Make sure all wires make good connection, even when wiggling them around. You could try soldering wires to everything, and use male header pins to plug them into the Arduino.  
The outer terminals of the potentiometer go to ground and 5V respectively, the center pin (wiper) goes to the analog input. If you experience noise (values fluctuating up and down 1 or 2 values), you could try the `master` version, which has improved filtering.  

Other than that, I don't know what could cause it. I would suspect a hardware issue, though.

TechJA, 2018-06-16:
Another tip: could you connect just one pot to A3 and after to A4 to check these input ports? Maybe they are damaged

rabbiccu, 2018-06-17:
The breadboard has an interupted power rail which I didn't realize was a thing. First project. Sadly though when I figured this out I was bridging them and the board fried.  I had an external 12V power supply, which I the leonardo is supposed to handle. 

There's no other way this could have fried using just the 5v, gnd, inputs and a bunch of pots and faders, right?

tttapa, 2018-06-17:
You could have shorted out 5V to ground, which would probably destroy the 5V regulator when powering it from a 12V source. 

What exactly fried? The MCU or some other part?

rabbiccu, 2018-06-17:
The ATmega32u4 is incredibly hot now, so I think that burned asking with some other things. 

I'm sure I sorted the 5v to gnd because I wasn't cautious to avoid that and tried a million configurations during troubleshooting. I guess I just thought the gnd could handle it. Oh well 10 bucks. 

So my pots are linear but my faders are logarithmic but there is code for mapping log sliders IIRC.

tttapa, 2018-06-17:
You can use the map function, together with the formula discussed here: https://github.com/tttapa/MIDI_controller/issues/43#issuecomment-396056679.

rabbiccu, 2018-06-17:
So would the 

```
value = (value < y_knee) ? (value * x_knee / y_knee) : ((value - y_knee) * (1023 - x_knee) / (1023 - y_knee) + x_knee); 
```
 come after the mapping function for that input? Will you do a sample code please?

tttapa, 2018-06-17:
```cpp
#include <MIDI_Controller.h> // Include the library

// Create a new instance of the class 'Analog', called 'potentiometer', on pin A0, 
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog potentiometer(A0, MIDI_CC::Channel_Volume, 1);

const int x_knee = 512;
const int y_knee = 102;

int mapCalibrated(int value) {
  value = (value < y_knee) ? (value * x_knee / y_knee) : ((value - y_knee) * (1023 - x_knee) / (1023 - y_knee) + x_knee); 
  return value;
}

void setup() {
  potentiometer.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}
```

rabbiccu, 2018-06-17:
And to add linear would they be added at the end of map calibration aka

```
#include <MIDI_Controller.h> // Include the library

// Create a new instance of the class 'Analog', called 'potentiometer', on pin A0, 
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog potentiometer(A0, MIDI_CC::Channel_Volume, 1);

const int x_knee = 512;
const int y_knee = 102;

int mapCalibrated(int value) {
  value = (value < y_knee) ? (value * x_knee / y_knee) : ((value - y_knee) * (1023 - x_knee) / (1023 - y_knee) + x_knee); 
  return value;
}

Analog potentiometer(A1, MIDI_CC::Channel_Volume, 2);

void setup() {
  potentiometer.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}

```

tttapa, 2018-06-17:
No, for the linear ones, you just don't use `potentiometer.map(mapCalibrated)` at all. The position in the code doesn't matter.

tttapa, 2018-06-17:
For example:
```cpp
#include <MIDI_Controller.h> // Include the library

// Create a new instance of the class 'Analog', called 'potentiometer', on pin A0, 
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog linearPots[] = {
  {A0, MIDI_CC::Channel_Volume, 1},
  {A1, MIDI_CC::Channel_Volume, 2},
  {A2, MIDI_CC::Channel_Volume, 3},
  {A3, MIDI_CC::Channel_Volume, 4},
};

Analog logPots[] = {
  {A4, MIDI_CC::Channel_Volume, 5},
  {A5, MIDI_CC::Channel_Volume, 6},
  {A6, MIDI_CC::Channel_Volume, 7},
  {A7, MIDI_CC::Channel_Volume, 8},
};

const int x_knee = 512;
const int y_knee = 102;

int mapLog(int value) {
  value = (value < y_knee) ? (value * x_knee / y_knee) : ((value - y_knee) * (1023 - x_knee) / (1023 - y_knee) + x_knee); 
  return value;
}

void setup() {
  for (Analog &potentiometer : logPots)
    potentiometer.map(mapLog); // apply the 'mapLog' function on the analog input of this potentiometer
}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}
```

rabbiccu, 2018-06-17:
Wow amazing, and it all makes sense.

closed 2018-06-23 by rabbiccu