tttapa/MIDI_controller #48: Faders jump when I push through 50%

mkothencz, 2018-06-16:
Hello,

I built a big MIDI controller with 16 faders, 150 buttons and 8 rotary encoders.
I used Tennsy 3.5 and 10k faders.

I have a little problem: when I push the faders through 50% they jump to maximum (or minimum depend on push direction) for a very short time.

How could I solve this? With programing or this is hardware problem?

tttapa, 2018-06-16:
If you use the AnalogReadSerial example of the IDE, doe you experience the same problem?

mkothencz, 2018-06-16:
No, I can't find any problem if I monitor with serial monitor.

tttapa, 2018-06-16:
That's very strange. Are you using release 3.0.1 or the latest version from the `master` branch?

mkothencz, 2018-06-16:
Yes, I use the latest version.

tttapa, 2018-06-16:
There could be a problem with implicit type conversions from unsigned to signed integer types, IIRC, they can be different on ARM.

I have exams at the moment, so I don't have time to look into it right now. If you want to debug yourself, I'd start by looking at `src/Helpers/EMA.h`, `src/Helpers/Hysteresis.h` and `src/Helpers/Hysteresis.cpp`.

mkothencz, 2018-06-22:
[image]

This is what I get, when I move the faders up-down.

I don't really know how I could debug.

What about changing refresh/analog reading rate? How could I try this out?

tttapa, 2018-06-22:
Could you post your exact code?

mkothencz, 2018-07-22:
#define USE_ROTARY_ENCODER
#include <MIDI_Controller.h>
const int speedMultiply = 1;

//BUTTONS
AnalogMultiplex multiplexer1(4, {0,1,2,3} );
AnalogMultiplex multiplexer2(5, {0,1,2,3} );
AnalogMultiplex multiplexer3(6, {0,1,2,3} );
AnalogMultiplex multiplexer4(7, {0,1,2,3} );
AnalogMultiplex multiplexer5(8, {0,1,2,3} );
AnalogMultiplex multiplexer6(9, {0,1,2,3} );
AnalogMultiplex multiplexer7(10, {0,1,2,3} );
AnalogMultiplex multiplexer8(11, {0,1,2,3} );
AnalogMultiplex multiplexer9(12, {0,1,2,3} );

Digital button1(53, 0x1, 15, 127);
Digital button2(26, 0x2, 15, 127);
Digital button3(27, 0x3, 15, 127);
Digital button4(28, 0x4, 15, 127);
Digital button5(29, 0x5, 15, 127);
Digital button6(30, 0x6, 15, 127);
Digital button7(31, 0x7, 15, 127);

Digital buttons1[] = {
  {multiplexer1.pin(0), 0x1, 1},
  {multiplexer1.pin(1), 0x2, 1},
  {multiplexer1.pin(2), 0x3, 1},
  {multiplexer1.pin(3), 0x4, 1},
  {multiplexer1.pin(4), 0x5, 1},
  {multiplexer1.pin(5), 0x6, 1},
  {multiplexer1.pin(6), 0x7, 1},
  {multiplexer1.pin(7), 0x8, 1},
  {multiplexer1.pin(8), 0x9, 1},
  {multiplexer1.pin(9), 0xA, 1},
  {multiplexer1.pin(10), 0xB, 1},
  {multiplexer1.pin(11), 0xC, 1},
  {multiplexer1.pin(12), 0xD, 1},
  {multiplexer1.pin(13), 0xE, 1},
  {multiplexer1.pin(14), 0xF, 1},
  {multiplexer1.pin(15), 0x10, 1},
};
Digital buttons2[] = {
  {multiplexer2.pin(0), 0x1, 2},
  {multiplexer2.pin(1), 0x2, 2},
  {multiplexer2.pin(2), 0x3, 2},
  {multiplexer2.pin(3), 0x4, 2},
  {multiplexer2.pin(4), 0x5, 2},
  {multiplexer2.pin(5), 0x6, 2},
  {multiplexer2.pin(6), 0x7, 2},
  {multiplexer2.pin(7), 0x8, 2},
  {multiplexer2.pin(8), 0x9, 2},
  {multiplexer2.pin(9), 0xA, 2},
  {multiplexer2.pin(10), 0xB,2},
  {multiplexer2.pin(11), 0xC,2},
  {multiplexer2.pin(12), 0xD, 2},
  {multiplexer2.pin(13), 0xE, 2},
  {multiplexer2.pin(14), 0xF, 2},
  {multiplexer2.pin(15), 0x10,2},
};
Digital buttons3[] = {
  {multiplexer3.pin(0), 0x1, 3},
  {multiplexer3.pin(1), 0x2, 3},
  {multiplexer3.pin(2), 0x3, 3},
  {multiplexer3.pin(3), 0x4, 3},
  {multiplexer3.pin(4), 0x5, 3},
  {multiplexer3.pin(5), 0x6, 3},
  {multiplexer3.pin(6), 0x7, 3},
  {multiplexer3.pin(7), 0x8, 3},
  {multiplexer3.pin(8), 0x9, 3},
  {multiplexer3.pin(9), 0xA, 3},
[... 156 lines cut ...]
    return map(val, 1001, 1023, 1023, 1023);
  }
  
}

void setup() {  
  potentiometer0.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer1.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer2.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer3.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer4.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer5.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer' 
  potentiometer6.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer7.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer8.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer9.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer10.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer11.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer12.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer13.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer14.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer'
  potentiometer15.map(mapCalibrated); // apply the 'mapCalibrated' function on the analog input of 'potentiometer' 
  
}
void loop() {MIDI_Controller.refresh();}

tttapa, 2018-07-26:
There is no way I can ever try that code, because I don't have the same hardware you have.

Could you please try to post a minimal example that shows the behaviour you mentioned?

mkothencz, 2018-07-26:
https://www.youtube.com/watch?v=qbX7qgLZIo0&feature=youtu.be In this video you can see when I push up fader 1,5,6,9,10.

tttapa, 2018-07-26:
I see. Is it always on the same faders? Or is it just random?

The strange thing is that I haven't been able to reproduce it, so I don't really have a clue where to start debugging.  
I'll check the mathematics behind the analog input circuitry, tonight. They should be alright, but who knows, maybe there's an overflow somewhere.

mkothencz, 2018-07-26:
It is just random.

I tried use capacitor on analog inputs, but only slowed the faders down, but jumps at same speed happens.

tttapa, 2018-07-26:
I feel so stupid right now.  
The version I have locally on my computer has a filter factor of 2, so the intermediate values are 10 + 2*2 bits wide and fit in a 16-bit int. However, the version on GitHub has [a filter factor of 3](https://github.com/tttapa/MIDI_controller/blob/1ecbd2c86e2268b36391db0ae5a5390cdbe90b8a/src/Settings/Settings.h#L4), so the values are 10 + 2*3 bits wide, and don't fit in a 16-bit int.

The solution: just change the filter factor to 2, or make the `int_t` for the filter 32 bits wide instead of 16. The latter will come at a performance and memory cost.

mkothencz, 2018-07-27:
Thank you! I can test it next week, but I am sure it will solve my problem.

mkothencz, 2018-07-30:
This two change options solved my problem, thanks! :)

Now I have only bit noise problem between 0-20% and 80-100%.
Do you have any idea, what I can try to make a bit smoother?

tttapa, 2018-07-30:
There are two options: 

1. Increase the filter factor again (note that this will also increase the latency if you make it too high), and change the filter type from `int16_t` to `int32_t`.
2. Filter the input before applying the map function. This is the best approach, I think, and it's the default in the next release. Move up [this line](https://github.com/tttapa/MIDI_controller/blob/8437236ddac62571b44f6a5af493850839b96c23/src/MIDI_Outputs/Analog.cpp#L16) two lines.

davidgauze, 2019-03-05:
ola, precisava ajuda, tenho nocçao basica de arduino mega 2560, queria criar algo com 10 fader, 3 encoder e 40 botao poderiam me ajudar???