tttapa/MIDI_controller #50: Calibrating a potentiometer

mikereynoldss, 2018-06-19:
Hey dude 

I'm having some problems calibrating a potentiometer. Some guy online helped me with the code but I can't get it to work. Do you think this is not supported by the library?

Here is the code I used:

`#include <MIDI_Controller.h>

AnalogMultiplex multiplexerF(A13, {34, 35, 36, 37} );
Analog potentiometersF[] = {
  { multiplexerF.pin(12), 0x1, 2 },
};

int potentiometerCalibrate(int val) {

  if (val <= 44)
  {
    return map(val, 0, 44, 0, 255);
  }
  if (val > 44 && val < 975)
  {
    return map(val, 45, 974, 255, 767);
  }
  if (val >= 975 )
  {
    return map(val, 975, 1023, 767, 1023);
  }
  
}

void setup() {
    potentiometersF[0].map(potentiometerCalibrate);
  }
void loop() {
  MIDI_Controller.refresh();
}`

And here is the potentiometer curve (the blue one, ignore the others):

https://scontent.fopo1-1.fna.fbcdn.net/v/t1.15752-9/35464607_2026512084057598_4902622490263552000_n.png?_nc_cat=0&oh=b1596ff439d3c7c9ad577a867d871832&oe=5B9EAF48

y axis is analog value and x axis is % of slide.

tttapa, 2018-06-19:
That code compiles just fine for me. Mappings like that are supported by the library.

mikereynoldss, 2018-06-19:
The code is compiling to me too. But when I upload it and plug the MIDI device in, my software even recognizes two potentiometers and is working awfully. Do you think the code could be done in a different way?

mikereynoldss, 2018-06-19:
The values only change from 0 and 1% on my fader half of the potentiometer does nothing

tttapa, 2018-06-19:
I tested it myself on a Teensy 3.2 (without a mutliplexer), and it works just as expected, so I suspect that there's something wrong with your wiring.

closed 2018-06-19 by tttapa

mikereynoldss, 2018-06-19:
Yeah I checked my cabling and its working now. Thanks a bunch