tttapa/MIDI_controller #53: Multiplexer's enable input

mkothencz, 2018-07-05:
Hello,

I would like to connect three button multiplexer's output to one pin on an Arduino Leonardo.
I would like to use multiplexer's enable input to select them in sequence.

Is this possible? (I tried to write ,,digitalWrite(Pin, LOW/HIGH)" before each Digital buttonsX[]={....   but not work.)

My code now:

#include <MIDI_Controller.h>
AnalogMultiplex multiplexer1(13, {4,3,12,6,} );
AnalogMultiplex multiplexer2(13, {4,3,12,6,} );
AnalogMultiplex multiplexer3(13, {4,3,12,6,} );

Digital buttons1[] = {
  {multiplexer1.pin(0), 0x1, 1},
  {multiplexer1.pin(1), 0x2, 1},
  {multiplexer1.pin(2), 0x3, 1},
  {multiplexer1.pin(3), 0x4, 1},
  {multiplexer1.pin(4), 0x5, 1},
  {multiplexer1.pin(5), 0x6, 1},
  {multiplexer1.pin(6), 0x7, 1},
  {multiplexer1.pin(7), 0x8, 1},
  {multiplexer1.pin(8), 0x9, 1},
  {multiplexer1.pin(9), 0xA, 1},
  {multiplexer1.pin(10), 0xB, 1},
  {multiplexer1.pin(11), 0xC, 1},
  {multiplexer1.pin(12), 0xD, 1},
  {multiplexer1.pin(13), 0xE, 1},
  {multiplexer1.pin(14), 0xF, 1},
  {multiplexer1.pin(15), 0x10, 1},
};
Digital buttons2[] = {
  {multiplexer2.pin(0), 0x1, 2},
  {multiplexer2.pin(1), 0x2, 2},
  {multiplexer2.pin(2), 0x3, 2},
  {multiplexer2.pin(3), 0x4, 2},
  {multiplexer2.pin(4), 0x5, 2},
  {multiplexer2.pin(5), 0x6, 2},
  {multiplexer2.pin(6), 0x7, 2},
  {multiplexer2.pin(7), 0x8, 2},
  {multiplexer2.pin(8), 0x9, 2},
  {multiplexer2.pin(9), 0xA, 2},
  {multiplexer2.pin(10), 0xB,2},
  {multiplexer2.pin(11), 0xC,2},
  {multiplexer2.pin(12), 0xD, 2},
  {multiplexer2.pin(13), 0xE, 2},
  {multiplexer2.pin(14), 0xF, 2},
  {multiplexer2.pin(15), 0x10,2},
};
Digital buttons3[] = {
  {multiplexer3.pin(0), 0x1, 3},
  {multiplexer3.pin(1), 0x2, 3},
  {multiplexer3.pin(2), 0x3, 3},
  {multiplexer3.pin(3), 0x4, 3},
  {multiplexer3.pin(4), 0x5, 3},
  {multiplexer3.pin(5), 0x6, 3},
  {multiplexer3.pin(6), 0x7, 3},
  {multiplexer3.pin(7), 0x8, 3},
  {multiplexer3.pin(8), 0x9, 3},
  {multiplexer3.pin(9), 0xA, 3},
  {multiplexer3.pin(10), 0xB,3},
  {multiplexer3.pin(11), 0xC, 3},
  {multiplexer3.pin(12), 0xD, 3},
  {multiplexer3.pin(13), 0xE, 3},
  {multiplexer3.pin(14), 0xF, 3},
  {multiplexer3.pin(15), 0x10,3},
};

void setup() {  
}
void loop() {MIDI_Controller.refresh();}

tttapa, 2018-07-05:
No, that is not possible at the moment. 

However, there's nothing stopping you from editing the [AnalogMultiplex source code](https://github.com/tttapa/MIDI_controller/blob/1ecbd2c86e2268b36391db0ae5a5390cdbe90b8a/src/ExtendedInputOutput/AnalogMultiplex.cpp#L5-L15) to add support for the enable pin. Just enable it upon entry of the `AnalogMultiplex::analogRead` and `AnalogMultiplex::digitalRead` functions, and disable it again before returning.

tttapa, 2019-07-31:
This has been implemented in tttapa/Control-Surface@c69bac999aaa2c373d4f4df503b217ada6f88165.

closed 2019-07-31 by tttapa