tttapa/MIDI_controller #54: Adding Sustain CC 64.

rabbiccu, 2018-07-06:
Everything so far is working great, I am very happy with my midified organ. So far I have all potentiometers. How do I add a sustain pedal? This would be a digital input on cc# 64.

tttapa, 2018-07-09:
The master branch has a class DigitalCC that you could use.

rabbiccu, 2018-07-12:
All the examples I can find are for triggering a note. Is there an example of using digital reads to send midi cc, the same way the analog class isul used?

tttapa, 2018-07-12:
You could use something like this:
```cpp
const pin_t sustainPin = 2;
const uint8_t channel = 1;

DigitalCC sustainPedal = { sustainPin, MIDI_CC::Damper_Pedal, channel };
```

rabbiccu, 2018-08-16:
So this didn't work although I'm not sure I am doing this right. 

The sustain pedal just passes 5v through it to the appropriate input on my Leonardo. As I'm very familiar now with the analog input scheme is there a way to just use that scheme instead of the DigitalCC class, which isn't working and I have no idea how to use?

 Analog potentiometer(A0, MIDI_CC:: Damper_Pedal, 1);

tttapa, 2018-08-16:
Usually, switches are wired as explained [here](https://github.com/tttapa/MIDI_controller/wiki/Hardware#pushbuttons-and-switches).

If you cannot connect it to ground, you need to invert it in software: https://github.com/tttapa/MIDI_controller/blob/79070e9531d738d1e47a8c6d94b4a5c4e502fe28/src/MIDI_Outputs/DigitalCC.cpp#L15-L18

rabbiccu, 2018-08-16:
Seems like the footswitch, which only has two terminals, just closes the circuit. When the circuit is open it would pass 0v and when closed would pass 5v?  Like this from Dave's MIDI CC List says:

> On/Off switch that controls sustain. (See also Sostenuto CC 66)
0 to 63 = Off, 64 to 127 = On

tttapa, 2018-08-16:
No, if the switch is open (not pressed), the voltage on the Arduino input is 5V, because it's pulled up through the internal pull-up resistor (it's inside of the Arduino). If the switch is closed, it pulls the voltage low to ground, so the input is at 0V.

That's all you have to worry about, the MIDI Controller library will handle the rest for you.

[image]

rabbiccu, 2018-08-16:
Ok so I'll just throw your code above in before i name the inputs. Will I use a digital input or an analog input?

tttapa, 2018-08-16:
Digital, because a switch is digital.

rabbiccu, 2018-08-16:
```
#include <MIDI_Controller.h> // Include the library

 void DigitalCC::invert()  // Invert the button state 
 { 
   invertState = true; 
 } 

const pin_t sustainPin = 2;
const uint8_t channel = 1;

DigitalCC sustainPedal = { sustainPin, MIDI_CC::Damper_Pedal, channel 1};

//here all my working analog input code

void setup() {}

void loop() {

  MIDI_Controller.refresh();
}
```

Is this good code to start with if the pedal is between Digital Input 2 and 5V? BTW thanks!

tttapa, 2018-08-16:
No. I don't know if your pedal can be wired up to ground, but if so, you have to follow the schematic I posted above.

Also delete lines 3-6.

rabbiccu, 2018-08-16:
I'm saying it cant be wired to ground and doesnt have a ground terminal, just two wires coming out of the pedal.

https://www.gopandy.com/wp-content/uploads/2017/02/maudiosustainpedalsp23-750x750.jpg

So in that case I would need to have lines 3-6 to invert, right?

tttapa, 2018-08-16:
What do you mean "it can't be wired to ground"? Tip to digital input pin, sleeve to ground, done.

rabbiccu, 2018-08-16:
Sorry, so then no 5V?

tttapa, 2018-08-16:
No, the switch in the schematic (S1) just has two terminals. The resistor is inside of the Arduino.

rabbiccu, 2018-08-29:
This worked.

tttapa, 2018-08-29:
Great! Glad to hear.

closed 2018-08-29 by tttapa