tttapa/MIDI_controller #57: buttons doing unexpected stuff

MistrKernnunos, 2018-08-13:
Hi, i have a problem.
I am using buttons conected to teensy 3,2 through mux (74hc4051). the core of it is that, i have two buttons, that when i press onei recive midi signals like i pushed both.

**Arduino board**:  ? Arduino UNO, Arduino Leonardo, Arduino DUE, Arduino MEGA, Arduino Micro, Teensy 3.2, ... ?  
**Schematic**: ?  

### Software versions:

**MIDI Controller library**: ? 3.1.0 ?  
**Arduino IDE: 1.8.5?  
**Operating System**: ? Windows
**Operating System version**: ? 10
(**Teensyduino**): ? 1.42 ? 

### Settings in the IDE
midi+usb, max clock speed, but not overc
### Full code 

```cpp

#define USE_ROTARY_ENCODER
#include <MIDI_Controller.h>

const uint8_t Channel = 1;       // MIDI channel 1
const int speedMultiply = 1;     // No change in speed of the encoder (number of steps is multiplied by 1)

RotaryEncoder encoder1(1, 0, 22, Channel, speedMultiply, 4, TWOS_COMPLEMENT);
RotaryEncoder encoder2(3, 2, 24, Channel, speedMultiply, 4, TWOS_COMPLEMENT);
RotaryEncoder encoder3(5, 4, 25, Channel, speedMultiply, 4, TWOS_COMPLEMENT);
RotaryEncoder encoder4(7, 6, 26, Channel, speedMultiply, 4, TWOS_COMPLEMENT);
RotaryEncoder encoder5(8, 9, 27, Channel, speedMultiply, 4, TWOS_COMPLEMENT);
RotaryEncoder encoder6(10, 11, 28, Channel, speedMultiply, 4, TWOS_COMPLEMENT);
RotaryEncoder encoder7(12, 13, 23, Channel, speedMultiply, 4, TWOS_COMPLEMENT);
RotaryEncoder encoder8(14, 15, 21, Channel, speedMultiply, 4, TWOS_COMPLEMENT);

    //buttons
  AnalogMultiplex multiplexers[] = {
    
  {16,{19, 20, 21}},
  {17,{19, 20, 21}},
  {18,{19, 20, 21}}
  
  };

  DigitalCC buttons[] = {
{multiplexers[0].pin(0), 6, 1},
{multiplexers[0].pin(1), 5, 1},
{multiplexers[0].pin(2), 4, 1},
{multiplexers[0].pin(3), 2, 1},
{multiplexers[0].pin(4), 7, 1},
{multiplexers[0].pin(5), 8, 1},
{multiplexers[0].pin(6), 3, 1},
{multiplexers[0].pin(7), 1, 1},

{multiplexers[1].pin(0), 37, 1},
{multiplexers[1].pin(1), 38, 1},
{multiplexers[1].pin(2), 44, 1},
{multiplexers[1].pin(3), 39, 1},
{multiplexers[1].pin(4), 32, 1},
{multiplexers[1].pin(5), 45, 1},
{multiplexers[1].pin(6), 43, 1},
{multiplexers[1].pin(7), 33, 1},

{multiplexers[2].pin(0), 35, 1},
{multiplexers[2].pin(1), 31, 1},
{multiplexers[2].pin(2), 36, 1},
{multiplexers[2].pin(3), 42, 1},
{multiplexers[2].pin(4), 41, 1},
{multiplexers[2].pin(5), 34, 1},
{multiplexers[2].pin(6), 30, 1},
{multiplexers[2].pin(7), 40, 1},
  };

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
MIDI_Controller.refresh();
}
```

## Steps taken to try to diagnose or solve the problem
photo of midi monitor
[image]

tttapa, 2018-08-13:
Most of these problems turn out to be a wiring problem of the address lines of the mux. Are you 100% sure that you selected the correct pins, and that they are connected to the correct address pins?

tttapa, 2018-08-21:
Did you manage to solve it?

tttapa, 2018-08-21:
Could you try the following test?

```cpp
#include <MIDI_Controller.h> // Include the library

constexpr pin_t addr_1 = 19;
constexpr pin_t addr_2 = 20;
constexpr pin_t addr_3 = 21;

// Create an instance of 'AnalogMultiplex' with the output pin of the multiplexer connected to
// analog input pin A0 and the address pins connected to pins 19, 20, 21
AnalogMultiplex multiplexer(A0, { addr_1, addr_2, addr_3 } );

void setup() {
  multiplexer.begin();
  Serial.begin(115200);
  while(!Serial);
}

void loop() {
  for (uint8_t i = 0; i < 8; i++) {
    multiplexer.analogRead(i);
    Serial.print(i);
    Serial.print(": ");
    Serial.print(digitalRead(addr_3));
    Serial.print(digitalRead(addr_2));
    Serial.print(digitalRead(addr_1));
    Serial.println();
    delay(500);
  }
  delay(10000);
}
```
Then open the Serial monitor (`CTRL+SHIFT+M`) and inspect the result. It should look exactly like this:
```
0: 000
1: 001
2: 010
3: 011
4: 100
5: 101
6: 110
7: 111
```

If you get a different result, there is a bug in the library. However, I've checked it again, and tested the above on my Teensy 3.2, and it works as expected, so I think that's rather unlikely (but not impossible, of course).  
If you get the same result, but the multiplexers still don't work, you should look at the hardware again. Your results seem to suggest that something is wrong with address line 3 (pin 21).  
The fact that this only happens sometimes leads me to believe that it's a hardware issue as well.

tttapa, 2018-09-05:
What were the results of the test I proposed?

MistrKernnunos, 2018-09-05:
It turned out right

closed 2022-02-20 by MistrKernnunos