tttapa/MIDI_controller #60: Current midi value for each analog input

squirelo, 2018-08-16:
Hi,

I'm using 4 faders and I need to know their current midi value (0-127) to show it in a small oled screen connected to my Arduino. 

Is there an easy way to get the current value for each fader from your library? 
A small example would be awesome :)

Sorry if this question looks noobish, I spent consequent time digging your code to understand how to do it by myself but it seems I need a little extra help.

Thx a lot

squirelo, 2018-08-19:
Awesome, thanks a lot for your reactivity, much appreciated.

I had to learn about hierarchy and how to derive a class, but here it is for other who will have the same need than me.

`#include <MIDI_Controller.h>

const int buttonPin = 7;     // the number of the pushbutton pin
volatile int buttonState = 0;         // variable for reading the pushbutton status

// Derived AnalogCC class, look this https://playground.arduino.cc/Code/Hierarchy
class AnalogCCOldVal : public AnalogCC {
  public:  
  
    AnalogCCOldVal(pin_t analogPin, uint8_t controller, uint8_t channel)
      : AnalogCC(analogPin, controller, channel) {}
      
    uint16_t getOldVal();  
    
};

uint16_t AnalogCCOldVal::getOldVal() { 
  return oldVal; 
}

AnalogCCOldVal faders[] = {
    {A0, MIDI_CC::Channel_Volume, 1}, // Create a new instance of class 'Analog' on pin A0, controller number 0x07 (channel volume), on MIDI channel 1.
    {A1, MIDI_CC::Channel_Volume, 2},
    {A2, MIDI_CC::Channel_Volume, 3},
    {A3, MIDI_CC::Channel_Volume, 4},
};

void setup() {
  Serial.begin(9600);
    // initialize the pushbutton pin as an input:
  pinMode(buttonPin, INPUT);
  // Attach an interrupt to the ISR vector
 // Attach an interrupt to the ISR vector
  attachInterrupt(digitalPinToInterrupt(7), pin_ISR, RISING);

  }

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}

void pin_ISR() {
  buttonState = digitalRead(buttonPin);
  doSomethingWithValues();
  }

void doSomethingWithValues() {
  Serial.println(faders[1].getOldVal());
  Serial.println(faders[2].getOldVal());
  Serial.println(faders[3].getOldVal());
  Serial.println(faders[4].getOldVal());
}
        `

closed 2018-08-19 by squirelo

tttapa, 2018-08-20:
Great!

One thing to watch out for: Sending Serial messages and other things that rely on interrupts can cause serious problems if you use them inside of an ISR (where interrupts are disabled automatically).  
The best solution is to set `buttonState` to `true` (keeping the ISR as short as possible) and then check the value of `buttonState` in your main loop.

For example:
```cpp
volatile bool buttonWasPressed = false;

void loop() {
  MIDI_Controller.refresh();
  if (buttonWasPressed) {
    doSomethingWithValues();
    buttonWasPressed = false;
  }
}

void pin_ISR() {
  buttonWasPressed = true;
}
```