tttapa/MIDI_controller #68: a consistent increase in the value of the rotary encoder

santta, 2018-11-26:
Hello, I thank you for the wonderful library!!!
I want to use your library for Lightroom, too. 
Faced with the problem of using rotary encoder-only one value of the selected midi CC is sent when rotating to the right, also when rotating to the left, the same value is sent.
How can I adjust the sequential increase in the value of the desired midi CC, when turning the rotary encoder, like an analog potentiometer?

## Description of the problem or question

## Steps to reproduce the problem

### Hardware

**Arduino board**:  ? Arduino UNO, Arduino Leonardo, Arduino DUE, Arduino MEGA, Arduino Micro, Teensy 3.2, ... ?  
**Schematic**: ?  

### Software versions:

**MIDI Controller library**: ? 3.0.0 ?  
**Arduino IDE**: ? 1.8.4 ?  
**Operating System**: ? Windows, OSX, Ubuntu, ... ?  
**Operating System version**: ? 10, 16.04, ... ?  
(**Teensyduino**): ? 1.39 ? 
(**Encoder library**): ? 1.4.1 ?  
(**MIDIUSB library**): ? 1.0.3 ? 

### Settings in the IDE

### Full code 

```cpp
// Your code here
```

## Steps taken to try to diagnose or solve the problem
? Tried the MIDI debug mode, used a MIDI monitor, ... ?

## The goal of your project and aditional information

closed 2018-11-26 by santta

tttapa, 2018-11-27:
This is the expected behavior of rotary encoders, they are inherently relative, while a potentiometer is absolute. Most MIDI software will allow you to use relative or incremental modes for encoders.

santta, 2018-11-30:
thanks, all worked well