tttapa/MIDI_controller #74: LCD

pandeagus, 2019-01-21:
Before opening a new issue, please read the [installation instructions](https://github.com/tttapa/MIDI_controller#installation) in the README. Also check if your question is answered in the [Wiki](https://github.com/tttapa/MIDI_controller/wiki) or in a [previous issue](https://github.com/tttapa/MIDI_controller/issues?q=is%3Aissue).   
If you do create a new issue, please follow the template below.  
--- (delete these first lines) ---

## Description of the problem or question

## Steps to reproduce the problem

### Hardware

**Arduino board**:  ? Arduino UNO, Arduino Leonardo, Arduino DUE, Arduino MEGA, Arduino Micro, Teensy 3.2, ... ?  
**Schematic**: ?  

### Software versions:

**MIDI Controller library**: ? 3.0.0 ?  
**Arduino IDE**: ? 1.8.4 ?  
**Operating System**: ? Windows, OSX, Ubuntu, ... ?  
**Operating System version**: ? 10, 16.04, ... ?  
(**Teensyduino**): ? 1.39 ? 
(**Encoder library**): ? 1.4.1 ?  
(**MIDIUSB library**): ? 1.0.3 ? 

### Settings in the IDE

### Full code 

```cpp
// Your code here
```

## Steps taken to try to diagnose or solve the problem
? Tried the MIDI debug mode, used a MIDI monitor, ... ?

## The goal of your project and aditional information

tttapa, 2019-01-21:
It seems like you didn't add anything to the issue. If this was a mistake, you can edit your post, if you don't have any questions, just let me know if I should delete it.

tttapa, 2019-01-22:
The MIDI controller doesn't support this, but you can use the Control Surface library.

Be sure to use the revision-4 branch until further notice.
It's not finished yet, but it should compile and work.

https://github.com/tttapa/Control-Surface/blob/revision-4/examples/MIDIOutput/PushButtonsLEDs/PushButtonsLEDs.ino

pandeagus, 2019-01-23:
IM SORRY IM ASKING YOU AGAIN..
I GET ERROR WHEN COMPILE THIS CODE

[image]

pandeagus, 2019-01-23:
Sorry
after I enter several libraries from the adafruit error the compile is gone. i try code from you but the leds still don't turn on. did I make a mistake?  I've led the series to ground.

tttapa, 2019-01-23:
I'm sorry about the error, I have to remove the Adafruit depenencies when I do an official release.

The LED relies on the software on the other side to turn it on or off. For example, if you have a DJ or digital audio workstation on your computer, the LED will also turn on if you click the button on your computer. 

What software are you using?

pandeagus, 2019-01-23:
I use amplitube and bias fx.. i just want make midi foot controller for
that software.. thank very much for your help..

tttapa, 2019-01-23:
Sadly, Amplitube doesn't support MIDI feedback. ([source](https://cgi.ikmultimedia.com/ikforum/viewtopic.php?f=5&t=16741))
This means that the Arduino can never know the state of the effects. 
The only thing you can do is keep the state on the Arduino, but if you enable the effect using your computer, or when it was already on when you plugged in the Arduino, the LED will _always_ be wrong.

If you want the LED to work correctly, you'll have to wait for Amplitube to fully support MIDI controllers, or use different software.

pandeagus, 2019-01-23:
Hm.. thanks you very much for your help... bu can ask you again..
how can one button when press.. can MIDI SEND NOTE and MIDI SEND CONTRO
CHANGE together..
with your library function..

i send the screenshot result of midi ox..
i ask this because if can SEND MIDI NOTE and MIDI CONTROL CHANGE together
.. Bias fx software can be control..
i dont Know why if just sending MIDI Control Cange cant respon on bias fx
software.. and if just send MIDI send note still cant respon on bias fx
software
but if sending together MIDI NOTE and MIDI CONTROL CHANGE can control that
Software.

I'm sorry for interrupting your time. and if possible, may I know your real
name ?
one againt thank you very much for your help..

Pada tanggal 24 Jan 2019 1.57 AM, "tttapa" <notifications@github.com>
menulis:

Sadly, Amplitube doesn't support MIDI feedback. (source
<https://cgi.ikmultimedia.com/ikforum/viewtopic.php?f=5&t=16741>)
This means that the Arduino can never know the state of the effects.
The only thing you can do is keep the state on the Arduino, but if you
enable the effect using your computer, or when it was already on when you
plugged in the Arduino, the LED will *always* be wrong.

If you want the LED to work correctly, you'll have to wait for Amplitube to
fully support MIDI controllers, or use different software.

pandeagus, 2019-01-23:
[image]

tttapa, 2019-01-24:
According to the [Bias FX documentation](https://help.positivegrid.com/hc/en-us/articles/207744223-How-to-Use-MIDI-Control-in-Standalone-Version-), using just Control Change should work.

Did you manage to get the LEDs working?

tttapa, 2019-01-24:
I have an exam tomorrow, so I can't really help you today.  
Feel free to post any questions you have, but I won't be able to answer before tomorrow 17:00 CET.

pandeagus, 2019-01-24:
Thank for the bias fx documentation..  I tried it using an example from the ProgramChanger from library but it didn't work. and i dont know why. can you give me clue ?

and for the LED i just try make new function code like this . its working but sometime little slow respons 
for On LEd. and i dont why.. but its okay 

void LEDswitch1 (){
int val1 = 0;
int ledOn1 = 0;
int pushed1 = 0;

val1 = digitalRead(switch1);

if (val1 == HIGH && ledOn1 == LOW){
  pushed1 = 1-pushed1;
  delay(5);
}
ledOn1 = val1;
if(pushed1 == HIGH){
  Serial.println("led1 OFF");
  digitalWrite(led1, LOW);
} else{
  Serial.println("led1 ON");
  digitalWrite(led1, HIGH);
}
  delay(5);
   
}

tttapa, 2019-01-24:
I don't know why the ProgramChanger doesn't work, last time I tested it, it worked, but I've made some changes after that, so I could have messed it up somehow. (The channel could be wrong, I think.)

Does the program changer show anything in MIDIOX? Did you connect the push buttons correctly (pins 5 & 6 via a push button to ground)?

I'll write you a better LED sketch tomorrow.

pandeagus, 2019-01-26:
[image]

hello brother i get error when try your example PushButtonsLEDsNoMIDIFeedback.ino.
how to solve?

tttapa, 2019-01-26:
I'm sorry, I later added another commit where I used the standard library, which is not available on AVR, and I forgot to exclude it when compiling for AVR. 

It should be fixed now: https://github.com/tttapa/Control-Surface/commit/dd8e1385c517306506cd2feb64ddb98b39d4a9a7

pandeagus, 2019-01-27:
hello.. can you give me example using 74HC4051 multiplexer with 8 button 
change MIDI CC and MIDI NOTE using your library. 
im sorry i ask many thing to you.
thank you..

tttapa, 2019-01-27:
Did you see the [Getting Started](https://tttapa.github.io/Control-Surface/Doc/Doxygen/db/df0/md__home_pieter_GitHub_Control-Surface_doc_Getting-Started.html) page and the example that goes with it?