tttapa/MIDI_controller #75: DIN midi output?

lameller, 2019-01-25:
Hi

How would i go about using a standard DIN midi socket to send midi from the arduino instead of through the usb port?

tttapa, 2019-01-25:
Connect the hardware like this: https://tttapa.github.io/Arduino/MIDI/Chap02-MIDI-Hardware.html
And you could use the following code:
```cpp
HardwareSerialMIDI_Interface midi(Serial, MIDI_BAUD);
```

closed 2019-04-20 by tttapa

Rolbista, 2019-05-13:
After using it like this before `setup()`
it throws an error:
```
Verifying...
DynamicCCPot:42: error: no matching function for call to 'HardwareSerialMIDI_Interface::HardwareSerialMIDI_Interface(Serial_&, int)'
 HardwareSerialMIDI_Interface midi(Serial, MIDI_BAUD);
                                                    ^
/Users/wojciechg/repos/arduino-midi/DynamicCCPot.ino:42:52: note: candidates are:
In file included from /Users/wojciechg/Documents/Arduino/libraries/MIDI_controller-master/src/MIDI_Controller.h:26:0,
                 from /Users/wojciechg/repos/arduino-midi/DynamicCCPot.ino:3:
/Users/wojciechg/Documents/Arduino/libraries/MIDI_controller-master/src/./MIDI_Interfaces/SerialMIDI_Interface.h:57:3: note: HardwareSerialMIDI_Interface::HardwareSerialMIDI_Interface(HardwareSerial&, long unsigned int)
   HardwareSerialMIDI_Interface(HardwareSerial &serial, unsigned long baud) : SerialMIDI_Interface(serial, baud) {}
   ^
/Users/wojciechg/Documents/Arduino/libraries/MIDI_controller-master/src/./MIDI_Interfaces/SerialMIDI_Interface.h:57:3: note:   no known conversion for argument 1 from 'Serial_' to 'HardwareSerial&'
/Users/wojciechg/Documents/Arduino/libraries/MIDI_controller-master/src/./MIDI_Interfaces/SerialMIDI_Interface.h:54:7: note: constexpr HardwareSerialMIDI_Interface::HardwareSerialMIDI_Interface(const HardwareSerialMIDI_Interface&)
 class HardwareSerialMIDI_Interface : public SerialMIDI_Interface<HardwareSerial>
       ^
/Users/wojciechg/Documents/Arduino/libraries/MIDI_controller-master/src/./MIDI_Interfaces/SerialMIDI_Interface.h:54:7: note:   candidate expects 1 argument, 2 provided
/Users/wojciechg/Documents/Arduino/libraries/MIDI_controller-master/src/./MIDI_Interfaces/SerialMIDI_Interface.h:54:7: note: constexpr HardwareSerialMIDI_Interface::HardwareSerialMIDI_Interface(HardwareSerialMIDI_Interface&&)
/Users/wojciechg/Documents/Arduino/libraries/MIDI_controller-master/src/./MIDI_Interfaces/SerialMIDI_Interface.h:54:7: note:   candidate expects 1 argument, 2 provided
exit status 1
```

tttapa, 2019-05-24:
What board are you using? Different boards have different inheritance structures for the class of "Serial", requiring me to manually add support for each board.

Rolbista, 2019-05-24:
Yeah, I have figured that out now, using
`HardwareSerialMIDI_Interface midi(Serial1, MIDI_BAUD);`
it compiled. I still need to test it with actual components but it should be ok.

CaBachhav, 2019-06-29:
What if We want both way output simultaneously , like over USB & TX out also. All professional gears have both type of output.