tttapa/MIDI_controller #76: Rotary encoder sends four signals instead of one.

santta, 2019-02-17:
Before opening a new issue, please read the [installation instructions](https://github.com/tttapa/MIDI_controller#installation) in the README. Also check if your question is answered in the [Wiki](https://github.com/tttapa/MIDI_controller/wiki) or in a [previous issue](https://github.com/tttapa/MIDI_controller/issues?q=is%3Aissue).   
If you do create a new issue, please follow the template below.  
--- (delete these first lines) ---

## Description of the problem or question
Rotary encoder sends four signals instead of one. 
I tried to correct for different versions of the values ​​rotaryencoder.hpp (normal_encoder) and speedmultiplay (in the sketch). But the encoder still sends 4 signals at one turn. How can I fix this problem?

## Steps to reproduce the problem
RotaryEncoder.hpp

### Hardware
Mega2560

### Software versions:

MIDI Controller library: 3.1.0
Arduino IDE:  1.8.4
Windows10 
[image]

### Settings in the IDE

### Full code 

```cpp
// Your code here
```

## Steps taken to try to diagnose or solve the problem
? Tried the MIDI debug mode, used a MIDI monitor, ... ?

## The goal of your project and aditional information

tttapa, 2019-02-17:
Then you have a very strange encoder. Did you try using `NORMAL_ENCODER * 4`?

santta, 2019-02-18:
Yes, I tried this value, but nothing has changed. There may be a problem in the encoder itself, I will try to connect another one.

santta, 2019-02-18:
I installed another encoder (100% good) - four signals are sent again.

tttapa, 2019-02-18:
I have no idea, to be honest.

Does changing the value to 1 result in 16 messages?

santta, 2019-02-18:
I have now determined that the problem with the arduino IDE is on my PC. For some reason, the program does not work correctly.

closed 2019-04-20 by tttapa