tttapa/MIDI_controller #79: Arduino DUE don't works whit Button Matriz Example.

Gratiman, 2019-03-04:
## Description of the problem or question

When I compile the example "Ex.11.Button-Matrix" whith the Arduino Due board I get the following error and it does not compile the code, it just happens with that board:

`"C:\\Users\\Gratiman\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb -Os -Wl,--gc-sections "-TC:\\Users\\Gratiman\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.11\\variants\\arduino_due_x/linker_scripts/gcc/flash.ld" "-Wl,-Map,C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152/Ex.11.Button-Matrix.ino.map" -o "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152/Ex.11.Button-Matrix.ino.elf" "-LC:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152" -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--start-group -u _sbrk -u link -u _close -u _fstat -u _isatty -u _lseek -u _read -u _write -u _exit -u kill -u _getpid "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\sketch\\Ex.11.Button-Matrix.ino.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\Banks\\Bank.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\Banks\\BankSelector.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\ExtendedInputOutput\\AnalogMultiplex.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\ExtendedInputOutput\\ExtendedIOElement.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\ExtendedInputOutput\\ExtendedInputOutput.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\ExtendedInputOutput\\ShiftRegisterOut.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\Helpers\\Hysteresis.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\MIDI_Controller\\MIDI_Controller_Class.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\MIDI_Interfaces\\MIDI_Interface.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\MIDI_Outputs\\Analog.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\MIDI_Outputs\\ButtonMatrix.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\MIDI_Outputs\\Digital.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\MIDI_Outputs\\DigitalCC.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\MIDI_Outputs\\DigitalLatch.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDI_controller-master\\MIDI_Outputs\\MIDI_Control_Element.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\libraries\\MIDIUSB-master\\MIDIUSB.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152\\core\\variant.cpp.o" "C:\\Users\\Gratiman\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.11\\variants\\arduino_due_x/libsam_sam3x8e_gcc_rel.a" "C:\\Users\\Gratiman\\AppData\\Local\\Temp\\arduino_build_318152/..\\arduino_cache_113158\\core\\core_arduino_sam_arduino_due_x_dbg_5bbfd5192b497a66a4690bfd3a3ea30f.a" -Wl,--end-group -lm -gcc
C:\Users\Gratiman\AppData\Local\Temp\arduino_build_318152\sketch\Ex.11.Button-Matrix.ino.cpp.o: In function `__static_initialization_and_destruction_0':

C:\Users\Gratiman\Documents\Arduino\libraries\MIDI_controller-master\examples\Ex.11.Button-Matrix/Ex.11.Button-Matrix.ino:30: undefined reference to `ButtonMatrix<4u, 3u>::ButtonMatrix(unsigned short const (&) [4], unsigned short const (&) [3], unsigned char const (&) [4][3], unsigned char, unsigned char)'

C:\Users\Gratiman\AppData\Local\Temp\arduino_build_318152\sketch\Ex.11.Button-Matrix.ino.cpp.o: In function `_GLOBAL__sub_I_buttonmatrix':

C:\Users\Gratiman\Documents\Arduino\libraries\MIDI_controller-master\examples\Ex.11.Button-Matrix/Ex.11.Button-Matrix.ino:37: undefined reference to `ButtonMatrix<4u, 3u>::~ButtonMatrix()'

collect2.exe: error: ld returned 1 exit status

Usando librería MIDI_controller-master con versión 3.1.0 en la carpeta: C:\Users\Gratiman\Documents\Arduino\libraries\MIDI_controller-master 
Usando librería MIDIUSB-master con versión 1.0.3 en la carpeta: C:\Users\Gratiman\Documents\Arduino\libraries\MIDIUSB-master 
exit status 1
Error compilando para la tarjeta Arduino Due (Programming Port).`

## Steps to reproduce the problem
1. open the "Ex.11.Button-Matrix" example.
2. Select in the Arduino IDE: "Tools/Board/Arduino Due(Programming port)".
3. Click on compile or upload button.

### Hardware

**Arduino board**:  Arduino DUE

### Software versions:

**MIDI Controller library**: 3.1.0  
**Arduino IDE**: 1.8.8
**Operating System**: Windows 
**Operating System version**: 10 & 8.1 
(**Teensyduino**): N/A 
(**Encoder library**): 1.4.1  
(**MIDIUSB library**): 1.0.3

### Settings in the IDE

[image]

### Full code 

```cpp
// /*
This is an example of the "ButtonMatrix" class of the MIDI_controller library.
Connect a 4 × 3 matrix of buttons with the rows to pins 2, 3, 4 and 5, 
and the columns to pins 6, 7 and 8.
Pull-up resistors are not necessary, because the internal ones will be used. 

If you want to be able to press multiple buttons at once, add a diode 
in series with each button, as shown in the schematic on the Wiki:
https://github.com/tttapa/MIDI_controller/wiki/Hardware

The note numbers are specified in the 'addresses' array.
Map accordingly in your DAW or DJ software.

Written by tttapa, 24/09/2017
https://github.com/tttapa/MIDI_controller
*/

#include "MIDI_Controller.h" // Include the library

const uint8_t velocity = 0b1111111; // Maximum velocity (0b1111111 = 0x7F = 127)
const uint8_t addresses[4][3] = {   // the note numbers corresponding to the buttons in the matrix
  {  1,  2,  3 },
  {  4,  5,  6 },
  {  7,  8,  9 },
  { 10, 11, 12 }
};

// Create a new instance of the class 'ButtonMatrix', called 'buttonmatrix', with dimensions 4 rows and 3 columns, with the rows connected to pins 2, 3, 4 and 5
// and the columns connected to pins 6, 7 and 8, that sends MIDI messages with the notes specified in 'addresses' on MIDI channel 1, with velocity 127
ButtonMatrix<4, 3> buttonmatrix({2, 3, 4, 5}, {6, 7, 8}, addresses, 1, velocity);

void setup() {}

void loop() {
  // Refresh the buttons (check whether the states have changed since last time, if so, send it over MIDI)
  MIDI_Controller.refresh();
}
```

## Steps taken to try to diagnose or solve the problem
None.

## The goal of your project and aditional information
I only need to use that example to do a full keyboard whit the Arduino DUE.

tttapa, 2019-03-04:
I have no idea why I only wrote that for AVR, but it should be fixed now.

closed 2019-03-04 by tttapa

myklemykle, 2019-05-10:
Hi,

Appears not to be fixed.  Today I am getting the same error when I try to compile the same example for Teensy 3.0 .

`/Applications/Utilities/Arduino1.8.7.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-gcc -O2 -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=1557485075 -T/Applications/Utilities/Arduino1.8.7.app/Contents/Java/hardware/teensy/avr/cores/teensy3/mk20dx128.ld -lstdc++ -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/Ex.11.Button-Matrix.ino.elf /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/sketch/Ex.11.Button-Matrix.ino.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/Banks/Bank.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/Banks/BankSelector.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/ExtendedInputOutput/AnalogMultiplex.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/ExtendedInputOutput/ExtendedIOElement.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/ExtendedInputOutput/ExtendedInputOutput.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/ExtendedInputOutput/ShiftRegisterOut.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/Helpers/Hysteresis.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/MIDI_Controller/MIDI_Controller_Class.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/MIDI_Interfaces/MIDI_Interface.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/MIDI_Outputs/Analog.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/MIDI_Outputs/ButtonMatrix.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/MIDI_Outputs/Digital.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/MIDI_Outputs/DigitalCC.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/MIDI_Outputs/DigitalLatch.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/MIDI_controller-3.1.0/MIDI_Outputs/MIDI_Control_Element.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/libraries/SoftwareSerial/SoftwareSerial.cpp.o /var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/../arduino_cache_78806/core/core_6bcf93ee8c5f42f75706a568501fef48.a -L/var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611 -larm_cortexM4l_math -lm
/var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/sketch/Ex.11.Button-Matrix.ino.cpp.o: In function `__static_initialization_and_destruction_0':
/Users/mykle/Documents/Arduino/libraries/MIDI_controller-3.1.0/examples/Ex.11.Button-Matrix/Ex.11.Button-Matrix.ino:30: undefined reference to `ButtonMatrix<4u, 3u>::ButtonMatrix(unsigned short const (&) [4], unsigned short const (&) [3], unsigned char const (&) [4][3], unsigned char, unsigned char)'
/var/folders/t7/qqyds22x055f1kdfcy7fn3m00000gn/T/arduino_build_350611/sketch/Ex.11.Button-Matrix.ino.cpp.o: In function `_GLOBAL__sub_I_buttonmatrix':
/Users/mykle/Documents/Arduino/libraries/MIDI_controller-3.1.0/examples/Ex.11.Button-Matrix/Ex.11.Button-Matrix.ino:37: undefined reference to `ButtonMatrix<4u, 3u>::~ButtonMatrix()'
collect2: error: ld returned 1 exit status`

commit 629564ba (2019-03-04): Fix #79, include ButtonMatrix implementation non-AVR