tttapa/MIDI_controller #83: BankSelector - several toggle switches (latch) and several LEDs.

santta, 2019-03-24:
## Description of the problem or question
In my project, I use 4 switches (latch) and 4 LEDs.
I need to use unique bankSetting on each of the switches, how can I do this?

## Steps to reproduce the problem
I want to use BankSelektor like this: when all switches are turned off, bankSetting 1 is selected. When I turn on each switch (one at a time), I need to switch bankSetting 2, bankSetting 3, bankSetting 4, bankSetting 5 and at the same time the LED turns on.

This is necessary in order to establish specific values ​​of the channel or notes when changing banks.

### Hardware
Arduino MEGA 2560

### Software versions:

**MIDI Controller library**: 3.1.0 
**Arduino IDE**: 1.8.8
**Operating System**:  Windows 10

### Full code 

```cpp
// #define USE_ROTARY_ENCODER
#include <MIDI_Controller.h>

//const uint8_t Controller = 0x14; // MIDI Control Change controller number
const int speedMultiply = 1;     // No change in speed of the encoder (number of steps is multiplied by 1)

// Create a new instance of the class 'RotaryEncoder', called 'enc', on pin 2 and 3, controller number 0x14, on channel1,
// no change in speed (speed is multiplied by 1), it's used as a jog wheel, and the sign mode is set to two's complement.

//Энкодер
RotaryEncoder encoder1(12, 13, 0x01, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);//01
RotaryEncoder encoder2(48, 49, 0x03, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder3(38, 39, 0x05, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder4(28, 29, 0x07, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder5(50, 51, 0x09, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder6(46, 47, 0x0B, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder7(36, 37, 0x0D, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder8(26, 27, 0x0F, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder9(A10, A11, 0x11, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);//БАНК
RotaryEncoder encoder10(44, 45, 0x13, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder11(34, 35, 0x15, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder12(24, 25, 0x17, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder13(A12, A13, 0x19, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);//БАНК
RotaryEncoder encoder14(42, 43, 0x1B, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder15(32, 33, 0x1D, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder16(22, 23, 0x1F, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder17(A14, A15, 0x21, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);//БАНК
RotaryEncoder encoder18(40, 41, 0x23, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder19(30, 31, 0x25, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder20(20, 21, 0x27, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
//RotaryEncoder encoder21(10, 11, 0x29, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder22(6, 7, 0x2B, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder23(14, 15, 0x2D, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder24(18, 19, 0x2F, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
//RotaryEncoder encoder25(12, 13, 0x31, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder26(8, 9, 0x33, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder27(10, 11, 0x35, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);
RotaryEncoder encoder28(16, 17, 0x37, 1, speedMultiply, NORMAL_ENCODER, TWOS_COMPLEMENT);

//Кнопка энкодера для банка
Digital button9(A3, 0x12, 1, 127);
Digital button13(A4, 0x1A, 1, 127);
Digital button17(A5, 0x22, 1, 127);

//Мультики
AnalogMultiplex multiplexer1(A0, { 2, 3, 4, 5 } );
AnalogMultiplex multiplexer2(A1, { 2, 3, 4, 5 } );
AnalogMultiplex multiplexer3(A2, { 2, 3, 4, 5 } );

//Кнопки Энкодеров_мультиков
Digital button2[] = {
  {multiplexer2.pin(0), 0x02, 1},//КНОПКА ЭНКОДЕРА 1
  {multiplexer2.pin(3), 0x04, 1},//КНОПКА ЭНКОДЕРА 2
  {multiplexer1.pin(7), 0x06, 1},//КНОПКА ЭНКОДЕРА 3---------
  {multiplexer1.pin(9), 0x08, 1},//КНОПКА ЭНКОДЕРА 4-------------
  {multiplexer2.pin(12), 0x0A, 1},//КНОПКА ЭНКОДЕРА 5
  {multiplexer2.pin(8), 0x0C, 1},//6
  {multiplexer1.pin(8), 0x0E, 1},//-------7
  {multiplexer1.pin(10), 0x10, 1},//--------8
  {multiplexer2.pin(6), 0x14, 1},//10
  {multiplexer2.pin(4), 0x16, 1},//11
  {multiplexer1.pin(11), 0x18, 1},//---------12
  {multiplexer2.pin(5), 0x1C, 1},//14
  {multiplexer2.pin(15), 0x1E, 1},//15
  {multiplexer1.pin(12), 0x20, 1},//-----------16
  {multiplexer2.pin(7), 0x24, 1},//18
  {multiplexer2.pin(10), 0x26, 1},//19
  {multiplexer1.pin(13), 0x28, 1},//----------20
  {multiplexer2.pin(1), 0x2A, 1},//21
  {multiplexer2.pin(9), 0x2C, 1},//22
  {multiplexer2.pin(11), 0x2E, 1},//23
  {multiplexer1.pin(14), 0x30, 1},//--------24
  {multiplexer2.pin(2), 0x32, 1},//25
  {multiplexer2.pin(13), 0x34, 1},//26
  {multiplexer2.pin(14), 0x36, 1},//0х36
  {multiplexer1.pin(15), 0x38, 1},//-------------28
  {multiplexer3.pin(1), 0x39, 1},//КНОПКА_2
  {multiplexer3.pin(2), 0x3A, 1},//КНОПКА_3
  {multiplexer3.pin(0), 0x3B, 1},//КНОПКА_4
  {multiplexer3.pin(6), 0x3C, 1},//КНОПКА_5
  {multiplexer3.pin(5), 0x3E, 1},//КНОПКА_6
  {multiplexer3.pin(4), 0x3F, 1},//КНОПКА_7
  {multiplexer3.pin(3), 0x46, 1},//КНОПКА_8
  {multiplexer3.pin(7), 0x47, 1},//КНОПКА_9
  {multiplexer3.pin(8), 0x48, 1},//КНОПКА_10
  {multiplexer3.pin(9), 0x49, 1},//КНОПКА_11
  {multiplexer3.pin(10), 0x50, 1},//КНОПКА_12
  {multiplexer3.pin(12), 0x51, 1},//КНОПКА_17
  {multiplexer3.pin(11), 0x52, 1},//КНОПКА_18
  {multiplexer3.pin(13), 0x53, 1},//КНОПКА_19
  {multiplexer3.pin(14), 0x54, 1},//КНОПКА_20
  {multiplexer1.pin(0), 0x55, 1},//КНОПКА_21
  {multiplexer1.pin(2), 0x56, 1},//КНОПКА_22
  {multiplexer1.pin(3), 0x57, 1},//КНОПКА_23
  {multiplexer3.pin(15), 0x58, 1},//КНОПКА_24
  {multiplexer1.pin(4), 0x59, 1},//КНОПКА_25
  {multiplexer1.pin(5), 0x5A, 1},//КНОПКА_26
  {multiplexer1.pin(1), 0x5B, 1},//КНОПКА_27
  {multiplexer1.pin(6), 0x5C, 1},};//КНОПКА_28};

//BANK1 КРАСНЫЙ
Bank bank1(1);
BankSelector bankSelector1 (bank1, A6, 53, BankSelector::TOGGLE);

//BANK2 ЖЕЛТЫЙ
Bank bank2(2);
BankSelector bankSelector2 (bank2, A7, 53, BankSelector::TOGGLE);

//BANK1 ЗЕЛЕНЫЙ
Bank bank3(3);
BankSelector bankSelector3 (bank3, A8, 53, BankSelector::TOGGLE);

//BANK2 СИНИЙ
Bank bank4(4);
BankSelector bankSelector4 (bank4, A9, 53, BankSelector::TOGGLE);

void setup() {
//КРАСНЫЙ банк1_кнопки
bank1.add(button9, Bank::CHANGE_CHANNEL);
bank1.add(button13, Bank::CHANGE_CHANNEL);
bank1.add(button17, Bank::CHANGE_CHANNEL);
//крассный банк1_Энкодеры
bank1.add(encoder9, Bank::CHANGE_CHANNEL);
bank1.add(encoder13, Bank::CHANGE_CHANNEL);
bank1.add(encoder17, Bank::CHANGE_CHANNEL);

//ЖЕЛТЫЙ банк2_кнопки
bank2.add(button9, Bank::CHANGE_CHANNEL);
bank2.add(button13, Bank::CHANGE_CHANNEL);
bank2.add(button17, Bank::CHANGE_CHANNEL);
//желтый банк2_Энкодеры
bank2.add(encoder9, Bank::CHANGE_CHANNEL);
bank2.add(encoder13, Bank::CHANGE_CHANNEL);
bank2.add(encoder17, Bank::CHANGE_CHANNEL);

//ЗЕЛЕНЫЙ банк3_кнопки
bank3.add(button9, Bank::CHANGE_CHANNEL);
bank3.add(button13, Bank::CHANGE_CHANNEL);
bank3.add(button17, Bank::CHANGE_CHANNEL);
//зеленый банк3_Энкодеры
bank3.add(encoder9, Bank::CHANGE_CHANNEL);
bank3.add(encoder13, Bank::CHANGE_CHANNEL);
bank3.add(encoder17, Bank::CHANGE_CHANNEL);

//СИНИЙ банк4_кнопки
bank4.add(button9, Bank::CHANGE_CHANNEL);
bank4.add(button13, Bank::CHANGE_CHANNEL);
bank4.add(button17, Bank::CHANGE_CHANNEL);
//синий банк4_Энкодеры
bank4.add(encoder9, Bank::CHANGE_CHANNEL);
bank4.add(encoder13, Bank::CHANGE_CHANNEL);
bank4.add(encoder17, Bank::CHANGE_CHANNEL);}

void loop() {
  // Refresh the encoder (check whether the position has changed since last time, if so, send the difference over MIDI)
  MIDI_Controller.refresh();
}
```

## The goal of your project and aditional information

I create my midi controller for working with images

santta, 2019-03-26:
and is it possible to somehow set specific change parameters for CHANGE_CHANNEL and CHANGE_ADDRESS?

tttapa, 2019-04-20:
You probably want to use this setting: 

> - Multiple momentary switches (push buttons)
>
> Pressing one of the buttons selects the respective output:
> When starting the program, bankSetting 1 is selected,
> When the second button is pressed, bankSetting 2 is selected,
> When the n-th button is pressed, bankSetting n is selected.
>
> `BankSelector(bank, { button 1 pin, button 2 pin, ... , button n pin } )`

See <https://github.com/tttapa/MIDI_controller/wiki/Library-documentation#appendix-a-bankselector-modes>