tttapa/MIDI_controller #87: Triggering multiple notes from single button press - including MIDI notes not in code!

the-duckchild, 2019-06-07:
## Description of the problem or question

Hi - I currently have thirteen buttons hooked up to two digital pins (0,1)  on a Teensy 3.2 through two 74HC4051 muxes. The Muxes are controled by pins 10,11,12. Some of the buttons are DigitalLatch, some are simple Digital (see code below)

Everything is working as it should up to a point. However the DigitalLatch buttons on Pin 7 and Pin 3 of MUX 1 seem to alternate between firing both pins at a very fast rate when either button is pushed. I would assume this was a soldering issue, but if it is I am struggling to find it - and then something even stranger is happening:
On Pin7 of Mux 2 the Digital button appears to trigger MIDI note 16, as well as 12 (which it is allocated)..... and it triggers both again at a very fast repeat, rather than held note.

The only 16 I can see in the code is:

"const static byte E0 = 16;" but I cannot think that this is the issue, or it would surely affect every button?

I am stumped as to how this is occuring, even if there is a soldering issue somewhere..

## Steps to reproduce the problem

### Hardware
Teensy 3.2

### Software versions:
**MIDI Controller library**: 3.10
**Arduino IDE**:  1.8.5 
**Operating System**: Windows 10  
  
(**Teensyduino**): 1.4.6
(**Encoder library**): 1.4.1
(**MIDIUSB library**): ? 1.0.3 ? 

### Settings in the IDE

### Full code 

#include <Encoder.h>
 
#include <MIDI_Controller.h>
      //Midi controller library

#include <ResponsiveAnalogRead.h>
      // analog smoothing
     
#include <Bounce.h>
      // include the Bounce library for 'de-bouncing' switches -- removing electrical chatter as contacts settle
      // usbMIDI.h library is added automatically when code is compiled as a MIDI device

using namespace ExtIO;
     // use standard arduino digital write pin mode etc. 

AnalogMultiplex buttonsmux1(1, { 10, 11, 12 } );
AnalogMultiplex buttonsmux2(0, { 10, 11, 12 } );
     // creating muxers
     //AnalogMultiplex(pin_t analogPin, { pin_t addressPin1, addressPin2, ... addressPinN } )
     //analogPin: the analog input pin connected to the output of the multiplexer
     //addressPin#: the digital output pins connected to the address lines of the multiplexer

DigitalLatch Arcadebuttons[] = {
  {buttonsmux1.pin(0), 0, 14, 127, 10},
  {buttonsmux1.pin(1), 1, 14, 127, 10},
  {buttonsmux1.pin(2), 2, 14, 127, 10},
  {buttonsmux1.pin(3), 3, 14, 127, 10},
  {buttonsmux1.pin(4), 4, 14, 127, 10},
  {buttonsmux1.pin(5), 5, 14, 127, 10},
  {buttonsmux1.pin(6), 6, 14, 127, 10},
  {buttonsmux1.pin(7), 7, 14, 127, 10},
     // Create 8 new instances of the class 'DigitalLatch', on the 8 pins of the buttonsmux1
     // that send MIDI notes on channel 14
     //Digital(pin_t pin, uint8_t note, uint8_t channel, uint8_t velocity = 127, latch time)
     //pin: the digital input pin to read the state from, the internal pull-up resistor will be enabled
     //note: the MIDI note number [0, 127]
     //channel: the MIDI channel [1, 16]
     //velocity: the MIDI velocity of the Note events [1, 127], how hard the key/button is struck 
};

DigitalLatch Arcadebuttonextras[] = {
  {buttonsmux2.pin(3), 8, 14, 127, 10},
  {buttonsmux2.pin(4), 9, 14, 127, 10},
     //extra two arcade buttons
};

Digital Navigation[] = {
  {buttonsmux2.pin(5), 10, 14, 127},
  {buttonsmux2.pin(6), 11, 14, 127},
  {buttonsmux2.pin(7), 12 , 14, 127},
      // Create 3 new instances of the class 'Digital' for navigation buttons, on the 3 pins of the buttonsmux2
     // that send MIDI notes on channel 14
     //Digital(pin_t pin, uint8_t note, uint8_t channel, uint8_t velocity = 127, latch time)
     //pin: the digital input pin to read the state from, the internal pull-up resistor will be enabled
     //note: the MIDI note number [0, 127]
     //channel: the MIDI channel [1, 16]
     //velocity: the MIDI velocity of the Note events [1, 127], how hard the key/button is struck 
};
  
//CONSTANT VALUES

  const static byte Channel_Volume = 0x7; 
     // controller number 7 is defined as Channel Volume in the MIDI implementation.
  const static size_t analogAverage = 8; 
     // Use the average of 8 samples to get smooth transitions and prevent noise
  const static byte velocity = 127; 
     // the maximum velocity, since MIDI uses a 7-bit number for velocity.
  const static int  latchTime = 3000;  
     // the amount of time (in ms) the note is held on. Read documentation or see source code for more information.
  const static byte C4 = 60; 
     // note number 60 is defined as middle C in the MIDI implementation.
  const static byte E0 = 16; 
    // note number 60 is defined as middle C in the MIDI implementation, so 16 is E0
  const byte Channel = 14; 
     // MIDI channel 14
  const byte Controller = 0x14; 
     // MIDI controller number
  const int speedMultiply = 1; 
      // no change in speed of the encoder
      

void setup() {

}      
void loop() {

MIDI_Controller.refresh();
}

## Steps taken to try to diagnose or solve the problem

Changed MIDI notes triggered - similar issues. 
Reconnected all control pins and rechecked continuity/orientation
Checked for continuity/short circuit across all Mux and Teensy pins
Tried various latch lengths in code. 
Removed "  const static byte E0 = 16; 
    // note number 60 is defined as middle C in the MIDI implementation, so 16 is E0
  const byte Channel = 14; "
Experimented with affected buttons on different pins/replacement buttons - problem is linked to pins themselves. Moving button from Pin 7 to Pin 3 on Mux2 created same issue on Pin 6 of Mux 2 (?)
Grounded all floating pins. Tried with and without "Enable" pins grounded.

the-duckchild, 2019-06-20:
Problem now solved with help from PJRC forum, which I will share here in case anyone experiences similar issues and googles this - Teensy 3.2 was running the code so fast that signals were not give time to stabilise. Adding 5 microseconds delay between lines 7 & 8 of the AnalogMultiplex.cpp library file (in src/ExtendedInputOutput) resolved the issue and all buttons now working correctly.

Also worth adding the same to the Analog section of AnalogMultiplex.cpp

tttapa, 2019-06-20:
Thank you for posting your solution!

I'll add the fix to the library as soon as my exams are over, leaving the issue open for now.

the-duckchild, 2019-06-20:
no problem. And thanks so much for the library, it has made this project so much easier than it would otherwise have been. Good luck with the exams!

closed 2019-10-09 by tttapa