tttapa/MIDI_controller #88: Change MIDI Message of Control Object

smp4488, 2019-06-18:
## Description of the problem or question
First off, thank you for creating this library. It has been very useful for the project i'm working on. Is it possible to change the MIDI message of a control after it has been created? I ask because i'm using Guitarix and it does not support MIDI channels. I have four potentiometers and five bank selectors. I have the bank selectors working and they change the channel appropriately, but Guitarix doesn't honor the channel on the CC message. So instead of changing the channels of the pots i'm wondering if it would be possible to change the message that is sent by the control? 

## Steps to reproduce the problem

### Hardware

**Arduino board**:  ? Arduino UNO, Arduino Leonardo, Arduino DUE, Arduino MEGA, Arduino Micro

**Schematic**: ?  

### Software versions:

**MIDI Controller library**: ? 3.1.0  
**Arduino IDE**: ? 1.8.9
**Operating System**: ? Windows
**Operating System version**: ? 10
(**MIDIUSB library**): ? 1.0.4 

### Settings in the IDE

### Full code 
```cpp
#include <MIDI_Controller.h>

DigitalLatch switches[] = {
    {2, MIDI_CC::Effects_1, 1},
    {3, MIDI_CC::Effects_2, 1},
    {4, MIDI_CC::Effects_3, 1},
    {5, MIDI_CC::Effects_4, 1},
    {6, MIDI_CC::Effects_5, 1}
};

Analog knobs[] = {
    {A0,  MIDI_CC::General_Purpose_Controller_1, 1},
    {A1,  MIDI_CC::General_Purpose_Controller_2, 1},
    {A2, MIDI_CC::General_Purpose_Controller_3, 1},
    {A3, MIDI_CC::General_Purpose_Controller_4, 1}
};

Bank bank(5);
BankSelector bankSelector(bank, { 2, 3, 4, 5, 6 } );

void setup()
{
    bank.add(knobs, Bank::CHANGE_CHANNEL);
}

void loop() // Refresh all inputs
{
    MIDI_Controller.refresh();
}
```
## Steps taken to try to diagnose or solve the problem
I tried to create a new object on ```knobs[0]``` but i get compile errors about the function not existing. I'm not sure how I would swap out the objects.

## The goal of your project and additional information
I'm creating a MIDI controller for Guitarix to control individual effects.

tttapa, 2019-06-18:
Did you try the following?
```cpp
bank.add(knobs, Bank::CHANGE_ADDRESS);
```
Or does that not fit your needs?  
It should change the controller number instead of the channel number.

By the way, the number of channels in your case is 4, not 5, so you could use `Bank bank(4);` (it's not related to the number of bank selector pins).

smp4488, 2019-06-18:
That was exactly what I needed. I'm not sure how I missed that. Thank you very much for the quick comment and support!

closed 2019-06-18 by smp4488

tttapa, 2019-06-19:
Glad to hear you got it working!