tttapa/MIDI_controller #90: I2C Expander

dealero, 2019-07-14:
Hi, I'm using Arduino Leonardo and your Library works perfect! However I'd like to make MIDI controller using I2C expanders such as PCF8574. Is there possible to implement this? I tried to do this, but no success. In my MIDI controller I want to put 8 encoders with buttons and additionally 16 more switches.

tttapa, 2019-07-23:
You can implement this rather easily by implementing the ExtendedIOElement interface:

https://github.com/tttapa/MIDI_controller/blob/815c86950e0a7323a2b89e123e59471158b88e04/src/ExtendedInputOutput/ExtendedIOElement.h#L31-L39

The documentation of ExtIO in the MIDI Controller library is rather poor, so I'd recommend checking out its successor, the Control Surface library. 

Take a look at the interface here:  
https://github.com/tttapa/Control-Surface/blob/a310ba8a19a4b929c6ef7d4758bcc1bb70436b1b/src/Hardware/ExtendedInputOutput/ExtendedIOElement.hpp#L70-L142  
You have to implement all pure virtual functions (`virtual ... = 0;`).

Take a look at how the AnalogMultiplex class is implemented:  
https://github.com/tttapa/Control-Surface/blob/master/src/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp

Do note that you can't use rotary encoders with I2C expanders, as they require interrupts. You could implement it, but it's not supported by the library.

cferrarini, 2019-11-12:
I´m trying to address some digital pins to a PCF8574 i2c port expander but my programing skills are not so good, maybe some ready library could create virtual i2c pins so this midi  library would not need to include the i2c code and would see them as regular pins. Any ideas?