tttapa/MIDI_controller #92: MIDI Controller - Rotary Encoder

nickho74, 2019-08-03:
Hi Pieter,

I was testing the Rotary Encoder.hpp and would like to suggest a function.

1 - Rotary Encoder :
a) When turning Encoder to the Left, send a Note On. (direction +1)
b) When turning Encoder to the Right, Send a differen Note On. (direction -1)

Can you advise me where to do modification on the code?

2 - Rotary Encoder to Multiplexer :
a) Is it possible to assign Rotary Encoder to Multiplexer 16ch?

Please advise.

Thanks
Nick Ho

tttapa, 2019-08-04:
1. https://github.com/tttapa/MIDI_controller/blob/5781d45369ffa3c4add30258b94d6221071122f5/src/MIDI_Outputs/RotaryEncoder.hpp#L67-L80

In Control Surface, it's even easier:  
https://github.com/tttapa/Control-Surface/blob/cd08dea14772fa4c57ee0d39d764ae19b8c44f2c/src/MIDI_Senders/RelativeCCSender.hpp#L45-L55
```cpp
    static void send(long delta, MIDICNChannelAddress address) {
        while (delta != 0) {
            // Constrain relative movement to +/-15 for
            // Mackie Control Universal compatibility
            long thisDelta = constrain(delta, -15, 15);
            uint8_t msgVal = mapRelativeCC(thisDelta);
            // send a Control Change MIDI event
            Control_Surface.MIDI().sendCC(address, msgVal);
            delta -= thisDelta;
        }
    }
``` 

2. No, because you cannot refresh it quickly enough. I might add support for IO expanders though, probably one with interrupt capability.

nickho74, 2019-08-05:
ok Thanks for the reply. I will try the code.

closed 2019-10-09 by tttapa