tttapa/MIDI_controller #95: Connecting multiple multiplexers to analog inputs for micro

Shuvam1997, 2019-08-24:
I love the code and it works perfectly for a single multiplexer...
However, I need to use 4 multiplexers and 32 knobs attached to them for my project and I tried several combinations to do this...

It shows the following error
[image]

The code I typed had the modifications as below:
AnalogMultiplex multiplexer[] = {
{A0, {10, 16, 14}},
{A1, {10, 16, 14}},};

closed 2019-08-24 by Shuvam1997

tttapa, 2019-08-24:
You declared it as an array, so you have to access the array element first, before calling the .pin method on it.

You could try something like this:

```cpp
AnalogMultiplex multiplexers[] = {
  {A0, {10, 11, 12, 13}},
  {A1, {10, 11, 12, 13}},
  {A2, {10, 11, 12, 13}},
  // ...
};

Analog potentiometers[] = {
  { multiplexers[0].pin(0), 0x1, 1 },
  { multiplexers[0].pin(1), 0x2, 1 },
  { multiplexers[0].pin(2), 0x3, 1 },
  // ...
  { multiplexers[1].pin(0), 0x11, 1 },
  { multiplexers[1].pin(1), 0x12, 1 },
  // ...
  { multiplexers[2].pin(0), 0x21, 1 },
  // ...
};
```

Shuvam1997, 2019-08-25:
Wow dude, thanks a lot....it works great...

I am really new to the arduino IDE and all this and it means a lot to me
that you replied so fast...