tttapa/MIDI_controller #96: Can the name of the board be changed?

sincos5, 2019-10-02:
I want to create a midi device but would like to change the name of it. Can that be achived? 
thanks!

tttapa, 2019-10-02:
It depends on the board you're using. For Teensy, just edit the src/MIDI_Interfaces/Teensy-USB-MIDI-name.c. For most Arduinos, you'll have to edit the MIDIUSB library.

sincos5, 2019-10-04:
thanks! im usingan arduino pro micro and it works really fine so far. 
thank you alot for the library.

sincos5, 2019-10-08:
After a short search I found out this>
http://liveelectronics.musinou.net/MIDIdeviceName.php
a mini tutorial to edit your board name

closed 2019-10-09 by tttapa

sincos5, 2021-08-05:
hey ttapa! I'm coming back after a long time wanting to know if its possible to connect two different boards without creating interference with one another or the only way to do it is assigning different midi channels for each board.

I will explain the issue in detail>
created one midi controller (a footswitch) a long time ago, and now, i created another (a gitarhero guitar) but now when i connect them to the PC, ableton does recognize them as individual midi controllers (IE, i can assign each to one individual track), but the notes that both have in common get overlapped.
for more info, both are made with Arduino micro, and i had manipulated the code to give them different names.
Hope you are ok, and as always thanks a lot.

tttapa, 2021-08-05:
This sounds like an Ableton-related issue, the computer should see them as two separate devices. If Ableton doesn't, you'll have to find another way to discriminate the messages, e.g. by using different channels.

sincos5, 2021-08-05:
could it be that im using in both devices the same midi information?
in one device im using a button on pin 3 that is a general porpouse controller 1 in channel 1
in the other device, im using a button on pin 17, general purpose controller 1 in channel 1.

if thats the case, is there a way to separate them or they always will be linked?

thanks a lot, this library is the best.

tttapa, 2021-08-05:
Like I said, your computer sees them as separate messages from separate devices, but it sounds like Ableton merges them. I have no experience with Ableton, so I'm afraid I can't really help you with that.

sincos5, 2021-08-05:
you where right, looking into ableton manual i found this>
`The MIDI Ports table lists all available MIDI input and output ports. To use an input port for remote control of Live, make sure the corresponding switch in its Remote column is set to ”On.” You can use any number of MIDI ports for remote mapping; **Live will merge their incoming MIDI signals.** `
font>
https://www.ableton.com/en/manual/midi-and-key-remote-control/