tttapa/MIDI_controller #98: How to send midi over serial

tbulaon1337, 2019-10-31:
Hi All, 

I have no idea what I'm doing and i am looking at sending midi over serial but I am unsure as to how to do it. Here is my code:

/*
This is an example of the "Digital" class of the MIDI_controller library.
Connect a push buttons to digital pin 2. Connect the other pin of the button to the ground, 
a pull-up resistor is not necessary, because the internal one will be used. 
This button will play MIDI note C4 when pressed.
Map it in your DAW or DJ software.

Written by tttapa, 08/09/2017
https://github.com/tttapa/MIDI_controller
*/

#include <MIDI_Controller.h> // Include the library

const uint8_t velocity = 0b1111111; // Maximum velocity (0b1111111 = 0x7F = 127)
SerialMIDI_Interface(Serial_t &serial, unsigned long baud)

// Create a new instance of the class 'Digital', called 'button', on pin 2, that sends MIDI messages with note 'C4' (60) on channel 1, with velocity 127
Digital button1(2, 1, 1, velocity);
Digital button2(3, 2, 1, velocity);
Digital button3(4, 3, 1, velocity);
Digital button4(5, 4, 1, velocity);
Digital button5(6, 5, 1, velocity);
Digital button6(7, 6, 1, velocity);
Digital button7(8, 7, 1, velocity);
Digital button8(9, 8, 1, velocity);

void setup() {}

void loop() {
  // Refresh the button (check whether the button's state has changed since last time, if so, send it over MIDI)
  MIDI_Controller.refresh();
}

Im not sure what to put in the bracketed sections of SerialMIDI_Interface

tttapa, 2019-11-02:
The most versatile option is:
```cpp
SerialMIDI_Interface<decltype(MySerial)> midi = {MySerial, baud_rate};
```
For hardware Serial ports, you can use the following:
```cpp
HardwareSerialMIDI_Interface midi = {Serial1, baud_rate};
```
For the default `Serial` port, you can use:
```cpp
USBSerialMIDI_Interface midi = {baud_rate};
```

tbulaon1337, 2019-11-05:
Solved. I just bought a new board that lets me use midi over USB. 

Thanks.

closed 2019-11-09 by tttapa