tttapa/MIDI_controller #99: unsupported board: Adafruit Bluefruit Feather nRF52840 Express

lxschwalb, 2019-12-15:
## Description of the problem or question
This is the full error message when my code includes the line "#include <MIDI_Controller.h>"

WARNING: library MIDI_controller-3.1.1 claims to run on (avr, sam, samd, teensy) architecture(s) and may be incompatible with your current board which runs on (nrf52) architecture(s).
WARNING: library MIDIUSB claims to run on (avr, sam, samd) architecture(s) and may be incompatible with your current board which runs on (nrf52) architecture(s).
In file included from /home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Controller/../MIDI_Interfaces/USBMIDI_Interface.h:11:0,
                 from /home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Controller/MIDI_Controller_Class.h:4,
                 from /home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/MIDI_Controller.h:10,
                 from sketch/buttons.cpp:1:
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:74:2: error: #error "Unsupported architecture"
 #error "Unsupported architecture"
  ^~~~~
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:125:2: error: 'EndpointDescriptor' does not name a type; did you mean 'MIDIJackinDescriptor'?
  EndpointDescriptor len;  // 9
  ^~~~~~~~~~~~~~~~~~
  MIDIJackinDescriptor
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:154:2: error: 'IADDescriptor' does not name a type; did you mean 'MIDI_EPDescriptor'?
  IADDescriptor                      iad;
  ^~~~~~~~~~~~~
  MIDI_EPDescriptor
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:156:2: error: 'InterfaceDescriptor' does not name a type; did you mean 'MIDI_ASInterfaceDescriptor'?
  InterfaceDescriptor                Audio_ControlInterface;
  ^~~~~~~~~~~~~~~~~~~
  MIDI_ASInterfaceDescriptor
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:160:2: error: 'InterfaceDescriptor' does not name a type; did you mean 'MIDI_ASInterfaceDescriptor'?
  InterfaceDescriptor                Audio_StreamInterface;
  ^~~~~~~~~~~~~~~~~~~
  MIDI_ASInterfaceDescriptor
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:208:1: error: expected class-name before '{' token
 {
 ^
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:213:2: error: 'EPTYPE_DESCRIPTOR_SIZE' does not name a type
  EPTYPE_DESCRIPTOR_SIZE epType[2];   ///< Container that defines the two bulk MIDI IN/OUT endpoints types
  ^~~~~~~~~~~~~~~~~~~~~~
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:222:21: error: 'USBSetup' has not been declared
   int getDescriptor(USBSetup& setup);
                     ^~~~~~~~
/home/alex/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:224:14: error: 'USBSetup' has not been declared
   bool setup(USBSetup& setup);
              ^~~~~~~~
In file included from /home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/MIDI_Controller.h:28:0,
                 from sketch/buttons.cpp:1:
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/SerialMIDI_Interface.h:82:61: error: 'Serial_' was not declared in this scope
 class USBSerialMIDI_Interface : public SerialMIDI_Interface<Serial_>
                                                             ^~~~~~~
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/SerialMIDI_Interface.h:82:61: note: suggested alternative: 'Serial'
 class USBSerialMIDI_Interface : public SerialMIDI_Interface<Serial_>
                                                             ^~~~~~~
                                                             Serial
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/SerialMIDI_Interface.h:82:68: error: template argument 1 is invalid
 class USBSerialMIDI_Interface : public SerialMIDI_Interface<Serial_>
                                                                    ^
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/SerialMIDI_Interface.h: In constructor 'USBSerialMIDI_Interface::USBSerialMIDI_Interface(long unsigned int)':
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/SerialMIDI_Interface.h:85:49: error: class 'USBSerialMIDI_Interface' does not have any field named 'SerialMIDI_Interface'
   USBSerialMIDI_Interface(unsigned long baud) : SerialMIDI_Interface(Serial, baud) {}
                                                 ^~~~~~~~~~~~~~~~~~~~
In file included from /home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/MIDI_Controller.h:29:0,
                 from sketch/buttons.cpp:1:
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/DebugMIDI_Interface.h: At global scope:
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/DebugMIDI_Interface.h:95:65: error: 'Serial_' was not declared in this scope
 class USBDebugMIDI_Interface : public SerialDebugMIDI_Interface<Serial_>
                                                                 ^~~~~~~
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/DebugMIDI_Interface.h:95:65: note: suggested alternative: 'Serial'
 class USBDebugMIDI_Interface : public SerialDebugMIDI_Interface<Serial_>
                                                                 ^~~~~~~
                                                                 Serial
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/DebugMIDI_Interface.h:95:72: error: template argument 1 is invalid
 class USBDebugMIDI_Interface : public SerialDebugMIDI_Interface<Serial_>
                                                                        ^
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/DebugMIDI_Interface.h: In constructor 'USBDebugMIDI_Interface::USBDebugMIDI_Interface(long unsigned int)':
/home/alex/Arduino/libraries/MIDI_controller-3.1.1/src/./MIDI_Interfaces/DebugMIDI_Interface.h:98:48: error: class 'USBDebugMIDI_Interface' does not have any field named 'SerialDebugMIDI_Interface'
   USBDebugMIDI_Interface(unsigned long baud) : SerialDebugMIDI_Interface(Serial, baud) {}
                                                ^~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Adafruit Bluefruit Feather nRF52840 Express.

## Steps to reproduce the problem
Select board as Adafruit Bluefruit Feather nRF52840 Express. Add "#include <MIDI_Controller.h>" to the code.

### Hardware
Adafruit Bluefruit Feather nRF52840 Express

### Software versions:

MIDI Controller library: 3.1.1  
Arduino IDE: 1.8.9  
Operating System: Lubuntu
MIDIUSB library: 1.0.4 (also doesn't support this board)

## Steps taken to try to diagnose or solve the problem
? Tried the MIDI debug mode, used a MIDI monitor, ... ?

## The goal of your project and aditional information
Make usb midi controller that can also receive midi over bluetooth

tttapa, 2019-12-15:
This board is not supported, and I have no plans to add support for it. 

The updated version of this library, Control Surface, does have experimental MIDI over BLE support, but only on ESP32 boards.

You can reuse the BLE MIDI parser, but you'll have to implement your own functions to receive and send Bluetooth packets.

lxschwalb, 2019-12-15:
Thanks for the quick response and suggestions