tttapa/MIDI_controller #100: Is the Nano Every board compatible?

nesdnuma, 2020-01-17:
## Description of the problem or question

Is the new Arduino Nano Every supposed to be compatible with the library?

The documentation tells this:

> The USB interface is managed by a dedicated SAMD11 that is factory loaded with the proper firmware to program the ATMega4809 microcontroller and let the Arduino NANO Every behave as it should when connected to a host computer for programming and serial communication. 

Thank you.

joshnishikawa, 2020-01-17:
This library was written specifically for the "Teensy" line of microcontrollers and those are the only thing I've personally tested on. I have no idea how well the library works on the Nano Every (if at all) but I hope you do get it working.

tttapa, 2020-01-18:
The ATMega4809 has no USB support, so direct MIDI over USB is not possible using this board.  
You could use MIDI over Serial (using 5-pin DIN plugs) or Hairless MIDI. If you find or write Serial to MIDI USB firmware for the SAMD11, you could in theory add MIDI USB capabilities, but that would be a huge waste of time, in my opinion. Just buying a Teensy (or an Arduino Micro) would be a much cleaner and easier solution.  
Also see the [MIDI over USB page](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html).

I no longer actively support this library. All new development happens in the [**Control Surface**](https://github.com/tttapa/Control-Surface) library. I'll push some fixes for the Nano Every later today, so that everything compiles. I cannot test anything, because I don't own a Nano Every board.

tttapa, 2020-01-22:
I forgot to mention: the Nano Every compatible code is on the `arduino-api` branch:  
https://github.com/tttapa/Control-Surface/tree/arduino-api

closed 2020-01-22 by tttapa

nesdnuma, 2020-01-23:
Thank you for the information. I'll probably use a Teensy then.