tttapa/MIDI_controller #101: How to Echo MIDI IN to MIDI OUT?

nickho74, 2020-01-18:
Hi Pieter,

I have been using most of your code in my MIDI project hardware in great success. Thank you !

One questions comes in mind, I have since added in a MIDI IN port into the project.

I would like to know where I can edit or add in in the Sketch?
I want to echo the incoming MIDI data from another MIDI hardware, into the existing project hardware. and then together output MIDI.
I have built a MIDI IN and MIDI OUT port on the hardware.

I have tested the FortySevenEffects MIDI library using the MIDI.turnThruOn() function and works fine in the project.

But I would like to know if there is a way in your sketch to do the same function.

Thank you.

tttapa, 2020-01-20:
The MIDI controller library does not support MIDI in. Its successor, [**Control Surface**](https://github.com/tttapa/Control-Surface) does. You can use the MIDI Pipes from the [**MIDI Routing module**](https://tttapa.github.io/Control-Surface-docDoxygen/df/ddc/group__MIDI__Routing.html).  
[**Example**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dc4/MIDI_Pipes-Routing_8ino-example.html)

Control Surface uses the last MIDI interface as its default interface. If you have multiple interfaces, it's recommended to add [`midi.setAsDefault()`](https://tttapa.github.io/Control-Surface-docDoxygen/d9/d73/classMIDI__Interface.html#a922dd87d272bb2be2771c43ac974481a) to your setup (before calling `Control_Surface.begin()`). Control Surface only initializes and updates its own MIDI interface, so you have to call `othermidi.begin()` and `othermidi.update()` yourself.

nickho74, 2020-02-03:
ok thank you.

I will try the Control Surface sketch.

closed 2020-02-03 by nickho74