tttapa/MIDI_controller #102: no response when change banks and instruments

xtramo, 2020-01-20:
## Description of the problem or question
Using Arduino leonrado connected via usb to pi rasberry 4 to play the yoshimi synth.
Im using your library and have successfully used the arduino to play on the yoshimi synth.
However I cant seem to get the midi to change the instruments on yoshimi?
Im using your code to test and nothing happens?
Ex.07.BankSelectors

## Steps to reproduce the problem

### Hardware

**Arduino board**:  ? , Arduino Leonardo, pi rasberry 4
**Schematic**: ?  using your example

### Software versions: pi rasbery rasbian latest , arduino 1.8.10

**MIDI Controller library**: ? 3.1.1 
**Arduino IDE**: ?1.8.10
**Operating System**: ? Windows, , ?  
**Operating System version**: ? 10,  and rasbian, arduino 

(**Encoder library**): ? 1.4.1 ?  
(**MIDIUSB library**): ? 1.0.3 ? 

### Settings in the IDE

### Full code 

```cpp
// Your code here
/*
  This is an example of the "Bank" class of the MIDI_controller library.
  Connect two potentiometers to analog pins A0 and A1,
  and two pushbuttons to pins 2 and 3.

  Connect push buttons to pins 11 and 12, and 4 LEDs (+ current limiting resitors) to pins 4, 5, 6 and 7).

  When bank 1 is selected:
    Potentiometer A is channel volume of track 1      (Controller number 0x07, MIDI channel 1)
    Potentiometer B is channel volume of track 2      (Controller number 0x07, MIDI channel 2)
    Mute button A is the mute button for track 1      (Note number 0x10, MIDI channel 1)
    Mute button B is the mute button for track 2      (Note number 0x11, MIDI channel 1)
    The LED on pin 4 lights up.

  When bank 2 is selected:
    Potentiometer A is channel volume of track 3      (Controller number 0x07, MIDI channel 3)
    Potentiometer B is channel volume of track 4      (Controller number 0x07, MIDI channel 4)
    Mute button A is the mute button for track 3      (Note number 0x12, MIDI channel 1)
    Mute button B is the mute button for track 4      (Note number 0x13, MIDI channel 1)
    The LED on pin 5 lights up.

  When bank 3 is selected:
    Potentiometer A is channel volume of track 5      (Controller number 0x07, MIDI channel 5)
    Potentiometer B is channel volume of track 6      (Controller number 0x07, MIDI channel 6)
    Mute button A is the mute button for track 5      (Note number 0x14, MIDI channel 1)
    Mute button B is the mute button for track 6      (Note number 0x15, MIDI channel 1)
    The LED on pin 6 lights up.

  When bank 4 is selected:
    Potentiometer A is channel volume of track 7      (Controller number 0x07, MIDI channel 7)
    Potentiometer B is channel volume of track 8      (Controller number 0x07, MIDI channel 8)
    Mute button A is the mute button for track 7      (Note number 0x16, MIDI channel 1)
    Mute button B is the mute button for track 8      (Note number 0x17, MIDI channel 1)
    The LED on pin 7 lights up.

  This allows you to control multiple tracks with only a limited amount of physical potentiometers and buttons

  Map accordingly in your DAW or DJ software.

  Written by Pieter P, 08-09-2017
[... 147 lines cut ...]
    bankSetting 1 is selected.

    BankSelector(bank, { button increment pin }, number of bankSettings);

- One momentary switch (push button) and multiple LEDs

    Pressing the button increments the bankSetting number and turns on the respective LED,
    When starting the program, bankSetting 1 is selected and LED 1 is on,
    When the button is pressed, bankSetting 2 is selected, LED 1 turns off and LED 2 turns on,
    When the button is pressed again, bankSetting 3 is selected, LED 2 turns off and LED 3 turns on.
    When the last bankSetting is selected, and the button is pressed,
    bankSetting 1 is selected, the last LED turns off and LED 1 turns on.
    
    BankSelector(bank, { button increment pin }, { led 1 pin, led 2 pin, ... , led n pin });

Note: a switch is 'off' or 'released' when it doesn't conduct. The digital value 
on the input will therefore be HIGH (because of the pull-up resistor)
*/
```

## Steps taken to try to diagnose or solve the problem
? Tried the MIDI debug mode, used a MIDI monitor, ... ? Im new to this how would I test as he usb is connected to the pi now and not the pc?

## The goal of your project and aditional information
To play the Yoshimi via the arduino buttons and sensors.

tttapa, 2020-01-20:
I haven't been able to reproduce this problem. I just tested the code you posted on an Arduino UNO, and it worked fine.  
The bank only changes the channel or the note number (depending on whether you're using `CHANGE_CHANNEL` or `CHANGE_ADDRESS`). It doesn't change the instrument.

Changing instruments is usually done by changing patches or programs using MIDI program change events.  
You could use something like this:

```cpp
MIDI_Controller.MIDI()->send(PROGRAM_CHANGE, channel, program);
// channel [1, 16] and program [0, 127]
```

xtramo, 2020-01-21:
Thanks for getting back to me.
Please explain as Im using a Leonardo? how did you verify that this works if you didnt duplicate my setup? the usb is connected to the rasberry?
How would I verify this if I wanted to test on a windows pc? I have fl studio on win 10.

tttapa, 2020-01-21:
90% of the code is the same, only the MIDI backend is different on Leonardo. All code has been tested on a Leonardo before.

MIDI is platform independent. I haven't tried Yoshimi, but I did confirm that the bank selector correctly changes the channel or address of the MIDI messages in question. You can confirm this by using a MIDI monitor on your PC or RPi.  
If it doesn't change the instrument in Yoshimi, this is most likely a problem with the configuration in Yoshimi.

> Im using your code to test and nothing happens?

What does "nothing happens" mean?

xtramo, 2020-01-21:
again thanks for the quick response.
Im not a pro when it comes to these things just getting into this amazing
tech.
so let me explain from a layman perspective and maybe Im trying to do the
wrong thing.
When the synth is open I have managed to map a potentiometer to the volume
control on the synth and can adjus this from the arduino.
Now Im trying to 1. change the "bank"? if that swhat it is from default to
say chorus, and then to change the instrument to a human voice and be able
to use a button to increment through different banks and instruments , so
when i use my arduino keypad it will then play those sounds.

this is the code Im using for the synth and arduino. Now I wanted to add
the bank program, instrument change and got stuck.?

#include “MIDI_Controller.h” // Include the library

const int c = 60;

const uint8_t velocity = 0b1111111; // Maximum velocity (0b1111111 = 0x7F =
127)
const uint8_t addresses[3][4] = { // button keymap
{c+0, c-5, c+2, c+3},
{c+4, c+5, c+6, c+7},
{c+12, c+9, c+14, c+11},
};

ButtonMatrix<3, 4> buttonmatrix({7, 6,5}, {11, 10, 9, 8} , addresses, 1,
velocity);

// Create a new instance of the class ‘Analog’, called ‘potentiometer’, on
pin A0,
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog potentiometer1(A0, MIDI_CC::Channel_Volume, 1);

// Create a new instance of the class ‘AnalogHiRes’, called
‘potentiometer’, on pin A1,
// that sends MIDI Pitch Bend messages on channel 1
AnalogHiRes potentiometer2(A1, 1);

// Create a new instance of the class ‘?’, called ‘potentiometer’, on pin
A2,
// that sends ??? on channel 1 Reverb
Analog potentiometer3(A2, MIDI_CC::Reverb, 1);

// Create a new instance of the class ‘?’, called ‘potentiometer’, on pin
A3,
// that sends ??? on channel 1 Echo
Analog potentiometer4(A3, MIDI_CC::Echo, 1);

void setup() {}

void loop() {
MIDI_Controller.refresh();
}

xtramo, 2020-01-21:
Hi,Im getting this error when trying the line you gave in your example code
07 :

MIDI_Controller.MIDI()->send(PROGRAM_CHANGE, [1, 16], [0, 127]);
.07.BankSelectors:58:1: error: 'MIDI_Controller' does not name a type

tttapa, 2020-01-21:
> MIDI_Controller.MIDI()->send(PROGRAM_CHANGE, [1, 16], [0, 127]);
> .07.BankSelectors:58:1: error: 'MIDI_Controller' does not name a type

You have to use that code inside of a function. By [1, 16], I meant "any number from 1 to 16", it's not actual code.

> Now Im trying to 1. change the "bank"? if that swhat it is from default to
> say chorus, and then to change the instrument to a human voice and be able
> to use a button to increment through different banks and instruments , so
> when i use my arduino keypad it will then play those sounds.

I don't think using a "MIDI Controller Bank" is what you're looking for. 
The voices and instruments are determined by the settings in Yoshimi, not by the notes sent by the MIDI Controller library.

You'll have to look through the documentation for Yoshimi to check if there are specific MIDI commands you can send to instruct it to change the instrument. Usually this is program change (see previous message) if there are less than 128 instruments, and a combination of MIDI bank select and program change if there are more instruments.  
MIDI bank select is different from the banking features of the MIDI Controller library.

A MIDI Bank Select event selects one of 16,384 different banks. Each bank has 128 patches/programs/instruments that can be selected using a MIDI Program Change event.

The MIDI Controller banking feature changes the address or channel of the MIDI control elements. For example, if you have four volume potentiometers, they send on MIDI channels 1-4 if the first bank is selected, on channels 5-8 if the second bank is selected, channels 9-12 if the third bank is selected, and so on.

xtramo, 2020-01-21:
Hi
Please can you send me the correct syntax to be able to select a bank and
then a patches/programs/instruments- this is what Im looking for and not as
a function but to be able to place this in the code I sent you .
 - much appreciated

tttapa, 2020-01-21:
I have exams right now, I don't have time. 

I already posted the code you need to change the program. Changing the banks is similar, but you'll have to do some research first, you can find a lot of information in the MIDI specs and probably in the Yoshimi documentation.

You'll then have to write code that decides when to change the instrument.

The Control Surface library has some of this built-in, like the Program Changer.

https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d7f/Program-Changer_8ino-example.html

xtramo, 2020-01-21:
please I have been looking really! ive got more than 100 forums/articles
Ive looked everywhere and nothing helps Ive changed code tried you name it !
thats why I reached out to you.
Im at the point where Ive just got no clue and all I need is that 1 or 2
lines of code that does the change - surely you have that - please

xtramo, 2020-01-21:
heads up the code you sent me now does work to a point!
so excited!
push button 3 works, it chooses the 3rd item in the 2nd row of the
instruments, button 4 works some time on the 1st row but button 1 doesnt.
but thats a huge start .
How do I change the bank?
yoshimi instruments are found under each bank:
Root 5, bank N where N goes from 1 to 115

xtramo, 2020-01-21:
my bad i figured out why it didnt play was because the instrument was
missing.
good luck with the exams