tttapa/MIDI_controller #104: Use Momentary as Toggle

fristaildg, 2020-01-30:
Hi! I'm super new at this and I'm sure this is a very simple task to achieve.
I need to send CC messages to control Bias FX pedals and have done this so far:
`#include <Control_Surface.h>

HairlessMIDI_Interface midi;

CCButton buttons[] = {
  {2, {MIDI_CC::Foot_Controller, CHANNEL_1}} 
};

void setup() {
  // put your setup code here, to run once:
  Control_Surface.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
  Control_Surface.loop();
}`

But every time I press the button it sends on (127) when I press and off (0) when I release the button. How can I change my code to make the button behave as a toggler?

Huge thanks in advance!!

tttapa, 2020-01-30:
You can use [`CCButtonLatched`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da6/classCCButtonLatched.html#details).  
Have a look at the [**MIDI Output Elements** module](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/dcd/group__MIDIOutputElements.html) for a complete overview.

luisFernandezDevSpark, 2020-01-30:
Yes I found it right after I asked you the question and it worked flawlessly.
Huge thanks for this library, it has made the life of a complete noob like me super easy!