tttapa/MIDI_controller #106: Problem to reload the code

dagame1981, 2020-03-31:
Hello
I followed the instructions you have detailed and everything works perfectly for me.
The problem comes when i want to add more code to the board when the ATMega has already been updated, when i upload the code in the arduino ID it gives me error and does not let me.
In the last part of your instructions you say that to put the code back you have to do it with the previously downloaded arduino fireware is only codes of numbers and letters and i do not understand it 
I hope it will help because i only need that to complete my project
Sorry for my bad training, i have started this project for a very short time
Thanks for your time
Greetings

tttapa, 2020-03-31:
You've probably followed the instructions for using Flip or dfu-programmer to flash the MIDI firmware arduino_midi.hex. Now you have to follow the same steps, but using the file Arduino-usbserial-atmega16u2-Uno-Rev3.hex instead of arduino_midi.hex. (In step 5, where you load the HEX file.)

dagame1981, 2020-04-01:
Done the steps that you indicate and when i give it execute in FLIP, I get an error in verify and it.
doesn't let me load any code.
Will i have to update the ATmega before those steps that you indicate?

dagame1981, 2020-04-01:
I realized that the resistor has not benn welded from the back.
Will it have something to do?

tttapa, 2020-04-01:
> Done the steps that you indicate and when i give it execute in FLIP, I get an error in verify and it.
doesn't let me load any code.

Please be more specific and post the exact error you're getting.  

Did you enter DFU mode by resetting the ATmega16U2?

> I realized that the resistor has not benn welded from the back.
> Will it have something to do?

No, this is only necessary on Rev1 boards.  
See this page: https://www.arduino.cc/en/Hacking/DFUProgramming8U2

dagame1981, 2020-04-01:
C:\Users\Pablo\Desktop\Arduino\MIDI_controller-3.1.1\examples\Ex.12.AnalogMultiplex\Ex.12.AnalogMultiplex.ino:20:13: warning: 'Analog' is deprecated

 Analog potentiometers[] = {

             ^

El Sketch usa 5210 bytes (16%) del espacio de almacenamiento de programa. El máximo es 32256 bytes.
Las variables Globales usan 490 bytes (23%) de la memoria dinámica, dejando 1558 bytes para las variables locales. El máximo es 2048 bytes.
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20

This is the error that appears when i try to upload the code to the board

tttapa, 2020-04-01:
Yes, I know that, but did you flash the Arduino-usbserial-atmega16u2-Uno-Rev3.hex firmware using Flip or dfu-programmer?

dagame1981, 2020-04-01:
yes i did but as  i told you before, when i flash it gives me an error in verify and appears in red

dagame1981, 2020-04-01:
[image]
Error in FLIP

tttapa, 2020-04-01:
Did you select the right HEX file? The filename seems wrong.

If it keeps failing, that means that the memory of the ATmega16U2 is corrupt.

dagame1981, 2020-04-01:
Yes ,select the correct HEX and it gives me that failure.
I'll try AVRDUDE.
THANKS

M1nCh3, 2020-04-01:
it keep telling me this error, how to fix it please?
exit status 1
'NORMAL_ENCODER' was not declared in this scope

tttapa, 2020-04-01:
@M1nCh3 This has nothing to do with the question at hand, please don't hijack issues, and open your own.

dagame1981, 2020-04-02:
My arduino seems to be damaged, i've tried another one and it worked for me,
Can i do this process with a Mega arduino? because i want to do my projectwith a Mega.
I would like to know if an LED can be turned on when i press a button and the LED stay on until i press the button again and it turns off

tttapa, 2020-04-02:
Yes, you could use a Mega, but I wouldn't recommend it. Using an Arduino with native USB MIDI support will be much easier.

There's an overview of Arduino boards and their USB MIDI capabilities here: [**MIDI over USB**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html)

dagame81, 2020-04-09:
Hello again, i am going to take a risk and will mount it with my Arduino Mega, i would like you to tell me how i can increase the codes of potenciometer and buttons , and tried to copy and paste the codes by changing the pins andwhen i have more than two it gives me error.
i have no idea about codes , i hope you help me
thanks

tttapa, 2020-04-09:
Please post the code you want help with, as well as your best attempt and the error message you get.

dagame81, 2020-04-09:
Arduino:1.8.10 (Mac OS X), Tarjeta:"Arduino/Genuino Uno"

/Users/pablogalindoaloy/Desktop/Arduino/MIDI_controller-3.1.1/examples/Ex.01.Potentiometer/Ex.01.Potentiometer.ino:22:13: warning: 'Analog' is deprecated
 Analog potentiometer(A0, MIDI_CC::Channel_Volume, 1);
             ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/pablogalindoaloy/Desktop/Arduino/MIDI_controller-3.1.1/examples/Ex.01.Potentiometer/Ex.01.Potentiometer.ino:23:13: warning: 'Analog' is deprecated
 Analog potenciometer(A1, MIDI_CC::Channel_Volume, 2);
             ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/pablogalindoaloy/Desktop/Arduino/MIDI_controller-3.1.1/examples/Ex.01.Potentiometer/Ex.01.Potentiometer.ino:24:13: warning: 'Analog' is deprecated
 Analog potenciometer(A2, MIDI_CC::Channel_Volume, 3);
             ^~~~~~~~~~~~~~~~~~~~~~~~
Ex.01.Potentiometer:24:22: error: redefinition of 'AnalogCC potenciometer'
 Analog potenciometer(A2, MIDI_CC::Channel_Volume, 3);
                      ^~
/Users/pablogalindoaloy/Desktop/Arduino/MIDI_controller-3.1.1/examples/Ex.01.Potentiometer/Ex.01.Potentiometer.ino:23:8: note: 'AnalogCC potenciometer' previously declared here
 Analog potenciometer(A1, MIDI_CC::Channel_Volume, 2);
        ^~~~~~~~~~~~~
Se encontraron varias bibliotecas para "SoftwareSerial.h"
Usado: /Users/pablogalindoaloy/Downloads/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial
Se encontraron varias bibliotecas para "MIDI.h"
Usado: /Users/pablogalindoaloy/Documents/Arduino/libraries/MIDI
Se encontraron varias bibliotecas para "MIDI_Controller.h"
Usado: /Users/pablogalindoaloy/Documents/Arduino/libraries/MIDI_controller-master
exit status 1
redefinition of 'AnalogCC potenciometer'

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

dagame81, 2020-04-09:
#include <MIDI.h>

/*
This is an example of the "Analog" class of the MIDI_controller library.
Connect a potentiometer to analog pin A0. This will be the MIDI channel volume of channel 1.
Map it in your DAW or DJ software.

Written by Pieter P, 08-09-2017
https://github.com/tttapa/MIDI_controller
*/

#include <MIDI_Controller.h> // Include the library

// Create a new instance of the class 'Analog', called 'potentiometer', on pin A0, 
// that sends MIDI messages with controller 7 (channel volume) on channel 1
Analog potentiometer(A0, MIDI_CC::Channel_Volume, 1);
Analog potenciometer(A1, MIDI_CC::Channel_Volume, 2);
Analog potenciometer(A2, MIDI_CC::Channel_Volume, 3);
void setup() {}

void loop() {
  // Refresh the MIDI controller (check whether the potentiometer's input has changed since last time, if so, send the new value over MIDI)
  MIDI_Controller.refresh();
}

tttapa, 2020-04-09:
Please note that the MIDI controller is no longer supported, and has been obsoleted by the [Control Surface](https://github.com/tttapa/Control-Surface) library.

The problem with your code is that you are creating two variables with the same name. This is not allowed.

It's best to create arrays instead of using a separate variable for each MIDI element:
- https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d7d/md_pages_Getting-Started.html
- https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d8d/Multiple-Control-Change-Potentiometers_8ino-example.html