tttapa/MIDI_controller #107: ESP32 RTPMIDI over Wifi cannot output CC

cferrarini, 2020-08-30:
Hi, I Tested The Applemidi interface example using ESP32 Doit board, connected to wifi and configured rtpmidi program in windows 8.
The NoteButton class worked ok, and the notes played in the DAW. But the CCPotentiometer did not work, the Daw does not receive any CC. although I tested the CCPotentiometer example in USBDebug Mode and the reading appeared in the Serial Monitor correctly. 
Assuming that the APPLEMIDI lib does not support CC, only notes... Is there a way to make it forward the CC to rtpmidi?
Its a shame because it works awesome via wifi!!!

The code I tried is this:
PBPotentiometer potentiometer = {
  15,        // Analog pin connected to potentiometer
  CHANNEL_1, // MIDI Channel 1
};

Best Regards!
Celso

tttapa, 2020-09-05:
AFAIK, AppleMIDI should support all MIDI messages. I'll try it later today. 

(`PBPotentiometer` sends MIDI Pitch Bend messages, not CC.)

cferrarini, 2020-09-05:
Sorry, I tried again with this same instance, in ESP32 all does fine with Hairless MIDI Interface (NoteButton and CCPotentiometer) , but with AppleMIDI Interface only NoteButton goes, CCPotentiometer does not respond at DAW.

// Instantiate a CCPotentiometer object
CCPotentiometer potentiometer = {
  15,                                   // Analog pin connected to potentiometer
  {MIDI_CC::Channel_Volume, CHANNEL_1}, // Channel volume of channel 1
};

The Whole Sketch:

#include <AppleMIDI.h>
USING_NAMESPACE_APPLEMIDI

#include <Control_Surface.h>
#include <MIDI_Interfaces/Wrappers/FortySevenEffects.hpp>

#include <ESPmDNS.h>
#include <WiFi.h>
#include <WiFiUdp.h>

#include "WiFi-Credentials.h" // See instructions above

// ----------------------------- MIDI Interface ----------------------------- //

// First create the AppleMIDI instance
APPLEMIDI_CREATE_INSTANCE(WiFiUDP, MIDI, "ESP32", 5004);
//                           │       │      │       └──── Local port number
//                           │       │      └──────────── Name
//                           │       └─────────────────── MIDI instance name
//                           └─────────────────────────── Network socket class

// Then wrap it in a Control Surface-compatible MIDI interface
FortySevenEffectsMIDI_Interface<decltype(MIDI) &> AppleMIDI_interface = MIDI;

// ------------------------------ MIDI Elements ----------------------------- //

// Add some MIDI elements for testing
using namespace MIDI_Notes;
NoteButton button1 = {
  5,                       // Push button on pin 5
  {note(C, 3), CHANNEL_1}, // Note C4 on MIDI channel 1
};
NoteButton button2 = {
  18,                       // Push button on pin 5
  {note(D, 3), CHANNEL_1}, // Note C4 on MIDI channel 1
};
NoteButton button3 = {
  19,                       // Push button on pin 5
  {note(E, 3), CHANNEL_1}, // Note C4 on MIDI channel 1
};
NoteButton button4 = {
  21,                       // Push button on pin 5
  {note(F, 3), CHANNEL_1}, // Note C4 on MIDI channel 1
};

// Instantiate a CCPotentiometer object
CCPotentiometer potentiometer = {
  15,                                   // Analog pin connected to potentiometer
  {MIDI_CC::Channel_Volume, CHANNEL_1}, // Channel volume of channel 1
};

// --------------------------- AppleMIDI callbacks -------------------------- //

void onAppleMidiConnected(const ssrc_t &ssrc, const char *name) {
  Serial << F("Connected to session ") << name << endl;
}
void onAppleMidiDisconnected(const ssrc_t &ssrc) {
  Serial << F("Disconnected") << endl;
}
void onAppleMidiError(const ssrc_t &ssrc, int32_t err) {
[... 69 lines cut ...]
[     6555.339] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Using teVirtualMIDI version: 1.3.0.43
[     6555.345] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Using teVirtualMIDI dll: C:\Windows\SYSTEM32\teVirtualMIDI32.dll
[     6555.350] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: DNSSD Apple™  <Bonjour> for Windows™ activated (333.10 (3331000))
[     6555.354] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Apple Inc./Bonjour: 3.0.0 (10), 3_0_0_10
[     6555.359] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: ---------------------------------------------------------------------------------------------------------------
[     6560.664] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: RX-MIDI:Horus:192.168.15.15:5005:RTP-Data: 80 61 92 96 00 C2 11 68 59 CA 66 FC 03 90 35 7F 
[     6560.670] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus: RTP-MIDI received
[     6560.892] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: RX-MIDI:Horus:192.168.15.15:5005:RTP-Data: 80 61 92 97 00 C2 1A 46 59 CA 66 FC 03 80 35 7F 
[     6560.899] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus: RTP-MIDI received
[     6561.199] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: RX-MIDI:Horus:192.168.15.15:5005:RTP-Data: 80 61 92 98 00 C2 26 1C 59 CA 66 FC 03 90 34 7F 
[     6561.204] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus: RTP-MIDI received
[     6561.250] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus:ESP32: Sending Receiver Feedback on control-port: 9298
[     6561.256] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: TX-CTRL:Horus:192.168.15.15:5004:Receiver Feedback: FF FF 52 53 FB 38 5C F8 92 98 00 00 
[     6561.411] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: RX-MIDI:Horus:192.168.15.15:5005:RTP-Data: 80 61 92 99 00 C2 2E A0 59 CA 66 FC 03 80 34 7F 
[     6561.417] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus: RTP-MIDI received
[     6561.692] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: RX-MIDI:Horus:192.168.15.15:5005:RTP-Data: 80 61 92 9A 00 C2 39 86 59 CA 66 FC 03 90 32 7F 
[     6561.698] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus: RTP-MIDI received
[     6561.961] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: RX-MIDI:Horus:192.168.15.15:5005:RTP-Data: 80 61 92 9B 00 C2 44 08 59 CA 66 FC 03 80 32 7F 
[     6561.967] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus: RTP-MIDI received
[     6562.251] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: TX-MIDI:Horus:192.168.15.15:5005:Synchronize: FF FF 43 4B FB 38 5C F8 00 00 00 00 00 00 00 00 03 E9 51 74 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[     6562.257] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus:ESP32: Sending Receiver Feedback on control-port: 929B
[     6562.262] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: TX-CTRL:Horus:192.168.15.15:5004:Receiver Feedback: FF FF 52 53 FB 38 5C F8 92 9B 00 00 
[     6562.398] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: RX-MIDI:Horus:192.168.15.15:5005:Synchronize: FF FF 43 4B 59 CA 66 FC 01 F0 39 0D 00 00 00 00 03 E9 51 74 00 00 00 00 00 C2 55 24 00 00 00 00 00 00 00 00 
[     6562.404] rtpMIDI_1_1_14_247 [ DIAG_VERBOSE ]: Horus: TX-MIDI:Horus:192.168.15.15:5005:Synchronize: FF FF 43 4B FB 38 5C F8 02 00 00 00 00 00 00 00 03 E9 51 74 00 00 00 00 00 C2 55 24 00 00 00 00 03 E9 57 75 
[     6562.410] rtpMIDI_1_1_14_247 [ DIAG_INFO    ]: Horus:ESP32: Latency=1537

tttapa, 2020-09-05:
The problem is that you're using GPIO15, which is connected to ADC2. You cannot use ADC2 while using the radio, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html.

cferrarini, 2020-09-05:
Sorry, I noticed it right now, its already working perfectly in pin 34 ADC1, Thanks!

closed 2020-09-05 by cferrarini