tttapa/MIDI_controller #109: Mapping logarithmic potentiometers to linear curve

kraftZeug, 2020-09-24:
First of all, I'd like to send you many many thanks for your code! It saved me a lot of time for my project (an electric guitar with onboard midi-controller) and it allowed me to focus on buiding and electronics

## Description of the problem or question
This is more of a question than an issue. I have no clue on how to integrate a function to map a logarithmic curve to a linear one.
Could you point me to the right direction?

**Arduino board**:  Teensy 3.2

tttapa, 2020-09-25:
As mentioned in the README, the MIDI Controller library is no longer supported. It has been replaced by the [Control Surface](/tttapa/Control-Surface) library.

With that library, you can use the [CCPotentiometer-Map](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d26/CCPotentiometer-Map_8ino-example.html) example.

You could use the following mapping function:
```cpp
analog_t log2lin(analog_t in) {
    constexpr analog_t max    = FilteredAnalog<7>::getMaxRawValue();
    // Tweak the position of the "knee" to match your potentiometer's spec
    constexpr uint32_t x_knee = max / 2; 
    constexpr uint32_t y_knee = max / 8;

    return in < y_knee
               ? in * x_knee / y_knee
               : (in - y_knee) * (max - x_knee) / (max - y_knee) + x_knee;
}
```

kraftZeug, 2020-09-26:
Yes! Thank you very much!
I'll go check Control Surface.

closed 2020-09-26 by kraftZeug